Ultimaker/Cura #2518: Ugly font with Russian locale

sadr0b0t, 2017-09-29:
Hello,

I am using Cura2.7 Linux on Ubuntu 16.04. Interface font with Russian locale is ugly in menu and dialogs. On the Print setup panel in the main window is ok.

See screenshot:
[image]

Ghostkeeper, 2017-10-04:
Doesn't reproduce for me on Ubuntu 17.04.

[image]

Let's see if other users of 16.04 find this, too.

sadr0b0t, 2017-12-21:
I have installed Cura 3.1 and have exactly same problem. It is pleasant to know, that Ubuntu 17.04 does not show this problem with Cura, but Ubuntu 16.04 LTS is going to be with us at least until 2021, so I hope this problem could be somehow fixed inside Cura build.

ChrisTerBeke, 2018-03-02:
3.3 will have updated fonts, should be fixed then. Please re-open if that's not the case.

closed 2018-03-02 by ChrisTerBeke

sadr0b0t, 2018-04-26:
Same story with Cura 3.3.1

[image]

sadr0b0t, 2018-04-26:
And now not only dialogs, but also print task settings are disturbed (in 3.1 they were ok)

[image]

sadr0b0t, 2018-04-26:
can't reopen

ianpaschal, 2018-04-26:
That was a message for us to re-open it.

I have just discussed this internally though, and our Estonian (but Russian speaking) colleague did some research in Russian to try and find a solution. He could not find any possible solution and since Russian Ubuntu users are a very, very small demographic of our total users we won't be able to spend any time on this issue.

Sorry about that.