Ultimaker/Cura #2519: Allow Import/Export of Machine Profiles

jed-frey, 2017-09-29:
I have a Creality CR-10 and the instructions for adding it to Cura 2.x on Windows mean digging around in the folder internals and dropping a profile in there.

Doing the same thing on Linux is a major PITA because of AppImage and not a set of instructions I would give to a newbie.

Allow Import/Export of Machine profiles similar to how material profiles can be imported & exported.

nallath, 2017-10-02:
It's not the best solution, but loading from project should also load the machine.

fieldOfView, 2017-10-02:
Note that it is not necessary to put the definition files in the application resources. You can already put them in the same folders inside the configuration folder (~/.local/shared/cura/2.7 or something like that)

jackha, 2018-08-16:
Creality CR-10 support is now bundled with Cura.

closed 2018-08-16 by jackha

fieldOfView, 2018-12-16:
Chances of Ultimaker putting development effort/time into this are slim to none. It could be implemented as a plugin.