Ultimaker/Cura #2520: all cura versions: rotating model changes slicing

jeroensomers, 2017-09-29:
When I rotate my model the slicer is filling up holes in my STL that is not good.

Example Cura 2.7.0
[image]
You can clearly see that the letters are sliced correctly.
[image]
Now I rotate the model 90 degrees
[image]
The letters are gone
[image]

Example Cura 15.04.6
[image]
You can clearly see that the letters are sliced correctly.
[image]
Now I rotate the model 90 degrees
[image]
The letters are gone
[image]

The model is made in sketchup
I added the model that I usesd for this.
[Fem sleutelhanger.zip](https://github.com/Ultimaker/Cura/files/1344633/Fem.sleutelhanger.zip)

fieldOfView, 2017-09-29:
Sketchup has a tendency to create 3d models that are unsuitable for 3d printing. They need to be "repaired".

What you could try (but no guarantees) is to use the "Search" in the Advanced sidebar to look for the "Union Overlapping Volumes" setting and turn it off.

jeroensomers, 2017-09-29:
Damnn, that was a very quick fix.

Thnx for the help!

closed 2017-09-29 by jeroensomers