Ultimaker/Cura #2525: Printer does not start at centre of print plate

dejavuNV, 2017-09-30:
Since upgrading to Cura 2.7.0, my object starts to print at head start position at bottom left corner of plate.
My printer in a Wanhao i3 v 2.1
In Cura the object is shown in the centre and saved but upon printing is printed on edje of plate.
Also a two piece object tries to print on the same spot, soon creating a tangled mess.
If I position the object in Cura at top right corner, the final print starts a little higher up than bottom left corner, does not overlap but is still way off required position.
If I print using the latest Cura that Wanhao have on their site, I have no problems at all
Any help would be appreciated.

ChrisTerBeke, 2017-10-04:
Sounds like the build volume in the machine settings in Cura do not match the actual dimensions of your printer. You can take a look in the "manage machines" window to see if the values there match your printer's physical properties.

Ghostkeeper, 2017-10-15:
In fact it seems like the "Origin at Center" machine setting is enabled. Can you look in the machine settings of your Custom FDM Printer and make sure that it's disabled?

dejavuNV, 2017-10-15:
Thanks ChrisTerBerke and GhostKeeper for your help. Sorry for late reply as I had been away.
Its turned out it was as you both said machine settings and the origin at centre was enabled and causing the problem. Another problem that has shown itself now is that the support material seems to bond to the object I'm printing more than it did with my previous version of Cura.
I have also noticed that the object prints much faster with the latest version and wonder if that's the problem. Maybe the main object is still a bit hot and not hardened enough before supports are added.
Have you any ideas with that one?
Thanks for your help, much appreciated.

Ghostkeeper, 2017-10-22:
> Another problem that has shown itself now is that the support material seems to bond to the object I'm printing more than it did with my previous version of Cura.

That could indeed be caused by many things. We didn't change the support settings outside of the UM3 profiles as far as I know, but maybe the temperatures or the speeds... In any case, take a look at the Support Z Distance setting. Maybe up that by one layer height. Or if you really think it's the speeds and the material not being solidified yet, you can increase the minimum layer time.

dejavuNV, 2017-10-25:
Thanks Ghostkeeper, I will check those settings you have suggested.

AIBABLY, 2017-11-08:
I have a printer type anite A6 program attached with the printer does not support files drawn from the Internet and thank you

closed 2017-11-14 by ChrisTerBeke

AIBABLY, 2017-11-14:
I have a printer type anite A6 program attached with the printer does not support files drawn from the Internet and thank you
من: ChrisTerBeke<mailto:notifications@github.com>
إرسال: ‏14 ‏تشرين الثاني,‏ ‏2017 ‏09:03 ‏م
إلى: Ultimaker/Cura<mailto:Cura@noreply.github.com>
نسخة: AIBABLY<mailto:Gendor58@hotmail.com>; Comment<mailto:comment@noreply.github.com>
الموضوع: Re: [Ultimaker/Cura] Printer does not start at centre of print plate(#2525)

Closed #2525<https://github.com/Ultimaker/Cura/issues/2525>.