Ultimaker/Cura #2526: Implement support for IPP 3D

wifiprintguy, 2017-09-30:
Implement support for IPP 3D, as described here:

http://www.pwg.org/3d/

nallath, 2017-10-02:
We already support 3mf, shouldn't that already be enough?

fieldOfView, 2017-10-02:
Cura creates gcode from 3d models. The goal of IPP 3D is specifically NOT sending g-code, but defines a high-level interface to 3D printers. The idea is you send models and how you want to have them printed to the printer, and the printer (or a cloud service) does the rest. 

When you say "Implement support for IPP 3D", what do you mean? Do you mean to create a IPP 3D server, which receives models+settings and automatically sends gcode to a connected printer (circumventing the UI) or do you mean to be able to have a highlevel interface to load/preview models and send those off to be printed (circumventing CuraEngine, and 90% of the rest of the GUI to set printer-specific settings)?

Ghostkeeper, 2017-10-11:
As far as I understand, this requires the IPP 3D service to start slicing and then printing. One way to implement that is to allow programs to send IPP 3D packages to Cura, which then slices it with the desired settings and allows you to send it to some printer (which?).

Problem is that the settings for this are intent-based and we don't have that sort of system in Cura yet. It's also a lot of work to implement this sort of thing.

ChrisTerBeke, 2017-10-11:
Depending on how you interpret the schematic the slicing can happen on either the printer or another (desktop) client. The protocol further describes all communication between 3D printer (output device), slicing/processing software and end-user. Also other parts like printer status and security are described. If Cura/Ultimaker would adopt this that would mean a rewrite of basically every part of the software stack.

[image]

(source: http://ftp.pwg.org/pub/pwg/candidates/cs-ipp3d10-20170210-5100.21.pdf)

nallath, 2017-10-11:
So i think it's safe to say; We won't do this.

closed 2017-10-11 by nallath

onesail, 2019-11-18:
IPP-3D should fit to your solution. it's mainly a protocol that handles job submission and job control from a client to the 3d-printer. The slicing-job is not part of the spec and can be done before or after job submission - depending on the printers capabilities. G-Code is explicitly allowed: Printers that do not support Slicing SHOULD support Documents conforming to a layered format such as PWG Safe G-Code [PWGGCODE]

Ghostkeeper, 2019-12-04:
I think then that implementing that IPP-3D protocol to submit jobs to a printer is precisely what Cura's output device plug-ins are designed to do. If someone has a printer that supports IPP-3D, then we'll leave it to them to implement a plug-in and perhaps submit it to Cura's plug-in browser. But I have no such printer available, so it's impossible to implement that reliably then.

onesail, 2019-12-04:
From my understanding Cura is required to print to Ultimaker Printers. The IPP-3D-protocol should be considered as an input interface to Cura. The protocol has been designed to interface between the "3d-modelling-software" and "print-3d-documents-software".
Is there on open API for input plugins, that allows to listen to network sockets? If yes, I guess we at http://ipp-software.com could implement such a plugin and offer it at the marketplace.
Does that make sende?

Ghostkeeper, 2019-12-06:
It can be done in a plug-in, yes. We haven't built that into Cura's framework (Uranium) though because we've never had the need for listening to a socket connection to accept an incoming model for slicing. Currently the main architecture for loading models assumes that you're reading a file. That is, you can write a [file reader plug-in](https://github.com/Ultimaker/UraniumExampleFileReaderPlugin), which is registered with a certain MIME type. When files of that MIME type are opened the `read` function of that plug-in gets called, which is responsible for reading out that file and placing an object in the scene that represents what's in that file. Uranium doesn't check if that file actually exists so in theory you could have a file with a different URI scheme as long as the extension is correct. But you won't be able to open it since Cura won't listen for the sockets then.

However there are also plug-ins that sideload models into the scene. For example the Support Blocker plug-in does this: It creates cubes in the scene whenever you click on the object. You can make an [Extension plug-in](https://github.com/Ultimaker/UraniumExampleExtensionPlugin) (the most generic type of plug-in). Upon starting, it can start a thread that listens for incoming connections. Upon receiving a connection it can read out the data from that stream and place an object in the scene.

Cura's framework is powerful enough for that to be implemented, but since we never had a need for it ourselves you'd have to do the legwork in the plug-in.

If you were unaware, you can offer it to be put on the marketplace here: https://contribute.ultimaker.com