Ultimaker/Cura #2527: Rotating a model and making the change permanent

thopiekar, 2017-09-30:
Hey,

I guess someone can answer the question easily. I need to rotate meshes in my nodes by 90 deg, but don't want that someone can undo it, eg. by right clicking on the model and reverting all transformations over there.
So what I did, is, that I rotate the node, request the transformed mesh, and return it back as the original mesh. Now, because I think the scene node will believe it is still rotate, I resetted the transformation matrix to a zero matrix. Sadly I get then a model, which is black, but seems to know about the mesh, because you can see the outline of the model still.
Am I missing something? Anything else I have to reset?

- https://github.com/thopiekar/CuraDassaultSystemesPlugins/commit/7e54d54877011f5bf1d81b132b0263ef3debc05e
- https://github.com/thopiekar/CuraDassaultSystemesPlugins/issues/20

Thanks!

fieldOfView, 2017-10-01:
How about instead of using SceneNode.rotate on the node, get the vertices of the node first, use UM.Mesh.MeshData.transformVertices to rotate the vertices and then put the vertices back into the SceneNode. That way the SceneNode does not know about the transformation.

thopiekar, 2017-10-01:
Just to understand the whole case.. How is model info handled in Cura?
When thinking of the most simple case, we have an STL for example. We read the file, know the location of the vertices and their relation to each other in a unitless (but we assume millimeter) space. Out of this info we are generating the mesh, which is used at the end to visualize the model inside the SceneView (preview widget in Cura - don't know the name for sure, but think I saw this name somewhere..) ?

I actually never had anything to do with the things that are going on in this depth of Cura, so I guess, I'm overriding the model info at a wrong place and you ( @fieldOfView ) are probably right with your suggestion. Will try it out later 😉

nallath, 2017-10-02:
Yep. A node can be transformed, but those are almost always un-do-able. If you actually transform the mesh data, it's not. 
If you need some info on how scene graphs work, feel free to drop by.

thopiekar, 2017-10-20:
Tried @fieldOfView hint, but still the model remembers the rotation.
Even when creating a new node and passing the transformed matrix or vertices, it still remembers the rotation. Really hard to tell the software to forget the rotation, which has been made.
@nallath Next week I'm back in NL. Hope we can find a solution for this.

nallath, 2017-10-23:
You can use meshData.getTransformed(matrix) to actually transform the mesh data (instead of having a node with a transformation)

closed 2017-11-22 by thopiekar