Ultimaker/Cura #2528: Skirt distance / size effects active bed size.

Thisismydigitalself, 2017-09-30:
When increasing skirt distance, the effective bed size shrinks down. this seems unnecessary.
if the model was bigger, Cura wont slice it. 

Skirt @ 70mm
[image]

Skirt @ 20mm
[image]

fieldOfView, 2017-10-01:
> this seems unnecessary.

It is not. Cura "reserves" space to print the brim. Say you have a model 5 mm from the edge of your buildplate, and then you try to print a 10 mm brim. That would not fit on the buildplate.

Thisismydigitalself, 2017-10-01:
But this is clearly not the case. 
as you can see the model is few cm big. skirt is so far away from both the object and the edge of bed. 
Most of the time i print skirts close to the model (5mm or so), but sometimes i chose to print it further away to check that the bed is leveled. 

I believe something is not right here.

fieldOfView, 2017-10-01:
Take your top image. Now drag the model closer to the edge of the available space (ie: the not shaded part of the buildplate). You will see that when the model just about hits the edge of the available space, the skirt will just about hit the edge of the buildplate. It can't go further.

Thisismydigitalself, 2017-10-01:
Thanks for sticking in with me.
Please check both snapshots made for the sake of the argument.
if this is by design then i stand correct. it just seems there can be more room for the skirt to print is such cases.

45mm skirt (Sliced)
[image]

50mm skirt (Slicing failed)
[image]

fieldOfView, 2017-10-01:
Determining the outline of the first layer is surprisingly much harder than finding the "shadow" that Cura draws. That shadow is used with the above logic.

Cura uses a simplification (that shadow as opposed to the outline of the first layer), by design.

Thisismydigitalself, 2017-10-01:
Understood.

Ghostkeeper, 2017-10-11:
To find the size of the first layer we'd have to slice first (otherwise we won't know what the first layer is). That just takes too much time because you typically want to see where you can place the model before slicing.

ianpaschal, 2018-03-09:
Relates to: #3371 and #3365.

Question has been answered though so closing.

closed 2018-03-09 by ianpaschal