Ultimaker/Cura #2529: 2.7 Model features vanish depending on orientation

davebelleville, 2017-10-01:
Changing the orientation (or not) will cause entire areas not to be printed. For instance, if this model is oriented on its side, you see the internal features. Placed vertical, all internal features vanish.

[EZ_BULB.zip](https://github.com/Ultimaker/Cura/files/1346963/EZ_BULB.zip)

smartavionics, 2017-10-01:
Freecad says your model has problems...

[image]

fieldOfView, 2017-10-01:
See https://github.com/Ultimaker/Cura/issues/2520

Thisismydigitalself, 2017-10-01:
Same here. I was trying to check your stl just to find out it has problems.

davebelleville, 2017-10-01:
I corrected the normals and it is fine now.

smartavionics, 2017-10-01:
Great, please close the issue.

closed 2017-10-02 by nallath