Ultimaker/Cura #2531: [REQUEST] Retract While Combing

dmezh, 2017-10-01:
A wonderful thing to bring back is Retract While Combing.

Leaky filaments, like PETG, suffer from long travels with no retracts. Too many retracts, especially with bowden tubes, isn't optimal, hence combing.

Adding [Retract While Combing] and [Combing Length for Retract] is the best solution. The logic is 

(if combingtraveldistance > {x mm}; retract during the movement).

Ghostkeeper, 2017-10-11:
Bring back? I don't remember ever having that sort of thing.

Ghostkeeper, 2017-10-11:
We did have some research in this about half a year ago, and got some minor quality improvements when retracting while combing. The point was that the retract move causes a constant low pressure inside the nozzle chamber, which keeps the material inside the chamber rather than oozing out. This then reduces blips and underextrusion.

The research manifested itself as a post-processing script which was never working properly. However it's better to actually implement this in the engine.

dmezh, 2017-10-11:
Thanks, Ghostkeeper.

I was referring to Dim3nsioneer's Retract While Combing, a plugin for Cura 1X.xx.

https://github.com/Dim3nsioneer/Cura-Plugins/wiki/Retract-While-Combing-plugin

This is less of an issue with UM2 and up, but the non-actively-cooled UMO hotend makes this more pressing, since oozing is worse, especially with PETG and others. 

Generally, I don't see this manifesting as badly on my UM2 w/Olsson, but it could still be improved.

Lastly, I think it's be smart to use a shorter retract length for combing to reduce filament wear on prints with a lot of combing travel.

brnl, 2017-12-17:
I think I also need this feature. I got very visible travel lines with PETG on the Anycubic i3 Mega with 'REAL PETG' from 123-3D.nl.

ctzsnooze, 2017-12-19:
I'd be grateful for this feature also.  I print complex objects in TPU with thin solid walls joining complex out-pointing corners (with holes and infill).  During very long free air travels between corners, a small blob forms (despite retraction).  This will get stuck on the next perimeter and is difficult to remove.  If I enable combing, I get much less blobbing, and that's great.  But the long travels are now inside the walls.  And there are a lot of travels over the same area.  Since the printhead is not de-pressurised during those travels, and because the walls are thin, a thin smear of TPU gets spread over the top of the wall during each travel, and eventually combines to form small blobs and flaps inside the wall, which then interfere with the wall, making it look 'unsmooth'.  

There would be no problem with the printhead traveling in combing mode over infill, would mess it up a bit, but heck, it's infill.

However when printing transparent materials, and when the combing travels are over thin solid edges or walls, the result isn't so good at all.

I am not sure it will entirely solve the problem, but I think if the printhead was de-pressurised by a retraction during those combing travels, the walls would turn out better.

So if it was possible to provide the option for retraction during combing, that would be really useful.

brnl, 2017-12-19:
I found out that if you turn off Combing, Cura retracts for every travel. That might help.

dmezh, 2017-12-19:
This is a good tip, and combined with zhop, it can help greatly in reducing
this effect. However, zhop isn't ideal on ultimaker-style z setups, prints
with many retracts, and it slows down prints. Sometimes quality issues can
be found as a result of zhop.

The ideal thing would be to comb as usual but deptessurize the nozzle while
doing it, as to keep the interior clean and avoid an empty nozzle when
starting the next printing move.

brnl, 2017-12-19:
@dmezh No, I found out that z-hop is actually making it worse. Disabling z-hop reduces stringing a lot!

My settings:
Retraction distance: 3mm @ 25mm/s
Combing mode: Off
Z Hop When Retracting: Disabled
Optimize Wall Printing Order: Enabled
Enable Coasting: Enabled
Coasting Volume: Default (cubed nozzle width)

ctzsnooze, 2017-12-19:
brnl if I set combing to off, I get long travels that go out through one perimeter then some distance away back in through another perimeter.  That puts blobs when re-entering the perimeter, even with retraction, because TPU is very oozy.  Combing stops those blobs by keeping the ooze inside the model, but without retraction, causing weird micro-blobbing inside thin walls.  

At the moment I can either have nice clean thin walls or blobs. 

I'm definitely not overextruding, travels are as fast as can be, retraction distance seems correct.

I'm fairly sure it would be better if combing could be combined with retraction.

dmezh, 2017-12-19:
Agreed!

odospace, 2017-12-30:
I have a similar problem with PETG and Anycubic Ultrabase, the first layer (inkluding the skirt) is pulled off, because the string cools down and takes the first lines away. Combining with retraction for the first level(s) should eliminate this problem (combining on the second level works without retraction).

smartavionics, 2017-12-30:
Hi @odospace, for PETG I never use any combing to avoid dribble but I do use retraction with z-hop and I never have a problem with the first layer getting trashed. I am printing onto blue tape at 75deg and using a slow speed for the first layer. No fan for lower layers and if I really don't want the corners to warp, I use 10mm of skirt that stops 0.5mm from the outer wall to keep the heat in! The PETG sticks to the tape like the proverbial to a blanket.

odospace, 2017-12-31:
@smartavionics, thank you for that information. Have a happy new year.

smartavionics, 2017-12-31:
But still, I can't help wonder whether a sort of (optional) semi-retract could be beneficial during combing. Just a small amount of retract that is carried out concurrently with the initial part of the travel move. At the moment, retract/primes always happen with the nozzle stationary. What I am suggesting is that as soon as the travel move is started at the beginning of the combing sequence, at little bit of retraction could occur to reduce the pressure in the nozzle. At the end of the combing travel, the same amount of filament could be extruded so that the net movement of filament is zero. Would that make combing be usable for runny materials?

ctzsnooze, 2018-01-03:
I agree entirely with smartavionics, retraction during the initial part of the combing travels, and priming at during the terminal part, would greatly improve the value of combing for me. 

I find the lack of retraction to be a big issue with parts like in the attached movie.   These can have 5-8 travels across the centre of thin walls, and each time there is a risk of micro blobs dropping off into the wall before the perimeters are printed.  Eventually these coalesce and, particularly with flexibles like TPU, result in sort of internal blobbing, leading to non-smooth walls.

While one can fiddle with infill first or last, and internal walls before outers, I get best results as per the movie with TPU.  

If I disable combing, the thin walls come out glass-clear and smooth, but without retraction, combing makes them messy and internally blobby.

So yes please, option for retraction on combing would be really appreciated.

brnl, 2018-01-03:
Another way to minimize the inconveniences of all these travels, is to minimize the travels... This is not meant to be a troll, but I started trying Simplify3D but this software generates much more intelligent paths with so many less (longer) travels. This is absolutely the way Cura should go!

ctzsnooze, 2018-01-03:
[Cura_combing.mov.zip](https://github.com/Ultimaker/Cura/files/1599801/Cura_combing.mov.zip)

Yeah second that.

Check how many travels happen down these thin flexible walls during the combing...  :-)

smartavionics, 2018-01-03:
>Another way to minimize the inconveniences of all these travels, is to minimize the travels... This is not meant to be a troll, but I started trying Simplify3D but this software generates much more intelligent paths with so many less (longer) travels. This is absolutely the way Cura should go!

The next release of Cura has some optimization of the order in which infill (and skin) lines are printed. For some models it can make a big difference to the amount of travel done.

smartavionics, 2018-01-03:
Hi @ctzsnooze , I can't show that video on my Linux box can you make it available in another format or put it onto youtube?

smartavionics, 2018-01-03:
BTW - I shall implement the scheme I mention above for doing a negative extrude at the start of combing and associated extrude at the end of the combing move so we can see if it is beneficial.

brnl, 2018-01-03:
@smartavionics Good to hear!

ctzsnooze, 2018-01-04:
@smartavionics thank you!

Here's that video on youtube.  I added a detailed explanation.

https://www.youtube.com/watch?v=4snZRGQkU_Y

PS - as far as the question of what could be 'excessive' combing travels, could I make a suggestion?  In the above layer, imagine there were just the four corners with no walls.  Cura would then complete each corner individually, depositing all line types until done, then make one travel to the next corner.  But when the same corners are joined by solid walls made only from perimeters, Cura no longer sees each corner as an 'island', and travels from corner to corner to complete each hole, each infill, each solid infill etc.  What I could imagine might help would be if Cura determined the condition where the 'corners' were only joined by solid walls, and then somehow applied the 'island' algorithm, completing all lines that were contained inside a corner first, then combing to the next corner and completing it, and so on, then doing the final perimeters.  That would reduce the travels along walls to just one, with great results I'd expect, especially with combing retraction.

Thanks so much for looking into this!  Cura rocks!

smartavionics, 2018-01-04:
Thanks for the video. I can see exactly what you are talking about. One of the problems that is noticeable in the video is that it (Cura) is using the wrong travel distances between the islands to decide the order in which to print the walls.

For example, at the beginning it prints the infill in the NW island and then goes all the way round to the NE island for the next infill region when it should have gone to the SW island. This is because the routing uses the as-the-crow-flies distances rather than the combed distances when working out which area to move to. I have fixed this a long time ago but the Cura devs have not yet incorporated my fix (why not @Ghostkeeper? it's been 3 months now). See https://github.com/Ultimaker/CuraEngine/pull/617 for details.

Your suggestion about detecting "islands" separated only by thin walls is possibly a good idea. Given how Cura currently works, I am not sure how easy it is to implement but I will contemplate how it could be done.

ctzsnooze, 2018-01-04:
Ah yes I wondered why combing was 'taking the long way around'.  Your suggestion in the above link is obviously a very big improvement, in my case most likely cutting the number of combing travels by half.  This, with retraction while making those combing travels, will probably just about solve the problem.  While it would be even better if the 'corners' could be perceived as 'islands' when joined only by perimeters, I do accept that it may be a big challenge to make robust code that would achieve this.

Ghostkeeper, 2018-01-22:
That pull request wasn't merged because we merged this one instead: https://github.com/Ultimaker/CuraEngine/pull/652

Was that not what you intended?

ctzsnooze, 2018-03-12:
I'm still having this issue.  Z hop on with combing off adds a lot of time to the print and accumulates blobs on the top of each layer which then sort of add up and cause problems.  It also Z hops while building support which adds massively to print time.  It would be so much better if there was a simple option to retract while combing.  I would be *so* grateful.

ianpaschal, 2018-03-12:
Duplicated in by #3245. Would close that one but it seems to be more current. Closing this one as it's confusing to have two topics about the same issue.

closed 2018-03-12 by ianpaschal

ctzsnooze, 2018-06-19:
3.4 retraction while combing is a massive improvement!  Thank you!