Ultimaker/Cura #2532: Dialogs display "Yes" / "No" buttons instead of actions

thinkyhead, 2017-10-01:
Human interface guidelines for OS applications have long stressed that you should avoid using generic buttons like "Yes" and "No" in dialogs that ask for a user response. Instead, dialogs should use buttons that express the precise actions that will be performed.

Notice how this dialog shows "Cancel" and "Replace" buttons:

[image]

This makes it easier to recognize the dialog and allows the user to proceed without the need to read the dialog text first.

I'd submit the change as a PR, but I'm just getting familiar with the codebase and have to focus on another project today. If this lingers a while, I will see about submitting a PR.

nallath, 2017-10-02:
Pull requests are of course always welcome, but i've notified our UX team in the meantime.

Ghostkeeper, 2017-10-11:
We do a lot of "OK" and "Cancel", but we never actually have "Yes" and "No".

ianpaschal, 2018-03-09:
I've just scanned through with the search and I can't find any cases of this. Considering this is also a pretty old issue I think it's safe to consider it fixed.

closed 2018-03-09 by ianpaschal

LukeChen-Ultimaker, 2018-03-27:
button that states for "action" is a common way to make user clear what to do and to some extend shows the meaning of that action. @thinkyhead do you have any concrete examples (screenshots) that we can look into? I guess reviewing strings of all buttons is unavoidable to get a better and coherent user experience.

fieldOfView, 2018-03-27:
Note that it is not always as simple as replacing a string. Qt - the UI framework we use - has standard dialogs with Yes/No options. Creating a dialog with custom buttons is more work, and it is actually harder to make these look like system-native dialogs.

thinkyhead, 2018-03-28:
> Qt - the UI framework we use - has standard dialogs with Yes/No options. Creating a dialog with custom buttons is more work, and it is actually harder to make these look like system-native dialogs.

Indeed! I set out to do this myself, examined the code that uses standard "yes/no" dialogs and compared them to the more involved dialogs that would require more work. Not being well-versed in Qt, I didn't get too far and left it for someone with more Qt savvy.

I think it will be worth the effort in the long run to eschew the Qt-standard "yes/no" dialogs altogether and implement an improved [cancel] [action] dialog function that does its best to emulate the dialog layout appropriate to the platform it's running on.