Ultimaker/Cura #2533: Cura cant resume after Pause

SacrilAI, 2017-10-01:
i was needing to change the filament, so i paused, then when i try to resume, i dont work, im using Cura 2.7

nallath, 2017-10-02:
Resuming over USB? Network? On the printer? What printer?

SacrilAI, 2017-10-02:
USB, my printer is Anet A8

Ghostkeeper, 2017-10-12:
I just tried pause and resume on an Ultimaker 2+. Works fine.

To implement the pause, Cura just stops sending normal commands but periodically requests the temperature (every 2 seconds I think) to keep the connection alive.

sadigov, 2018-01-07:
I had the same issue with UItimaker 2+ today. It looks like the printer's CPU/memory got overwhelmed, took a long time to resume, and when it resumed, it started with the wrong Z-axis height, by about 5mm, bumped into the build and melted part of it. I have restarted the print from scratch.

ianpaschal, 2018-03-09:
Needs more info for reproducibility but did not receive it. Closing.

closed 2018-03-09 by ianpaschal