Ultimaker/Cura #2534: "Automatically drop models on the buildplate" not slicing where object is visualized

alonsorobots, 2017-10-02:
Objects are displayed in a different space than they are being sliced in.

To reproduce the issue, I'm:

* Turning off  'Automatically drop models on the buildplate' 
* Loading a mesh
* Translating the cube up in Z by 20mm 
* Change 'View Mode' to 'Layers'

and I get this result:

[image]

I want to print a floating object, because I was using the "support mesh" option that allows me to specify where I want support structure. 

Thanks!

-Alonso Martinez

alonsorobots, 2017-10-02:
Interesting! when I turn on supports then it does the right thing!

I guess in that situation where it wouldn't make sense that you would want to print an object that was floating without supports, I would want something like a warning saying that this was not allowed and that the object would be placed on the buildplate.

Thanks!

ChrisTerBeke, 2017-10-11:
Indeed it seems that when no support is enabled an no part of the model is touching the build plate, CuraEngine decided to put it on the build plate anyways.

Devs: CURA-4442

fieldOfView, 2017-10-11:
CuraEngine has always done that.

nallath, 2017-10-11:
That behavior was changed some time ago. It used to indeed always drop the model, but it should be a -bit- smarter now. The expected behavior is indeed that the support is printed (some people use it as thick rafts).

ChrisTerBeke, 2017-10-17:
@alonsorobots We have updated the behaviour when no supports are selected so that the model will float in the air (and you can use the support mesh).

closed 2017-10-17 by ChrisTerBeke

alonsorobots, 2017-10-18:
Thanks Chris and everyone else!!! =)