Ultimaker/Cura #2535: Not extruding in Cura 2.7

skygunner58203, 2017-10-02:
Since updating to 2.7, Extruder motor doesn't extrude filament.  The torque or power and eSteps sent via gCode appear to be flawed or bugged.  Issue doesn't exist in 2.6 and prints fine in that version.  Identical files also print fine in S3D.  Have checked the profile and settings are identical to previous version.  Have reproduced this with several prints.

smartavionics, 2017-10-02:
Hello, please provide a link to the model or gcode.

ChrisTerBeke, 2017-10-04:
We also need info on which printer you're using and what settings you have in Cura for that printer.

skygunner58203, 2017-10-04:
[image]
My settings.

skygunner58203, 2017-10-04:
Here is the link to what I was printing.  It printed fine last week.
https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=0ahUKEwjIuu_DjtfWAhUI3YMKHY2zA8MQFggoMAA&url=https%3A%2F%2Fwww.thingiverse.com%2Fthing%3A1545913&usg=AOvVaw1LmJcfnJaVVGE0pxm8rB3w

Ghostkeeper, 2017-10-15:
I reproduced your settings, as far as they are visible, and sliced that Calicat. It seems to extrude fine, as displayed by this snippet of g-code:

    ;TYPE:SKIRT
    G1 F1500 X38.891 Y30.099 E0.03938
    G1 X39.49 Y29.584 E0.0788
    G1 X40.138 Y29.133 E0.11818
    G1 X40.829 Y28.749 E0.15762

So I can't reproduce the problem.

Ghostkeeper, 2017-10-15:
Are you sure it's the g-code that Cura outputs that's at fault here? It's not the extruder grinding into the material and not getting any grip any more, or the temperature being too low or something? Nozzle clogged?
Perhaps you could show us a snippet of g-code that shows that it's not extruding.

If it's really the g-code at fault, maybe you can provide us with a project file (in the File menu, then Save Project). This will export all of your settings and the currently loaded model, which should make it easier to reproduce your problem.

skygunner58203, 2017-10-15:
I finally managed to fix the issue.  Turns out my Marlin firmware went haywire.  Everytime time I would slice a model and send it to the printer, my mintemp setting would change from 170 to 210.  Re-flashing my firmware has fixed that particular issue and it extrudes normally now.  Unfortunately, my nozzle is now clogged beyond fixing and I await my replacement hotend from Amazon.  Thank you for all the time you guys put into trying to help me with this.

closed 2017-10-16 by nallath