Ultimaker/Cura #2536: Suggestion: Layers view in mm

Thisismydigitalself, 2017-10-02:
Please add to layer number marker the total height in mm as well.

maukcc, 2017-10-04:
not total height, but height at that layer or best: "current height"/"total height"

Thisismydigitalself, 2017-10-04:
> not total height, but height at that layer or best: "current height"/"total height"

Yes, that was my intention, sorry.  "current/market height" thank you.

Ghostkeeper, 2018-02-21:
Our project manager removed this from our planning because it was not seen as important enough.

maukcc, 2018-02-21:
The difference between excellent and just good enough is in the details :)

ianpaschal, 2018-02-21:
The difference between excellent and feature creep is also in the details :)

darkdvd, 2019-03-19:
Not important but make a huge difference !

Actual view :
[image]

View in mm :
[image]

View in Layer & mm :
[image]

So please add the possibility to show the height in mm.

Guyohms, 2020-07-21:
I'm using adaptive layers feature and for tuning the adaptive layers parameters, this feature would help so much.

I guess it's simple to implement, because since Cura knows the thickness of each layers under the current one, it should be just a summation.

steblerjun, 2020-09-18:
++++++++1 please add the possibility to show the height in mm in preview mode!

logiclrd, 2020-10-01:
I don't even think this needs to be a configurable option. Just show both the layer number and the Z value it's printed at.

Some people (like myself) feel comfortable poking through the G code file to check things out, even insert commands here and there. Cura writes comments like `;Layer:0` to the G code, but these numbers correspond to the raw layer of the model, and the Preview pane has its own numbering scheme. It starts at 1, and if you use a raft, the raft's layers get counted first, etc. Currently, it's really difficult to figure out which layer in the G code corresponds to what is being shown in the preview. Please, please just make it say what the Z value is, so I can go to the G code and search for ` Z1.23` or what have you.

logiclrd, 2020-10-01:
@Ghostkeeper Can you take this back to the project manager and ask them to reconsider? As far as I can see it's a tiny feature with no logic required, and it would make many people's lives so much easier.

konskarm, 2020-10-07:
Hi @logiclrd. This is already in our backlog, but there are other priorities at the moment. Devs, see CURA-6535.

logiclrd, 2020-10-07:
Cool beans, thanks for the update ;-)

Thisismydigitalself, 2020-10-18:
3 years since i started this thread. 
I guess it's better late than never.
Looking forward.

coskier, 2021-02-02:
Although who knows how long the backlog will take to clear...

kpishere, 2021-07-11:
I'm also looking for this feature.  You see, the printer shows its current working height in mm.  You want to reference the cura preview in the same units.  Esp. for adaptive layer heights, the math isn't just an annoyance.

Ghostkeeper, 2021-07-19:
Our project manager has deferred this issue now, since they don't expect to be working on this any time soon. We'll leave this ticket open on Github since it is a real issue (and one that we hear more often), but mark it as deferred to make it clear that it's not expected to be implemented any time soon.

logiclrd, 2021-07-24:
Does deferred mean that it's not in the backlog any more? This continues to be a regular issue. I need this feature on many projects. I'm not the only one, looking at the comments on this issue. Maybe you could open source Cura, and then I could write the feature myself. :-P

logiclrd, 2021-07-24:
Maybe I can figure out how to do it myself...

logiclrd, 2021-07-25:
So, I've been poking at it for an hour or two, and I'm not 100% confide I'm understanding where exactly the label is defined and how it gets its value. Is it the `SimulationSliderLabel` inside the upper/lower handle `Rectangle`? I see it getting its value through what looks like data binding -- the `LayerSlider.qml` file says simply `value: sliderRoot.upperValue`.

So, to implement this feature, `sliderRoot` would need to be given a new bindable property that provided the desired text including the actual physical height, and the implementation of that property would need to include a look-up into a list of Z values by layer number, and then that data would need to be piped in from a place that knows it.

Ghostkeeper, 2021-08-03:
> Does deferred mean that it's not in the backlog any more? This continues to be a regular issue. I need this feature on many projects. I'm not the only one, looking at the comments on this issue.

Practically, yes. It means we acknowledge it to be an issue but doesn't get enough priority for it to be planned in the priority list. In practice, most deferred bugs eventually get fixed in a refactor left or right. Most deferred feature requests are still open probably, but some get implemented later if priorities change. It wouldn't surprise me if this eventually makes it in through a redesign of the layer view slider.

> So, I've been poking at it for an hour or two, and I'm not 100% confident I'm understanding where exactly the label is defined and how it gets its value. Is it the `SimulationSliderLabel` inside the upper/lower handle `Rectangle`? I see it getting its value through what looks like data binding -- the `LayerSlider.qml` file says simply `value: sliderRoot.upperValue`.

Exactly. `sliderRoot.upperValue` is defined at the top of the file, where it's set to the maximum value (the total number of layers) by default. The state of the layer view is here tracked in QML which is bad practice according to our guidelines but not the end of the world either.

> So, to implement this feature, `sliderRoot` would need to be given a new bindable property that provided the desired text including the actual physical height, and the implementation of that [...]

I would rather implement a new `pyqtSlot` function in `UM.SimulationView` that takes a layer number and returns the Z coordinate of that layer, or perhaps a Z range for spiralize and wireframe modes. The label can then just display the result of `UM.Simulation.getLayerZ(sliderRoot.maximumValue)` and it'll update automatically when the `maximumValue` changes.

aandea, 2022-12-09:
I use a Creality printer. When I switch the layer from 0.2 mm to 0.12 mm, the calculations become a burden to track progress. I would love to see this feature implemented as well !

maruhe, 2023-01-27:
+++1
with adaptive layers i becomes more important also
users decide what is important for them! not a project manager..

lmcjerl, 2024-02-01:
++1 This should be a very easy display update!