Ultimaker/Cura #2537: Cura don't start on Mac OSX

xtensia, 2017-10-02:
Hi all,
I'am in trouble to start Cura on any of my differents Mac ! I have a couple of Mac and each of us can't start Cura ! (Macbook A1181 & MacMini A1176)
I have tried to start manualy Cura from his App folder and i have seen a strange error message 
Is it normal that we can see under the diffrent error lines some of that refered to a file that is presumed to be in /Users/ultimaker/build/2.7/buil.....etc because it's an path that do not exist in other machine outside of ultimaker developpers one !?

Thanks for your answer.

Regards
Frederic Boutinon

_/Applications/Cura.app/Contents/MacOS/cura_ ; exit;
macmini:~ fredericboutinon$ /Applications/Cura.app/Contents/MacOS/cura ; exit;
Error in sys.excepthook:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 665, in exec_module
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 10, in <module>
  File "ExtensionLoader_PyQt5_QtCore.py", line 22, in <module>
  File "ExtensionLoader_PyQt5_QtCore.py", line 14, in __bootstrap__
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/imp.py", line 342, in load_dynamic
ImportError: dlopen(/Applications/Cura.app/Contents/MacOS/lib/python3.5/PyQt5.QtCore.so, 2): Library not loaded: /usr/lib/libc++.1.dylib
  Referenced from: /Applications/Cura.app/Contents/MacOS/lib/python3.5/PyQt5.QtCore.so
  Reason: image not found

Original exception was:
Traceback (most recent call last):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12, in <module>
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21, in <module>
  File "/Users/ultimaker/build/2.7/build/inst/bin/cura_app.py", line 53, in <module>
ImportError: dlopen(/Applications/Cura.app/Contents/MacOS/lib/python3.5/Arcus.so, 2): Library not loaded: /usr/lib/libc++.1.dylib
  Referenced from: /Applications/Cura.app/Contents/MacOS/lib/python3.5/Arcus.so
  Reason: image not found_

ChrisTerBeke, 2017-10-04:
Seems like a shared library could not be found on your system. Which version of MacOS are you running? libc++ missing is a known issue for older versions of the Mac operating system.

xtensia, 2017-10-04:
Hi Chris,
Thanks for your help,

My Macmini run on 10.6 currently and i cannot find libc++ file on the /usr/lib folder ! So i try to install with macport commande "sudo port install libcxx" and that made lots of error message relative of dependency and freeze finaly !  Thanks for this i will search to install this needed lib and i thinks that work well after !

My Macbook run on 10.8.5 and don't start the newest Cura 2.7 (don't even bonce on taskbar), so i install an old 2.3.1 that accept to start and communicate with printer ! But when i try to load a STL file from open file windows ...select a file, click on open and nothing happen. If i drag from finder to the Cura windows that work and my STL object screened but never done the slicing process run constantly !  Cura's lagging on all menu but accept to quit after a moment.

Finally i can only use a old Cura 2.3.1 on one old production PC only in 32bits and that work fine !  

Did you now a way to remove the necessity of approving access from PC onto the printer because my UM3 change frequently is IP address (due to the DHCP) and never can automatically reconnect unless i make manually ! I do not found where i can fix IP Address to prevent constantly changing ?

Regards

Frederic Boutinon

Frédéric Boutinon
Gérant 
frederic.boutinon@xtensia.fr
90 rue Origet
37000 TOURS
Tél : + 33 953 500 815
Fax : +33 972 23 55 11
Port: +33 6 08 03 06 88

Ghostkeeper, 2017-10-15:
The libc++ needs to be exactly the same version as it was compiled with, or you can expect some bad errors.

The authentication should only be lost if you delete the printer. The printer is identified by its MAC address.

sbgraphic, 2018-02-15:
Hello,

Is there some progess about this osx "issue" with Cura ? 
I'm trying to launch Cura on osx 10.8.5 and can't event load/open version 2.5 to 3.2, the system log show just this line:

15.02.18 18:23:06.740 com.apple.launchd.peruser.501[258]: ([0x0-0x1b31b3].nl.ultimaker.cura[2381]) Exited with code: 1

Only previous version frome 15.6.3 are starting properly

Best,
Eric

Ghostkeeper, 2018-03-08:
Our deployment target is version 10.7, so I'd expect 10.8 to work. We've tested this on various MacOS versions, but Cura seems to start fine on our computers. I don't think we have a MacOS 10.8.5 at work though.

ianpaschal, 2018-03-09:
@sbgraphic Not sure when you last updated by if you're running version 10.8.5 that would be between 2.5 and 4.5 years ago. We try to support older versions as much as possible but everyone in our office running macOS is using 10.13 (High Sierra). 10.8 is so old in fact that while I have it purchased on the Mac App Store, I can't download it to even test on a virtual machine.

This means that we will not be be able work on this issue (even with a diagnosis we won't have any quick way to see if our coding changes make the fix).

My personal advice would be that you should immediately upgrade your system. Mac updates are very streamlined and smooth, as well as free of charge. Considering there are some extremely big security vulnerabilities in the last 2-4 years which these updates would address, I think it's good advice. Of course, upgrading immediately to the newest version brings its own risks but you should be safe upgrading to at least the well tested and supported Sierra (10.12) and I suspect it will solve numerous problems including being able to run newer versions of Cura produced and tested when that version was current.

diegopradogesto, 2019-09-10:
Closing this ticket because it's very old and without any updates. Feel free to reopen if you think we can still do something.

closed 2019-09-10 by diegopradogesto

sbgraphic, 2019-09-17:
Hi guys, thanks for your support and sorry for my late response, i've finally updated (only) to 10.11 on this old MacPro 3,1 yet _(I had many reasons to stay on this very stable 10.8.5 osx for old software/hardware compatibilities)_, and get a new iMacPro with Mojave, hopefully I can keep up in this "osx upgrading madness" for some time, best to all. This thread can be closed (again).

thanarious, 2020-04-02:
Was pleasantly surprised when I read that even the latest (4.5) version of Cura only needs OS X 10.7 on Mac, since I have a MacBook 4,1 running 10.7.5 at home. However, installing that version on this computer doesn't seem to lead anywhere, Cura just crashes as soon as it launches. Is this a known issue? Anyone knows of the latest version that supports such an old OS?

thanarious, 2020-04-03:
version 2.3 seems to be working, but crashes randomly.

Ghostkeeper, 2020-04-06:
That comment from me before is outdated. It's currently limited by the Qt version we're using: https://doc.qt.io/archives/qt-5.10/supported-platforms.html

We updated from Qt 5.8 to 5.10 for release 3.5. So most likely the latest version that you're able to use is 3.4.

thanarious, 2020-04-07:
Indeed, since you're using Qt, you should at least follow Qt's system requirements on your project as well, just doesn't make sense otherwise!