Ultimaker/Cura #2539: 3mf error

651851996, 2017-10-02:
cura read 3mf file without color information
but cura own save 3mf no problem。However, this other software will not be used

651851996, 2017-10-02:
[image]
[image]
[image]

651851996, 2017-10-02:
version is 2.7

ChrisTerBeke, 2017-10-04:
The color of the model represents the color of the material that is in the extruder that the model will be printed with. In this case, the 3 donuts are all configured to be printed with extruder 1, thus showing yellow in the 3D viewer.

651851996, 2017-10-08:
我的意思是
1.用cura打开别的软件的彩色3mf会出现变成整体或者相互分开（这会失去装配位置）的现象，不能很好地支持多材料打印
2.用cura保存的多材质3mf（给手动调节位置和材质保存）可以达到理想的效果，不过别的软件打不开

ChrisTerBeke, 2017-10-09:
Unfortunately I don't speak Chinese, could you respond in English please?

651851996, 2017-10-09:
you need google =.=b
I mean
1. With cura open other software color 3mf will appear as a whole or separate from each other (which will lose the assembly position) phenomenon, can not well support multi-material printing
2. Save the material with cura 3mf (to manually adjust the location and material preservation) can achieve the desired results, but other software can not open

ChrisTerBeke, 2017-10-09:
Haha, don't really wanna spend time doing lousy translations in Google :P

On topic: we are thinking about storing Cura specific information in project files, but we're not quite sure what the best solution would be. I'll have a look at how 3D Builder exports the color/material info and see if we can use that somehow.

Ghostkeeper, 2017-10-17:
What is precisely the desired behaviour here? That we find an extruder for the object that has a colour loaded that matches approximately with the object colour?

fieldOfView, 2017-10-17:
Perhaps if a 3mf file contains multiple objects with multiple materials, but no extruder information, a dialog could pop up to map materials to extruders? The dialog could list all materials found in the 3mf files, and a combobox with the extruders for the current configuration for each of these materials.

651851996, 2017-10-17:
Ideally, the multi-color 3mf file is loaded and split according to the different colors and combined

closed 2018-11-22 by Appesteijn