Ultimaker/Cura #2545: Suggestion: Measurements in Cura

Thisismydigitalself, 2017-10-02:
I bet many like me would like to be able to measure stls before printing.
As a Fusion360 designer i know it is very hard to measure stl objects so what I'm suggesting is a way to control the resolution of Cura Printbed's checkerboard pattern so we can use that to measure distances by rotating and dropping through the Bed floor and be able to estimate the sizing of the object.
If you'll tell me Cura 3.0 will have a way to roughly measure parts - that would be great.

fieldOfView, 2017-10-02:
Cura does show the size of the whole print in the lower right corner of the viewport. In Cura 3.0 the buildplate will have a light 1mm grid in addition to a 10mm grid (which it currently also has).

Thisismydigitalself, 2017-10-02:
Both Scale and lower right corner do not show the dimensions of the dissected model through the printbed. we need to see the ever changing dimensions when we scrub through the layer in layer's view or for the dimensions to be reflected in the lower right corner.

I hope my idea is now better explained.

nickthetait, 2017-10-02:
So in layer preview, you would like to see the x and y measurements just for that layer?

Thisismydigitalself, 2017-10-02:
Correct. 
When we resize models to fit specific usage.
Here's an example - Here we see the overall X/Y dimensions but the real important dimensions is the section currently sliced

[image]

nickthetait, 2017-10-03:
(I don't work for Ultimaker but) I would guess that this feature would be useful to a broad range of Cura users. Thanks for suggesting it @Thisismydigitalself !

DDDirk, 2017-10-04:
A more general approach could be similar to what I use often in Blender; choose orhtogonal view (Num5) and then one of the 6 basic views (eg Top (Num7) or Front (Num1). A grid is then visible that shows the dimensions. To see just one layer, one could Boolean Intersect a 'thin cube' (a plane with three dimensions) and move that up and down or sideways.

Ghostkeeper, 2017-10-12:
An orthogonal view is fairly easily implemented and makes it easy to measure any part against the build plate down below.

Ghostkeeper, 2017-10-12:
Well it works, but the funny thing is that you can't zoom, since the thing about orthographic projections is that it doesn't become larger when it comes closer.
[image]

See also the branches `feature_orthogonal_camera` on both Cura and Uranium.

Thisismydigitalself, 2017-10-13:
Why guess the size of a model if you can extract it from the model itself?
It's not about estimating the model sizing. we need accurate sizing. that's the main reason i have created this issue  and suggested this feature in the first place.

fieldOfView, 2017-10-13:
> since the thing about orthographic projections is that it doesn't become larger when it comes closer.

Yes, I ran into that when I tried doing ortographic views. Zooming would have to be implemented as "true" zooming instead of dollying..

DDDirk, 2017-10-13:
In Blender zoom always works, in orthogonal or perspective view (toggle Num5) or in dolly mode (Shift + +/-).

Anyway, if the idea is to do exact measurements a virtual caliper might be an idea. For example, use layer view and then add two parallel lines that can be rotated and then move towards the edges of the layer, with a number in between showing the distance (possibly with arrows, just like in technical drawings, to make the function clearer). This seems to me to be more logical and intuitive than rotating the model and dropping it through the build plate.
And maybe this could be expanded to other orientations, so not limited to horizontal 'layers'.

Thisismydigitalself, 2017-10-17:
We need to be careful not to over think this. over thinking tends to complicate things.
All ideas are great but we need basic and accurate measurements now. not tomorrow.
i lost count how many times i've printed a portion of a model just to measure it before printing the entire thing.

fieldOfView, 2017-10-17:
> i lost count how many times i've printed a portion of a model just to measure it before printing the entire thing.

This sounds like you could have (should have?) done meassuring while modelling. The only times I print to meassure is when I need to have exact holes or an exact fit, which is influenced by die swell and other material properties, and these don't (and won't) show up in Cura.

Thisismydigitalself, 2017-10-17:
This sounds like you haven't read my first post.. :)
_"As a Fusion360 designer i know it is very hard to measure stl objects but what I'm suggesting for downloaded public stl's ... "..._

I was talking about STL' from like thingiverse, yeggi, etc.. publicly shared models.

DDDirk, 2017-10-17:
And a slicer seems like the ideal tool for this (even if it's not for slicing, just for measuring) since it already does most of the calculations that are needed.
The caliper was just an idea for visual presentation.

I am not familiar with the inner workings of Cura, but this seems fairly straightforward once the model is sliced. Take a slice, find the lowest and highest x-value and you have a measurement. Similar for y. Basically, that is a 2D bounding box. For other angles, the slice (not the whole model) can be rotated internally (not in the GUI), same principle applied, and then 4 lines are drawn at that angle (and 90° - angle) through the four points found. This would effectively be two calipers at the same time, at a right angle.

I am aware that it is probably not as simple as I make it sound. It never is. :)
Also, this only applies to the convex shape.

Ghostkeeper, 2018-03-21:
Our project manager removed this from our planning. We won't get around to implementing this any time soon.

adiastra, 2018-07-30:
That sucks! a simple point to point measuring tool would be sooo useful.

fieldOfView, 2021-04-08:
There's a test-version of my measurement plugin available here:
https://community.ultimaker.com/topic/36573-measure-tool-plugin/
The plugin is currently a simple point to point measuring tool. Not sufficient for all measuring needs, but it beats counting rectangles.

fieldOfView, 2021-04-28:
The Measure Tool plugin is now available in the Marketplace