Ultimaker/Cura #2547: Cura 2.6.1 fails to slice

jpathuber, 2017-10-02:
I used the standard ppa (ppa:thopiekar/cura) to install Cura, which has been fine for months.  Now I'm getting a failure similar to the person using the latest build, but with a different error in cura.log:

DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] [ERROR] Trying to retrieve unregistered setting with no value given: 'xy_offset_layer_0'

This started today, all was well last Friday. Cura says it's slicing but then never gets to saving the file. Should I build it myself from github or what? BTW, I can pull in an stl file, muck about with it and save as .3mf no problem.

hogerson, 2017-10-03:
I have same problem. Was away some days, came home last Friday start up and not slice.
Works in windows on same computer

thopiekar, 2017-10-03:
Thanks @nallath , the reason is that the PPA is blocked by this: https://github.com/Ultimaker/Uranium/issues/286

thopiekar, 2017-10-03:
@jpathuber @hogerson Build of Uranium has passed now.. Cura 2.7.0 is on the way..

hogerson, 2017-10-03:
Thanks, After last update total crash

thopiekar, 2017-10-03:
@jpathuber Yeah, sorry for this! 😞

jpathuber, 2017-10-03:
No worries, I understand stuff happens. It's a great program and I'm looking forward to the fix!

gummybearoz, 2017-10-10:
2.7.0 fails to slice less i force infill to different % then back to what it was when cura open up last 
in last 3 days you have wrecked this program

ChrisTerBeke, 2017-10-10:
2.7.0 was released quite a while ago, so nothing in the last 3 days changed there...

jpathuber, 2017-10-10:
Okey-dokey let's just say AppImage is the way and close this out.....

maukcc, 2017-10-11:
Please re-open this until there is a solution.

jpathuber, 2017-10-11:
Re-opening at maukcc's reasonable request. Although, now it is a problem with 2.7.0 and not 2.6.1 so I'm not sure whether that's right. Also the symptoms are different - it crashes. Votes?

nallath, 2017-10-11:
I'd prefer it if people don't re-use issues. It makes it bloody hard for us to figure out why something is still open.

As for doing development; The best way to do that is to run it from source. If you only want to change profiles, you don't really need this. If you need help setting it up on source, send us a message. I'm sure we can work something out.

maukcc, 2017-10-12:
@nallath : thanks for the offer to help us setting it up from source. That would be wonderful.
We are both in the Netherlands? Maybe we should have a meeting sometime.

Ghostkeeper, 2017-10-20:
Try me over at r.dulek at ultimaker com (or any of the other e-mail addresses that I sign my commits off with).

closed 2018-02-27 by ianpaschal