Ultimaker/Cura #2548: time estimate

mypeopoly, 2017-10-02:
Hi Cura team,

We would like to learn what part of Cura should we look at if the print estimated time is off?  We are using Cura to slice for Moai, a SLA printer and the laser movement speed is about 4 times of the print speed set in Cura.  Obviously, the print time would be off and we like to adjust that. What is a good place to start looking? 

[image]

nallath, 2017-10-03:
I think that using the right print speeds & acceleration in Cura should do the trick.

mypeopoly, 2017-10-03:
@nallath 

thanks for the quick reply.  It won't because the firmware interprets the print speed different.  the actual laser spot speed movement is about 4 times of the print speed parameters.

nallath, 2017-10-03:
Then there is little you can do about this.

mypeopoly, 2017-10-03:
can I times the print speed by 4 in the code where it does the estimated time calculation?

nallath, 2017-10-03:
Yes, but it will be a pretty nasty hack to do it like that (as changing it on that spot will break the code for all other machines). I think it's better to not have a machine that interprets speed differently.

mypeopoly, 2017-10-03:
@nallath  we have a forked version just for Moai. Perhaps we will just mod that one for now and let the users know about the changes.   In the long run, we like to update the firmware back to interpret the value as it is.

nallath, 2017-10-03:
Ah, in that case; Have a look at the PrintInformation.py file; That's where the frontend receives / distributes the print times.

fieldOfView, 2017-10-03:
A "if moai then divideSpeed" (pseudocode, if you hadn't guessed) will make sure it does not break other printers too hard.

fieldOfView, 2017-10-03:
A clean(ish) way to do this without affecting other printers by the way would be to add a custom setting (eg "moai_speed_multiplier") to your definition file, and set the value (eg to "4"). Then in PrintInformation.py check for the existence of that setting and use its value. 

I do similar stuff for BlackBelt. 
Adding a custom setting: https://github.com/fieldOfView/Cura/blob/blackbelt/resources/definitions/blackbelt.def.json#L30
Using the custom setting: https://github.com/fieldOfView/Cura/blob/blackbelt/plugins/BlackBeltPlugin/BlackBeltPlugin.py#L108

mypeopoly, 2017-10-03:
thanks guys. will test and revert

closed 2018-03-02 by ChrisTerBeke