Ultimaker/Cura #2550: Fuzzy skin - outside only option

zazolabs, 2017-10-03:
Add option to fuzzy only the outside walls, like the brim outside only option.

Use case - when printing a two part box  top and bottom - they are open so all walls are considered outside and fuzzed on both sides. It would be better to have only the outside perimeter walls to be fuzzed.

Ghostkeeper, 2018-02-09:
Our project manager just rejected this ticket for being too much of an edge case. The use case is too slim to warrant a new option.

zazolabs, 2018-02-09:
Printing enclosures/boxes is and an edge case ?!?!? what is  your PM printing - egg scluptures ?

ianpaschal, 2018-02-09:
As I understand it, not boxes but fuzzy boxes.

zazolabs, 2018-02-09:
yes, not much fuzzy boxes because fuzzy is not very usable for boxes - try printing a box with a slot for a PCB with the fuzzy option - the slot is fuzzy can not fit the pcb, a box with a lid - the lid does not fit to a fuzzy base. fuzzy is nice it makes unique skin but it needs to be on the outside for an usable box. besides most of the code is already there.

ianpaschal, 2018-02-09:
> yes, not much fuzzy boxes because fuzzy is not very usable for boxes - try printing a box with a slot for a PCB with the fuzzy option - the slot is fuzzy can not fit the pcb, a box with a lid - the lid does not fit to a fuzzy base.

That's sort of our point. Making a nice outer surface texture actually is really nice for a sculpture as you put it and not so useful for trying to make interlocking parts. We seem to agree on this.

As for whether or not the code is already there; "Most of the code is already there" does not mean something is easy to implement or that it won't require lots testing and introduce new bugs. In this case applying different settings to different surfaces of a model is actually a very huge modification that needs to be done, even if most of the required parts are already there.

ianpaschal, 2018-02-09:
@Ghostkeeper I'm going to go ahead and close this if you've already spoken with @Appesteijn about it.

zazolabs, 2018-02-09:
Ok, let's see:

src/FffPolygonGenerator.cpp:
FffPolygonGenerator::processFuzzyWalls
....

    for (unsigned int layer_nr = 0; layer_nr < mesh.layers.size(); layer_nr++)
    {
        SliceLayer& layer = mesh.layers[layer_nr];
        for (SliceLayerPart& part : layer.parts)
        {
            Polygons results;
            Polygons& skin = (mesh.getSettingAsSurfaceMode("magic_mesh_surface_mode") == ESurfaceMode::SURFACE)? 
                        part.outline : part.insets[0];

.....

                  so the next line must be if (option enabled && skin is inside ) continue;

to be able to print with  correct dimensions and now where to print brim on outside - there must be a property indicating  if a skin is inside or outside. 

so, please, help me write that if statement.

ianpaschal, 2018-02-09:
> there must be a property indicating if a skin is inside or outside.

There is not. I think you don't realize how incredibly complicated it is to make a determination like that for any given model.

Also, even if it was a matter of adding a conditional in Cura Engine: how do you select what's inside and what's outside in Cura? What kind of interface logic has to be written? And where will the actual interface elements go and what will they displace? There is a LOT of code to be written for something like this.

I hope this helps to explain why we're not able to take this on right now.

zazolabs, 2018-02-09:
I'll repeat my self - the logic to determine  is exactly the same as brim on the outside only option that is already implemented.  It is probably something like  object bounding polygon /the most outer perimeter/ and a simple test if inside or outside that polygon where outside means on the most outer perimeter - og can check the generated polygon vertexes against that bounding box and skip adding if inside.  Fuzzy is generating more polygons and is a simple function that need to be changed - not much risk of bugs and side effects /and it's strange opensource project to be afraid of bugs/.

zazolabs, 2018-02-09:
yes it is there:

Polygon p = storage.getLayerOutlines(layer_nr, include_helper_parts,  external_only)
bool Polygons::inside(Point p, bool border_result) const

so it would be like this:

for (unsigned int layer_nr = 0; layer_nr < mesh.layers.size(); layer_nr++)
    {
Polygon *layer_bounding_outline  = storage.getLayerOutlines(layer_nr, false, true);
for (Point& p : poly) {
...  
     if (option && layer_bounding_outline->inside(p, false) continue;
}

Appesteijn, 2018-02-09:
Feel free to make a pull request, but we (the cura team) are not going to work on this in the near future (e.g. 6 months).

Ghostkeeper, 2018-03-05:
>     if (option enabled && skin is inside ) continue;
> so, please, help me write that if statement.

A per-mesh setting can be obtained via something like `mesh.getSettingBoolean("the_setting_name")`.

You're confusing our terminology a bit regarding skin vs. walls. We regard "skin" as the top/bottom sides of the print and "walls" as the left/right/front/back sides. So what you're looking for is inside walls. 

Finding inside walls is pretty easy if you have a `ConstPolygonRef` instance: If the `.area()` function returns a positive number, it is an outside wall. If it returns a negative number, it is an inside wall (a hole). Also, the 0th element of a `Polygons` instance is always the outermost perimeter, so always an outside wall.

dsvilko, 2018-07-21:
I came here hoping that this was possible but was disappointed to discover that not only it can't be done but there doesn't seen to be much interest is developing this feature. This is not an edge case, as I see it. Jitter, specially small amplitude, can have a nice look and very effectively hide a small imperfections in the prints. I am currently building a complicated enclosure for my project and would very much like a jittery outside wall without messing with the precise inside dimensions. 
I do understand that it could be tricky to define what is an 'outside' wall of an object but I propose a much simpler and more flexible solution. Add a possibility to modify the jitter parameters on overlap with another model (as you already can for a lot of other parameters). That way you would not only solve the problem of outside faces of boxes  but could also do cool things such as selectively masking out the eyes of an animal sculpture and jittering up the rest of the fur. Hopefully this would be much easier to add.

Ghostkeeper, 2018-08-31:
What would happen if the jitter causes a part of the wall to be moved outside of the volume that enables the jitter? Does it need to stop at the border?

betavr, 2019-01-19:
Please make the idea of @dsvilko happen. @Ghostkeeper stopping at the border is not neccessary imo, bc fuzzy skin makes the model bigger anyway.

swatti87, 2019-01-24:
Can we bring this back to light a bit?

Another idea how to implement this. How hard would it be to "select" certain walls instead of  automating the selection? 

This feature has insane amount of potential if refined just a little bit.

antoniobeta, 2019-05-17:
I love the idea. I am making a screw driver handle and it would be nice if only the exterior is fuzzy and no the interior where i have to fit the bit :(

pdzamore, 2019-05-24:
This is definitely not an edge case. Any container intended to House something will look good with fuzzy skin outside and be useless with fuzzy skin inside. I ended up here because I am making a filament sensor case. Fuzzy skin outside would make it look much more finished.

mbartlett21, 2019-05-31:
We can do this at the moment:
1. Create another mesh that will fill the area that you don't want fuzzy
2. set the Per Model Settings to "Modify setttings for overlap with other models"
3. Choose select settings and tick the Fuzzy Skin option.
4. Untick Fuzzy skin

Example:
[image]

Ghostkeeper, 2019-05-31:
That will create a border (of wall lines) within your model though. If you don't mind that, it is a good workaround.

zmgrant, 2019-06-25:
I also would like to see this feature implemented. My reasoning is that it is common for the settings needed for a strong part do not coincide with the settings needed for a pretty part.  For cases and enclosures with a need for strength (or printed with challenging materials) it would be nice to have a way to make the surface more pleasing while not compromising functionality.

I'll be the first to admit I have no clue how the code of the slicer is set up, but my instinct for an algorithm that can handle this would involve some mathematical morphology such as an iterative approach using opening and closing based on object shape to determine the objects outer most face, followed by a dilation that would contain the area to be affected by the fuzzy skin algorithm. This could be applied layer-wise with some special cases for the top/bottom or in 3 dimensions.

Ghostkeeper, 2019-06-27:
> I'll be the first to admit I have no clue how the code of the slicer is set up

The code for fuzzy skin is pretty contained. It's right here: https://github.com/Ultimaker/CuraEngine/blob/2f8560d9b9ec168bda3ed8191262fda90066ca0c/src/FffPolygonGenerator.cpp#L1055-L1118

> but my instinct for an algorithm that can handle this would involve some mathematical morphology such as an iterative approach using opening and closing based on object shape to determine the objects outer most face,

It's easier than that. Polygons in CuraEngine have a winding, so all we have to do to determine whether a shape is an inside shape is to see whether the surface area of the polygon is negative.

> followed by a dilation that would contain the area to be affected by the fuzzy skin algorithm

Fuzzy skin just adds new points along the border which get a certain offset along the normal vector. The distance of that offset can be limited without needing to do a dilation of the polygon edge.

> This could be applied layer-wise with some special cases for the top/bottom or in 3 dimensions.

Fuzzy skin currently only works in the X/Y plane. In the Z plane it would not print very well because you're going to produce islands for which you'd need to retract and which would need support on the bottom side.

sezeryalcin, 2019-07-08:
It's very important to have fuzzy skin applicable only to outer shell. I just had a print and outside is great but since inner walls also have fuzzy skin, out parts do not fit.

jdodgen, 2019-08-12:
I have been printing up some object that the fuzz would look nice but they have a hollow area and the fuzz in there makes this feature unusable.

connornishijima, 2019-10-28:
I'm also giving a YES vote for this feature, I've been trying all day to work around this issue.

mnewnham, 2019-12-06:
Big yes from me too. A response was that this is an edge case, but literally every plastic car trim part is printed textured on the outer facing part and untextured on the inside. Fuzzy printing at low fuzziness is great for making professional looking replacement interior pieces but completely messes up locator tabs and similar on the inside of the part

DKXAM, 2019-12-12:
This feature seems incomplete without an outside only version of it, I disagree on the edge case here

connornishijima, 2019-12-12:
Short of a nice "outside only" option, a neat alternative would be an option to "average" bumpy contours for inner walls and thus skin layers as well.

For example, if I manually model my own "outside only" fuzz with noisy displacement maps in Blender, I hardly save any print time due to there now being fuzz on not just the outside outer wall, but the inner and skin outline walls as well, wherever I manually added texture. This also creates quite an ugly outline on surface layers.

Would it be possible to add a "smooth inner walls" option first as a compromise to these requests?

smartavionics, 2019-12-13:
It's trivial to turn off fuzz for holes, is that what you want?

[image]

Of course, things would get more tricky when you have, say, a box that should have fuzzy outside walls and then the top few mm needs to be smooth walls so that it can fit inside the lid. As mentioned above, it would be very difficult for Cura to determine when an outside wall should be fuzzy or smooth but as you can set fuzzy walls on a per-mesh basis it would be straightforward for the user to use a modifier mesh to make some of the outer walls fuzzy and some not.

If this is useful, I can put the "no fuzz holes" option into my cura builds.

smartavionics, 2019-12-13:
Here's an example of what I mean by using a modifier mesh to turn off the fuzz on the top region of a box like thing...

[image]

sezeryalcin, 2019-12-13:
I think outer walls are already calculated/known in Cura. Having an option for it should not be a huge work. 
Modifier mesh could be difficult to apply to thin walls.

smartavionics, 2019-12-13:
> I think outer walls are already calculated/known in Cura. Having an option for it should not be a huge work.

Sorry, I don't understand what you are saying there. Can you please rephrase it?

>  Modifier mesh could be difficult to apply to thin walls.

The modifier mesh is applied to a slice of the whole model not just the walls.

sezeryalcin, 2019-12-13:
programming perspective, if you know clearly if wall is outer wall, it gets easy to add this feature.

smartavionics, 2019-12-13:
Have you looked at the images I posted above? The first image shows a model whose outside wall has been fuzzed but the hole's wall has not been fuzzed. That was achieved using a modified Cura that has a Fuzzy Skin Outside Only option.

[image]

All I want to know is whether this is what people want?

sezeryalcin, 2019-12-13:
Yes, this is what I was hoping for.

connornishijima, 2019-12-13:
That would be amazing!!!

smartavionics, 2019-12-14:
I have added the Fuzzy Skin Outside Only setting to my latest Cura release. My releases can be found at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. Please read the README.md file there. All feedback is welcome.

estcap2, 2019-12-24:
> Have you looked at the images I posted above? The first image shows a model whose outside wall has been fuzzed but the hole's wall has not been fuzzed. That was achieved using a modified Cura that [...]

Yes, absolutely. Would be great for vases and things with precise inners that should not be fuzzy.

smartavionics, 2019-12-24:
> Yes, absolutely. Would be great for vases and things with precise inners that should not be fuzzy.

Indeed but it won't work in spiralize mode which is often used for vases because when spiralizing there is only one line and so you can't make a single line have a smooth inner edge and a rough outer edge.

estcap2, 2019-12-24:
> Indeed but it won't work in spiralize mode which is often used for vases because when spiralizing there is only one line and so you can't make a single line have a smooth inner edge and a rough [...]

Oh yes I know... came to that realisation last week, after experimenting a bit. It didn't go too well. I plan on trying your feature, only outer walls fuzziness (without spiralizing) to get a "stone" approach with thin walls.

connornishijima, 2020-01-04:
@smartavionics , THANK YOU! I've been using this for a week now on my small farm, I (among other's I'm sure) sell a product that RELIES on fuzzy skin for the look and feel, and it has been amazing!

@Ghostkeeper Any chance you guys would reconsider and accept a pull request now that the work is done?

smartavionics, 2020-01-04:
PRs now in the queues.

swatti87, 2020-01-13:
Would it be too complicated to CHOOSE what walls to fuzzy?

smartavionics, 2020-01-13:
> Would it be too complicated to CHOOSE what walls to fuzzy?

Yes, that would be a lot of work to implement.

fieldOfView, 2020-01-13:
Using the "Per model settings" on an overlapping model, it should be possible to make only parts of a model fuzzy.

swatti87, 2020-01-13:
I've tried a lot with overlapping models with various levels of success

sezeryalcin, 2020-01-13:
With a mesh model, I cannot remove fuzzy skin if it's already set for the main object.
However I can "add" fuzzy skin for overlaps if main setting is set to no fuzzy skin.

sezeryalcin, 2020-01-20:
@smartavionics I tested your build on Linux and although it has issues with fonts and exceptional cpu load, it worked fine and printed my design with "Fuzzy Outer only"
I would be glad to see this feature on upcoming official Cura releases.

smartavionics, 2020-01-20:
Hell @sezeryalcin , thanks for the report. Could you please zip up a project file and attach to this issue so I can investigate the exceptional cpu load that you mention. Thanks.

sezeryalcin, 2020-01-20:
Extra CPU load occurs only during startup (compared to 4.4.1 Linux x64), not for slicing.
I see a font issue which may be related to CPU issue.
[image]

Ghostkeeper, 2020-01-31:
Thanks to that pull request, this feature is now merged into the mainline Cura version 4.5. It'll be included with the upcoming release. Thank you!

closed 2020-01-31 by Ghostkeeper

sezeryalcin, 2020-01-31:
Thank you all and @smartavionics

sezeryalcin, 2020-02-28:
This feature works on 4.5 beta. However they way it determines "outside" needs some improvement. When I have a model that has holes outside, fuzzy skin with outside only option still goes into inside and  does fuzzy along the hole. Here I am attaching a hole and this view is from INSIDE, but fuzzy skin continued inside between layers where hole is.

[image]

smartavionics, 2020-02-28:
Yes, the feature was implemented very simply. I suspect an implementation that could handle your example would be a lot more effort to create. Sorry, I'm too lazy to do that but perhaps someone else will be inspired to improve the feature.

Ghostkeeper, 2020-03-04:
That would also be ambiguous. It's hard to see from your picture, but let's say that you have a layer that looks something like this:
[image]
The hole on the bottom side there is your screw hole as pictured in your screenshot. Arguably, the circular shape can be considered to be on the "inside". But now let's take the following example:
[image]
Is the circle still part of the "inside"? Yeah, sorta. Let's take the following example:
[image]
Is the 3/4 circle still part of the "inside"? To some extent yes. If you'd ask humans where the inside of this model is, they'd probably point to the 3/4 circle. Let's take it a bit more extreme then:
[image]
Is the 1/4 circle still part of the "inside"? Well it's an inner curve. Except it's not a curve but a series of straight line segments that approximate a curve. But you can hardly say that this curve is on the "inside" of the model any more. And finally:
[image]
This piece still has one inner corner that originally belonged to the inner circle. But nobody in their right mind would call the top side of this shape the "inside" because of that single pixel offset there.

sezeryalcin, 2020-03-04:
Outer shell can be defined as "surface you see at a zenith point when you rotate the model 360 deg on both X and Y"
Yes, this creates a problem on figures 1,2 and 3 above where you would be seeing inside the circle where you may not wish to. So in this case, we can define another parameter like "Min Bypass Distance" This can help define a virtual outer shell where you don't need penetration to inner curves.

[image]

If you have 10mm holes on your design, when you set this to 5.1mm, fuzzy should not go further into cavities as it will get closer to a shell closer to 5.1mm. On the other hand, if there is a curvy opening on some other side of the model, it can still be assumed as outer.

This is just an idea, no claims to solve all edge cases related to this. Probably also need a point precision value regarding this as calculating this can be challenging for CPU.

comedyaddict, 2020-03-07:
> Here's an example of what I mean by using a modifier mesh to turn off the fuzz on the top region of a box like thing...
> 
> [image]

This is exactly what I'm looking for, I just can't make it work. First off, I'm a complete Cura noob, but am liking the fuzzy skin feature. I need it to be applied to the outside of the container (this works now) but prevent from being applied to male threaded container part.

[image]

I have your lastest Cura release and can get the outside only option to work fine, but have no idea how you add a modifier or change its type which I've seen mentioned in other posts elsewhere. It sometimes feels like I'm missing major parts of Cura when I watch other peoples videos lol But I digress... 

Basically, how do I add a modifier mesh as you've done in your picture? :)

Ghostkeeper, 2020-03-09:
> I have your lastest Cura release and can get the outside only option to work fine, but have no idea how you add a modifier or change its type which I've seen mentioned in other posts elsewhere. It [...]

You'd load in a cube into Cura and select it. Then in the tools on the left hand side, click on "Per Model Settings". Click on the option "Modify Settings for Overlaps" and uncheck the "Infill Only" option. Click on "Select Settings" to select the settings you want to change with this modifier mesh. Then in the 3D scene, drag and rescale your cube to overlap with the places where you want the setting adjusted (in your case, the entire threaded part).

Some people like to use the Support Blocker tool as well to place the cube. But this is essentially the same as just loading in your own cube. It doesn't have to be a cube either; you can design very specific shapes where the settings need to be adjusted.

comedyaddict, 2020-03-31:
Many thanks, that worked perfectly! I appreciate the quick response;
somehow I missed the email notification.

chambm, 2020-05-26:
Is it normal that the first layer doesn't respect the "Fuzzy skin outside only" option?
[image]

This seems like a really cool option btw to give a wheel better traction.

[12 more replies]