Ultimaker/Cura #2553: 2.7.0 Clicking "Print with OctoPrint" from Prepare does not change blue page indicator line

ScruffR, 2017-10-03:
When hitting the "Print with OctoPrint" button on the Prepare page, the application swithes over to the Print page (as expected) but the Prepare tab keeps the "active" indicator (blue page indicator)
See here (running on Window 10 Pro 64bit)
[image]

fieldOfView, 2017-10-03:
Ah yes, thanks. I found that too a while ago, but forgot to make an issue (or fix it). I don't think it is an OctoPrintPlugin problem, because AFAIK it does the same thing the UM3NetworkPrintingPlugin does (so that may be affected too).

ChrisTerBeke, 2017-10-11:
This is fixed in the 3.0 release, you can try the beta to see it in action. Details: https://github.com/Ultimaker/Cura/commit/eb39d857e9ba37b2fa53090791308aef2158687b.

closed 2017-10-11 by ChrisTerBeke