Ultimaker/Cura #2556: Q A Z W shortcuts

Thisismydigitalself, 2017-10-04:
It seems these works only when we click on these manipulation's icons on the left side panels.
if we click on bed, than select the model these Q A Z W do not work anymore. there is a need to highlight the model with mouse than click again on the manipulation's icons on the left side panels for these to work again.

Ghostkeeper, 2017-10-15:
This issue doesn't reproduce on my computer. I can click the bed (everything unselects), then select a model and immediately push Q, A, Z or W and the appropriate tools activate.

nallath, 2017-10-16:
Could you provide a bit more information about your system (operating system, graphics card, etc)?

Thisismydigitalself, 2017-10-16:
Following your reply I ran some tests and managed to reproduce this and so it looks like a bug. 
running cura 2.7 on win10/64/nvidia i750. stay tuned. will shortly let you know what invokes this behavior.

Thisismydigitalself, 2017-10-16:
- Start cura 2.7
- Load STL
- Touch built plate > model > Shortcuts = all is well.
- Touch built plate > model > Side panel manipulators > Shortcuts = not working anymore and need to restart cura to regain functionality.

Thisismydigitalself, 2017-10-16:
Just checked with Cura 3.0 - same.
looks like a bug in both 2.7 and 3.0

Drayson80, 2017-10-18:
in addition (sorry for adding it here) - is there a way to modify the choosen letters? just asking because on a german keyboard, z and y are changed... :-)

Ghostkeeper, 2017-10-29:
> Start cura 2.7 or 3.0 and load STL
>
>- Touch built plate > model > Shortcuts = all is well.
>- Touch built plate > model > Side panel manipulators > Touch built plate > model  = not working anymore and need to restart cura to regain functionality. 

It still doesn't reproduce on my end. Using Windows 8.1, Cura version 3.0.3 (so 64-bit). NVIDIA GeForce GTX 560, but that probably wouldn't matter for this sort of thing.

We're considering making those shortcuts configurable. You're not the first to request it: https://github.com/Ultimaker/Cura/issues/2523

Thisismydigitalself, 2017-11-29:
Same issue with 3.1.
Shortcuts wont work if you first click on print bed then select the object  - in this case shortcuts will not work. for the shortcuts to work one must first click on the manipulation icons on the left, then select the object(s) - only then shortcuts works for me.

Ghostkeeper, 2017-12-13:
Still not reproducing. To be clear, this is what I'm doing precisely:

* Start Cura.
* Load a model by going into File -> Open Recent -> Cube1cm.stl.
* Click the build plate.
* Click the cube.
* Press the A button.

Thisismydigitalself, 2017-12-14:
Please follow my instructions to the letter - just tried it with Cura 3.1Beta and the bug is still there:

Touch built plate > model > **Left Side panel manipulators** > Touch built plate > model = not working

Ghostkeeper, 2018-01-01:
Ah, yes sorry, I didn't read that properly.

* Start Cura.
* Load model by going into File -> Open Recent -> Cube1cm.stl.
* Click the build plate.
* Click the cube.
* Click on "Scale" button.
* Click the build plate (stuff deselects).
* Click the cube.
* Press the Q button.

Then it reproduces.

Ghostkeeper, 2018-01-01:
As a workaround, you can get it to work again by clicking any setting field and then the build plate. It seems to be related to focus. Something swallows those key presses.

Ghostkeeper, 2018-04-16:
Our project manager removed this from our planning. We won't be able to get around to fixing this soon.

closed 2023-06-24 by github-actions[bot]