Ultimaker/Cura #2558: Cura 2.7.0 Segfault in Ubuntu 16.04.3 LTS

noleks, 2017-10-05:
Fatal Python error: Segmentation fault

Thread 0x00007f26b7fff700 (most recent call first):
  File "/usr/lib/python3/dist-packages/UM/Backend/Backend.py", line 156 in _storeStderrToLogThread
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f26d4ff9700 (most recent call first):
  File "/usr/lib/python3/dist-packages/UM/Backend/Backend.py", line 148 in _storeOutputToLogThread
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f26d5ffb700 (most recent call first):
  File "/usr/lib/python3.5/threading.py", line 297 in wait
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1392 in wait
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1046 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f26d67fc700 (most recent call first):
  File "/usr/lib/python3.5/threading.py", line 297 in wait
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1392 in wait
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 915 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f26d6ffd700 (most recent call first):
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 829 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f26d7fff700 (most recent call first):
  File "/usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f26f8ff9700 (most recent call first):
  File "/usr/lib/python3.5/threading.py", line 293 in wait
  File "/usr/lib/python3.5/threading.py", line 424 in acquire
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f26f97fa700 (most recent call first):
  File "/usr/lib/python3.5/threading.py", line 293 in wait
  File "/usr/lib/python3.5/threading.py", line 424 in acquire
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f26f9ffb700 (most recent call first):
  File "/usr/lib/python3.5/threading.py", line 293 in wait
  File "/usr/lib/python3.5/threading.py", line 424 in acquire
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f26fa7fc700 (most recent call first):
  File "/usr/lib/python3.5/threading.py", line 293 in wait
  File "/usr/lib/python3.5/threading.py", line 424 in acquire
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f26faffd700 (most recent call first):
  File "/usr/lib/python3.5/threading.py", line 293 in wait
  File "/usr/lib/python3.5/threading.py", line 424 in acquire
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f26fb7fe700 (most recent call first):
  File "/usr/lib/python3.5/threading.py", line 293 in wait
  File "/usr/lib/python3.5/threading.py", line 424 in acquire
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f26fbfff700 (most recent call first):
  File "/usr/lib/python3.5/threading.py", line 293 in wait
  File "/usr/lib/python3.5/threading.py", line 424 in acquire
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f2700afd700 (most recent call first):
  File "/usr/lib/python3.5/threading.py", line 293 in wait
  File "/usr/lib/python3.5/threading.py", line 424 in acquire
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f270137e700 (most recent call first):
  File "/usr/lib/cura/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 58 in _updateThread
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Current thread 0x00007f2745605700 (most recent call first):
  File "/usr/lib/python3/dist-packages/UM/Qt/ListModel.py", line 63 in setItems
  File "/usr/lib/python3/dist-packages/cura/Settings/ExtrudersModel.py", line 224 in __updateExtruders
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 682 in run
  File "/usr/bin/cura", line 73 in <module>
Segmentation fault (core dumped)

mr2web, 2017-10-05:
I got simular issue on my ubuntu 16.04 LTS after upgrading from Cura 2.5.0 to 2.7.0. First it crashed on the changelog splash screen after the update. The I removed ~/.config/cura and ~/.local/shared/cura, after backing up the files naturally. Then I restarted Cura and now it launched as a fresh install asking you to ad a printer. When adding a custom FDM printer and are clicking the "Add printer" button Cura craches with a "Fatal Python error: Segmentation fault" error.

balinhansen, 2017-10-06:
> When adding a custom FDM printer and are clicking the "Add printer" button Cura craches with a "Fatal Python error: Segmentation fault" error.

Just ran into this myself, the easy work around: add a Printer that isn't a "Custom FDM Printer" and then modify all the settings.

What's with the fonts though? Should I just upgrade to 17.04 and get it over with ... or?

nallath, 2017-10-06:
Did you run via appimage?

FixerDave, 2017-10-07:
Same for me.  Everything working well and then along came the update to 2.7.0.  Crashdump on 
"Starting printer discovery. 
Fatal Python error: Segmentation fault..."

Only running one custom printer.  Haven't tried the "hack a non-custom printer" work-around yet.  Not keen on it as there are lots of custom profiles defined.

fieldOfView, 2017-10-07:
None of you mention, but I am guessing you all use the (unsupported) PPA? Please try the official AppImage instead.

The crash seems to be in the specific version of Qt that is available for your LTS Ubuntu. AFAIK, the AppImage comes with its own Qt.

ezavod, 2017-10-07:
Similar error on openSUSE Tumbleweed.

**System:**
 OS: openSUSE Tumbleweed 64bit
 Kernel: 4.13.4-1-default
 Desktop: KDE Plasma 5.10.5 (KFramework 5.38.0)
 Qt: 5.9.1
 Cura: 2.7.0

**Case 1:** Using OBS package https://build.opensuse.org/package/show/science/cura.
Launching cura I get a segmentation fault similar to the one in the opening thread: [cura_segv.txt](https://github.com/Ultimaker/Cura/files/1364521/cura_segv.txt). I can see the splash screen until "Loading Interface..." and then it dies. I also deleted my config files (as done by @mr2web) but sadly it still crashes.

**Case 2:** Then I tried to run the AppImage (as @nallath, @fieldOfView suggested) from https://ultimaker.com/en/products/cura-software and it also fails with segmentation fault: [cura_appimg_segv.txt](https://github.com/Ultimaker/Cura/files/1364525/cura_appimg_segv.txt). The splash is visible and it dies after "Loading Interface..." again.

Both were fresh installs.

OliverWieland, 2017-10-07:
Same problem here on Kubuntu 17.04
I tried the "non-custom-printer-trick", but same issue.
I have a nvidia card and tried nouveau as well a th nvidia drivers

FixerDave, 2017-10-07:
The appimage version worked for me and even pulled in my existing profile settings.  Yes, I was using the PPA.  My search on how to install Cura came up with that, and the recommended appimage option didn't come up anywhere near the top, at least that I noticed.  Thanks to fieldOfView for the note.

jpathuber, 2017-10-10:
Everything old is new again, I'll be switching to AppImage which old farts like me used to call "statically linked computer programs". Please don't bother explaining why they're "not the same".

thopiekar, 2017-10-12:
@jpathuber I didn't take a close look at the log, but this is what I get when using Cura 3.0.0.
However, I'm using my PPA here. Before resolving some breaking builds because of issues with the testing suite, Cura 2.7.99 from master was working fine.

```
thopiekar@home:~$ cura
UM.Resources.__initializeStoragePaths [342]: Initializing storage paths
UM.Resources.__initializeStoragePaths [353]: Config storage path is /home/thopiekar/.config/cura/3.0
UM.Resources.__initializeStoragePaths [361]: Data storage path is /home/thopiekar/.local/share/cura/3.0
UM.Resources.__initializeStoragePaths [373]: Cache storage path is /home/thopiekar/.cache/cura/3.0
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [103]: Trying OpenGL context 4.1...
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [113]: Yay, we got at least OpenGL 4.1 core: 4.1 Core profile
UM.Qt.QtApplication.__init__ [82]: Detected most suitable OpenGL context version: 4.1 Core profile
2017-10-12 10:36:27,244 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ConsoleLogger
2017-10-12 10:36:27,285 - INFO - CuraEngineBackend.CuraEngineBackend.__init__ [64]: Found CuraEngine at: /usr/bin/CuraEngine
2017-10-12 10:36:27,285 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CuraEngineBackend
2017-10-12 10:36:27,289 - WARNING - UM.PluginRegistry.loadPlugin [200]: Plugin ConsoleLogger was already loaded
2017-10-12 10:36:27,297 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SelectionTool
2017-10-12 10:36:27,306 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin RotateTool
2017-10-12 10:36:27,312 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin MirrorTool
2017-10-12 10:36:27,324 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CameraTool
2017-10-12 10:36:27,333 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin TranslateTool
2017-10-12 10:36:27,343 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ScaleTool
2017-10-12 10:36:27,348 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin FileLogger
2017-10-12 10:36:27,354 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OBJWriter
2017-10-12 10:36:27,360 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin STLWriter
2017-10-12 10:36:27,364 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OBJReader
2017-10-12 10:36:27,369 - WARNING - STLReader.STLReader.<module> [25]: Could not find numpy-stl, falling back to slower code.
2017-10-12 10:36:27,371 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin STLReader
2017-10-12 10:36:27,379 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LocalFileOutputDevice
2017-10-12 10:36:27,383 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SimpleView
2017-10-12 10:36:27,443 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin FirmwareUpdateChecker
2017-10-12 10:36:27,447 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin XRayView
2017-10-12 10:36:27,450 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CuraProfileReader
2017-10-12 10:36:27,463 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GodMode
2017-10-12 10:36:27,466 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ChangeLogPlugin
2017-10-12 10:36:27,484 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin X3DReader
2017-10-12 10:36:27,578 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UM3NetworkPrinting
2017-10-12 10:36:27,587 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade21to22
2017-10-12 10:36:27,591 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade25to26
2017-10-12 10:36:27,594 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade27to30
2017-10-12 10:36:27,598 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade22to24
2017-10-12 10:36:27,603 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade26to27
2017-10-12 10:36:27,616 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin RemovableDriveOutputDevice
2017-10-12 10:36:27,625 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SliceInfoPlugin
2017-10-12 10:36:27,630 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PerObjectSettingsTool                                                                          
2017-10-12 10:36:27,640 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin 3MFWriter                                                                                      
2017-10-12 10:36:27,649 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LayerView                                                                                      
2017-10-12 10:36:27,657 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin AutoSave                                                                                       
2017-10-12 10:36:27,665 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin MachineSettingsAction                                                                                                                                                                       
2017-10-12 10:36:27,690 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OctoPrintPlugin                                                                                                                                                                             
[... cut ...]
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1102 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Current thread 0x00007fa05fd3c740 (most recent call first):
  File "/usr/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 234 in initializeEngine
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 674 in run
  File "/usr/bin/cura", line 82 in <module>
Speicherzugriffsfehler (Speicherabzug geschrieben)
```

JEFF2A, 2017-10-12:
Same trouble for me, after and before python update 2.7 to 3.5, cura 2.7 & master crashed, first one at printer choise window, and the other just after the splash screen.
My Qt version is 4.8.7
Could it be an OpenGL trouble, due to a graphic card change ???

thopiekar, 2017-10-12:
@JEFF2A Have you already tried the AppImage? Same there?

JEFF2A, 2017-10-12:
going to try...

JEFF2A, 2017-10-12:
ok !! it seems working very better !!

thopiekar, 2017-10-12:
@JEFF2A Well, then we can only blame it on Ubuntu again, I guess.. Nothing in the logs is indicating something else.

mpsdskd, 2017-10-12:
I'm experiencing the same problem on Kubuntu 17.04.
PPA 2.6 worked, PPA 2.7 gives the mentioned error, Appimage 2.6.2 works as intended,
appimage 2.7 however does not show my current printer configuration, so it is of no use for me.

nallath, 2017-10-12:
If 2.7 does not show your printer configuration, the upgrader must have gone wrong. Could you share your logs?

OliverWieland, 2017-10-12:
I can confirm that the AppImage 2.7 works, but my settings were not upgraded. I copied the settings manually to the 2.7 folder, now it works.

mpsdskd, 2017-10-12:
I hope this is the log information you need:

`2017-10-11 21:57:20,379 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin FileLogger
2017-10-11 21:57:20,385 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin GCodeProfileReader
2017-10-11 21:57:20,398 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin GCodeReader
2017-10-11 21:57:20,404 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin GCodeWriter
2017-10-11 21:57:20,417 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ImageReader
2017-10-11 21:57:20,444 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin LayerView
2017-10-11 21:57:20,452 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin LegacyProfileReader
2017-10-11 21:57:20,462 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin LocalFileOutputDevice
2017-10-11 21:57:20,473 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin MachineSettingsAction
2017-10-11 21:57:20,660 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin OctoPrintPlugin
2017-10-11 21:57:20,692 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin PerObjectSettingsTool
2017-10-11 21:57:20,701 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin PluginBrowser
2017-10-11 21:57:20,713 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin PostProcessingPlugin
2017-10-11 21:57:20,731 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin RemovableDriveOutputDevice
2017-10-11 21:57:20,795 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SliceInfoPlugin
2017-10-11 21:57:20,804 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SolidView
2017-10-11 21:57:20,814 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin CameraTool
2017-10-11 21:57:20,824 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin MirrorTool
2017-10-11 21:57:20,837 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin RotateTool
2017-10-11 21:57:20,853 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ScaleTool
2017-10-11 21:57:20,861 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SelectionTool
2017-10-11 21:57:20,872 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin TranslateTool
2017-10-11 21:57:20,887 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin UltimakerMachineActions
2017-10-11 21:57:20,922 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin UM3NetworkPrinting
2017-10-11 21:57:20,932 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin UpdateChecker
2017-10-11 21:57:20,933 - INFO - UpdateChecker.UpdateCheckerJob.run [44]: Checking for new version of cura
2017-10-11 21:57:20,984 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin USBPrinting
2017-10-11 21:57:21,008 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin VersionUpgrade21to22
2017-10-11 21:57:21,017 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin VersionUpgrade22to24
2017-10-11 21:57:21,027 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin VersionUpgrade25to26
2017-10-11 21:57:21,036 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin VersionUpgrade26to27
2017-10-11 21:57:21,043 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SimpleView
2017-10-11 21:57:21,069 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin X3DReader
2017-10-11 21:57:21,085 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin XmlMaterialProfile
2017-10-11 21:57:21,094 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin XRayView
2017-10-11 21:57:21,097 - INFO - UM.Qt.QtApplication.__init__ [112]: Command line arguments: {'disable-textures': False, 'qmljsdebugger': None, 'file': [], 'external-backend': False, 'single_instance': False}
2017-10-11 21:57:21,105 - INFO - UM.VersionUpgradeManager.upgrade [112]: Looking for old configuration files to upgrade.
2017-10-11 21:57:21,469 - DEBUG - UM.Backend.Backend._logSocketState [174]: Socket state changed to Listening
2017-10-11 21:57:21,482 - INFO - UM.Backend.Backend.startEngine [75]: Started engine process: /tmp/.mount_G1OdhO/usr/bin/CuraEngine
2017-10-11 21:57:21,484 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] Calling engine with: ['/tmp/.mount_G1OdhO/usr/bin/CuraEngine', 'connect', '127.0.0.1:49674', '-j', '/tmp/.mount_G1OdhO/usr/bin/resources/definitions/fdmprinter.def.json', '']
2017-10-11 21:57:21,531 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] 
2017-10-11 21:57:21,534 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] Cura_SteamEngine version DEV
2017-10-11 21:57:21,537 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] Copyright (C) 2014 David Braam
2017-10-11 21:57:21,538 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] 
2017-10-11 21:57:21,540 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] This program is free software: you can redistribute it and/or modify
2017-10-11 21:57:21,542 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] it under the terms of the GNU Affero General Public License as published by
2017-10-11 21:57:21,544 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] the Free Software Foundation, either version 3 of the License, or
2017-10-11 21:57:21,545 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] (at your option) any later version.
2017-10-11 21:57:21,548 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] 
[... cut ...]
2017-10-11 22:00:46,911 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2017-10-11 22:01:16,124 - DEBUG - UM.Qt.QtApplication.windowClosed [267]: Shutting down cura
2017-10-11 22:01:16,137 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [244]: Attempting to kill the engine process
2017-10-11 22:01:16,138 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [250]: Killing engine process
2017-10-11 22:01:16,139 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [253]: Engine process is killed. Received return code -15
`

JEFF2A, 2017-10-13:
if it could help here is my log
[log_cura_jeff.txt](https://github.com/Ultimaker/Cura/files/1381783/log_cura_jeff.txt)

kyrios123, 2017-10-15:
@thopiekar are you sure it is Ubuntu that should be blamed ?
* Same issue on [openSUSE Tumbleweed](https://github.com/Ultimaker/Cura/issues/2558#issuecomment-334926608)
* On [Red Hat 27](https://bugzilla.redhat.com/show_bug.cgi?id=1494278)
* On [Solus](https://www.reddit.com/r/SolusProject/comments/76gvv5/cura_not_launching/)
None of these rely on PPA's but face similar issue

thopiekar, 2017-10-15:
Well, since the AppImage is working, I only the the difference that all the other distros are using their own dependencies. Ultimaker is using only mainline sources to build their AppImage.. Things like the Qt libraries are build and inside the AppImage already. Since most distros are patching a lot, I can only imagine of this reason. Additionally there was already something in the past, that Cura from my PPA was not able to run on Unity, but the AppImage was. Ok, there is no Unity anymore, but the issue back then was related to a patch made by Canonical.
Of course, I know to less about the current state of the patches Ubuntu and all the others are using, but when I see this behavior I come first to this conclusion. You can convince me of something else.. It is not difficult, if you argue good enough :wink:

kyrios123, 2017-10-15:
So it's most likely related to some Qt libraries versions ?
If it can help, it was build against Qt 5.9.1

thopiekar, 2017-10-15:
Well, this could be the reason including PyQt, which is on top.
Got here Qt 5.9.1 here, too, and PyQt 5.7.

What we could do, to see whether it is a upstream bug or a patch, is to build a AppImage with Qt 5.9.1 and PyQt 5.7 (just like the deps we are using on our distros).
https://github.com/Ultimaker/cura-build-environment/blob/master/projects/qt.cmake

kyrios123, 2017-10-15:
PyQt is a dependency of Uranium which is a dependency of Cura so it is automatically installed.
We currently use PyQt 5.8.2

ezavod, 2017-10-19:
I fixed my crashing AppImage by installing **libglvnd-devel**.
See https://ultimaker.com/en/community/39284-i-keep-getting-segfaults-when-running-cura for details.

zimmski, 2018-02-11:
I just installed cura on openSUSE Tumbleweed and it crashed too but installing "libglvnd-devel" as ezavod suggested solved the problem for me.

thopiekar, 2018-02-11:
@zimmski : For me it was the same. However, in Ubuntu 18.04 the issue is solved now.
From my side (PPA) the issue is solved.

tenten8401, 2018-02-13:
I'm running Cura on Solus Linux and the only fix was installing libglvnd-devel.

ikeydoherty, 2018-02-13:
Surely the issue stems from `cura_app.py` ?

```
#WORKAROUND: GITHUB-88 GITHUB-385 GITHUB-612
if Platform.isLinux(): # Needed for platform.linux_distribution, which is not available on Windows and OSX
    # For Ubuntu: https://bugs.launchpad.net/ubuntu/+source/python-qt4/+bug/941826
    if platform.linux_distribution()[0] in ("debian", "Ubuntu", "LinuxMint"): # TODO: Needs a "if X11_GFX == 'nvidia'" here. The workaround is only needed on Ubuntu+NVidia drivers. Other drivers are not affected, but fine with this fix.
        import ctypes
        from ctypes.util import find_library
        libGL = find_library("GL")
        ctypes.CDLL(libGL, ctypes.RTLD_GLOBAL)
```

ikeydoherty, 2018-02-13:
I lost way too much sanity even figuring out wth is going on, so instead I moved the `.so` links out of the `devel` subpackages on Solus: https://dev.solus-project.com/R3960:13ca2f42cf3e778bbdfa786761e3709cfa40edf9

Issue will fix itself there on Friday..

thopiekar, 2018-02-17:
@ikeydoherty All I know is that this section was needed in the past to get Cura working.
Because I had no imagination how far this hack is needed, I added the `if platform.linux_distribution()[0]` check, which has been removed in the meantime and therefore the hack is applied to add distros.
During the time, it might be that this section is superfluous and like I wanted to describe in the comment, this hack was only needed on systems, where the proprietary NVidia drivers have been used.
But good that you mentioned it because I can imagine it might be worth to try without this hack again.

ianpaschal, 2018-05-28:
Since this is old-ish and affects a relatively minuscule slice of users, _and sounds like it's resolved?_... I'm going to close it. If that's a mistake we can reopen it.

closed 2018-05-28 by ianpaschal