Ultimaker/Cura #2563: AppStream metadata directory should be changed to ${CMAKE_INSTALL_DATADIR}/metainfo

onitake, 2017-10-06:
According to https://wiki.debian.org/AppStream/Guidelines the AppStream metadata should be installed to `/usr/share/metainfo`, not `/usr/share/appdata`.

Please change the corresponding line in `CMakeLists.txt`:

```
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -51,7 +51,7 @@
     install(FILES ${CMAKE_BINARY_DIR}/cura.desktop
             DESTINATION ${CMAKE_INSTALL_DATADIR}/applications)
     install(FILES cura.appdata.xml
-            DESTINATION ${CMAKE_INSTALL_DATADIR}/appdata)
+            DESTINATION ${CMAKE_INSTALL_DATADIR}/metainfo)
     install(FILES cura.sharedmimeinfo
             DESTINATION ${CMAKE_INSTALL_DATADIR}/mime/packages/
             RENAME cura.xml )
```

awhiemstra, 2017-10-10:
The spec seems to agree with you: https://www.freedesktop.org/software/appstream/docs/chap-Quickstart.html. I made a quick PR to fix it.

LipuFei, 2017-10-10:
That PR has been merged.

PR #2574 (2017-10-10): Install AppStream metadata into share/metainfo
Rather than share/appstream. This is what is suggested by the specification.

Fixes #2563
commit f02cc66e (2017-10-10): Install AppStream metadata into share/metainfo

Rather than share/appstream. This is what is suggested by the specification.

Fixes #2563
closed 2017-10-10 by LipuFei