Ultimaker/Cura #2565: Cura 2.7 slicer failure

tsh-xx, 2017-10-07:
The attached STL slices incorrectly. Layer 75 (sliced at 0.2mm) has no gcode.
[reamer.zip](https://github.com/Ultimaker/Cura/files/1366517/reamer.zip)

Same problem is observed with 2.6

Print actually survived, but both octoprint and gcode.ws show a problem with that layer.

smartavionics, 2017-10-08:
> Print actually survived, but both octoprint and gcode.ws show a problem with that layer.

CraftWare and Cura itself (latest development code) do not show a problem with that layer and the gcode looks reasonable. If the gocde was actually missing the print would show an obvious line at that position, does it?

tsh-xx, 2017-10-08:
If i turn off 'sort layers by Z' in gcode.ws, everything looks fine.
The print had some marks (and did fail first time round) but might have been higher up on the print - so I think this is not a cura problem.

closed 2017-10-08 by tsh-xx