Ultimaker/Cura #2566: Cura 2.7.0 G-code between parts in wrong order.

norpchen, 2017-10-09:
Using "Print one at a time" mode -- the code between two parts looks like this: 

> ;TIME_ELAPSED:1043.351379
G1 F2100 E764.71404
M107
M190 S75                                                      ; wait for bed to reheat
G0 F15000 X116.67 Y172.905 Z27.429          ; lift head 5mm 
G0 X80.669 Y34.634
;LAYER_COUNT:85
;LAYER:0
G0 F3600 X66.376 Y31.2 Z0.25

...one problem...it waits for the bed to heat up BEFORE it lifts the head, leaving a nice melted blob on the finished part while it waits.  The M190 should occur AFTER the life by 5mm line.  

Also, can you expose the code added between parts somewhere for edit, like you do for start and end gcode?  I may not always want a forced M190 wait between parts.

Ghostkeeper, 2017-10-18:
Indeed, it's better to move the nozzle up first. I've changed it for you here: https://github.com/Ultimaker/CuraEngine/commit/f32d2e3972cf9a17f921367a81868c605b22176a

closed 2017-10-18 by Ghostkeeper

Ghostkeeper, 2017-10-28:
It's a bit difficult to expose, since there is a bit of logic in there. It won't wait if the build plate temperature during the first layer is the same as the build plate temperature for the rest of the print. We could expose the "move 5mm up" part but that won't help for your case. We could also put that logic in the setting as a formula but then you won't be able to implement something similar as a user using the GUI.