Ultimaker/Cura #2568: [feature request] Manual Supports

atlemo, 2017-10-09:
Being able to add single, manual supports for small areas that need support would be fantastic.
Similar to Simplify 3D.

I'm sorry if this is a duplicate, but I could not find anything in search.

Thisismydigitalself, 2017-10-09:
+1 and .. these days we can only chose either touching built-plate supports or everywhere but we cannot chose supports which are only inside the model. I miss this option daily.

nallath, 2017-10-10:
It's not the best way to do it (ux wise), but have a look at the anti-overhang meshes. They work in a similar way as the infill meshes.

ChrisTerBeke, 2017-10-10:
This feature requests pops up every now and then. It's not on our roadmap currently, as many people still prefer automatic supports (slice & forget), and pro users sometimes model their own supports when creating the original mesh.

The suggestion by @Thisismydigitalself about supports that only touch the model is another discussion and should be fairly easy to implement.

thesec, 2017-10-10:
How about a model touching auto-support with the option to remove parts of it.
Seems a fair compromise and close enough to a fully manual support 👍

DDDirk, 2017-10-10:
Slightly off topic, but Chris, doesn't modelling the support mess up the skin (because the support then becomes part of the skin)?

ChrisTerBeke, 2017-10-10:
I've seen people design specific tree structures to support their models (like DLP supports). Another example would be the small support structures in the Ultimaker robot print (between the hands and legs).

atlemo, 2017-10-13:
Making custom supports is not really an option in a lot of cases where you receive an STL from a client, and you're not able to spend hours creating support meshes, which in turn won't be read as support and get the benefits from support settings in Cura.

thesec, 2017-10-13:
Custom supports is off-topic, lets do not climb that tree any further :)

garv3, 2018-02-08:
Needs to be implemented. Not having this feature makes Cura kind of useless for many prints!

ChrisTerBeke, 2018-02-08:
How so? I literately never felt the need to use it.... Can you give some examples where it would help you out? Maybe there's another solution (read: simpler to implement) that will also get you there.

ChrisTerBeke, 2018-02-08:
Removing supports can already be done by setting the mesh type of for example a cube to 'remove supports' and placing that cube where you want to remove the supports. we're working on a plugin to make that workflow a bit easier (e.g. creating the cube automatically).

With regards to filament usage / placing supports, I personally still don't see the added value between fully automated supports (which could be improved) and manually designing your support in CAD software for specific use cases. Again, that's just my own opinion.

jackha, 2018-02-08:
You can actually add manual support. Load a model (i.e. a cube), then in custom mode select Mesh Type "Print as support"

Blaster1920, 2018-02-08:
You guys are totally crazy..  Just try Simplify 3D: manual support totally customizable in less than a sec.. Add a cube then copy it etc etc? You'd triple the slicer set-up time.

ChrisTerBeke, 2018-02-08:
It's fine to have an opinion about all of this, but calling us straight out crazy seems a bit overexcited...

Also Cura and Simplify are totally different slicers/code bases, so just because they currently have it means it's something we can implement in a respectable time frame.

thesec, 2018-02-11:
Will give the mesh type "remove support" a try. Getting that into an easy workflow sounds "almost there" to me.

On using CAD for support generation - in my opinion - that completely mixes up concerns: while I design in CAD and might test the model itself for production usage it is the slicer where the model is placed, added support to ( not part of the model ) and sliced for a optimum result.

Ghostkeeper, 2018-02-21:
Our project manager removed this from our planning, saying he doesn't think it's important enough. I don't know if he actually read the ticket before removing it from our planning though...

thesec, 2018-02-23:
Doesn't sound like a very agile team ... more like 80ies and waterfall? ;P

ChrisTerBeke, 2018-02-23:
Actually it is a very known practice in Agile to clean up the backlog with stories that no longer seem relevant. This is called backlog grooming. It also doesn't mean it can't re-appear if it becomes important again.

thesec, 2018-02-23:
Now we have Agile :))

superjamie, 2018-04-18:
This has been a request for a long time: https://github.com/Ultimaker/Cura/issues/386

superjamie, 2018-04-18:
Oh I agree with you! The recent Support Blocker feature in Cura 3.3 gets at least part of the way there.

I more meant to say this Issue is a duplicate of your original so this could be closed.

fieldOfView, 2018-04-18:
> regarding the manual modelling in CADs, it's not enough. Support structures have different densities, different ways to be constructed than the part itself, e.g. with (or without) a support interface, and the same support structures might be printed different ways depending on the printer

That can already be remedied by printing select models "as support". See the Mesh Type selection in Per Model Settings.

Ghostkeeper, 2018-05-03:
> I more meant to say this Issue is a duplicate of your original so this could be closed.

I agree. Let's close this one as duplicate of #386 and keep the conversation together with the rest a bit.

closed 2018-05-03 by Ghostkeeper