Ultimaker/Cura #2569: No "Move" Button Cura 2.7.0

cool-junk, 2017-10-10:
There is no "move" button on my Cura 2.7.0 
This is really driving me crazy. I have searched everywhere and can't find a solution for this. I even uninstalled and reinstalled to no avail..... Still no "Move" Button...... 
I have no way to move things on the build bed.
See image for reference.

[image]

smartavionics, 2017-10-10:
The move buttons doesn't become active until one or more objects are selected. That object on the build plate doesn't have a blue outline so it isn't selected?  What happens when you click on the object?

ChrisTerBeke, 2017-10-10:
Mm that button should definitely be there. Seems like the plugin that exposes it didn't properly load. Can you post some logs here? Information on how to find the logs can be found at https://github.com/Ultimaker/Cura#logging-issues.

ChrisTerBeke, 2017-10-10:
@smartavionics The button would indeed not be activated, but it should still be listed in the toolbar at all times.

smartavionics, 2017-10-10:
> @smartavionics The button would indeed not be activated, but it should still be listed in the toolbar at all times.

Where does it get listed? I can't find it other than on the LHS?

smartavionics, 2017-10-10:
Sorry, it's too early here, please ignore my above comments!

nallath, 2017-10-10:
Could you share logs? If the tool can't be loaded, it should be clear from the logs.

cool-junk, 2017-10-10:
Thanks for the replies fellas. I will get the logs and post them this evening when I get home from work.

cool-junk, 2017-10-11:
Here is my Log file. 

[cura.log](https://github.com/Ultimaker/Cura/files/1373851/cura.log)

cool-junk, 2017-10-11:
I noticed that the Logs go back way before I tried to reinstall that last time. I just uninstalled and manually cleaned out the app data etc... I am re-downloading it not and will post my results after this install. Fingers Crossed......

cool-junk, 2017-10-11:
Okay, my issue is now resolved. 
1st: I uninstalled Cura through "uninstall programs" on Windows. 
2nd: I did a search for "Cura" in windows explorer and manually deleted any associated files I could find. 
3rd: I downloaded and reinstalled Cura from the website.
All Fixed!

closed 2017-10-11 by cool-junk

ChrisTerBeke, 2017-10-11:
The logs contain the following line:

2017-08-06 12:45:57,433 - DEBUG - UM.PluginRegistry.loadPlugin [148]: Plugin TranslateTool was disabled

This means that the move tool plugin was disabled somehow, which is weird because it's a required plugin.

fieldOfView, 2017-10-11:
It is possible to (accidentally) disable required plugins by clicking their names. Fixed in https://github.com/Ultimaker/Uranium/pull/294

ChrisTerBeke, 2017-10-11:
Nice catch!