Ultimaker/Cura #2570: Skirt or Brim not going around object base completely

chuffpdx, 2017-10-10:
Load "Creepy_Hand" object at the following:
https://www.thingiverse.com/thing:1524972

Set brim or skirt and notice it cuts through middle of object and not around base completely. 

I was able to confirm this in v2.7 and v2.6.1

smartavionics, 2017-10-10:
The bottom of the model isn't flat, just lower it into the build plate a little like this:

[image]

Thisismydigitalself, 2017-10-10:
hmm.. was that a sneak peek teaser of Cura 3.0 built plate?..

fieldOfView, 2017-10-10:
It is no secret, there are more screenshots of the new UI to be found around here

smartavionics, 2017-10-10:
And you can see it for yourself if you install the beta 3.0 which has been announced today.

Thisismydigitalself, 2017-10-10:
> And you can see it for yourself if you install the beta 3.0 which has been announced today.
Wow, - I must have been asleep 👍 
I can only see the announcement. no file to download.

smartavionics, 2017-10-10:
Yes, you're asleep because the link is in the posting!

http://software.ultimaker.com/Cura_open_beta/

chuffpdx, 2017-10-10:
My mistake, thank you.  I rotated -1 degree and it skirted properly.

closed 2017-10-10 by chuffpdx