Ultimaker/Cura #2573: Please disable "call home" features by default

petterreinholdtsen, 2017-10-10:
In the configuration, there are two 'call home' features, the 'Check for updates on start' and 'Send (anonymously) print information'.  These features will submit among the obvious information also the users IP address and when Cura is used to Ultimaker, which is exposing private information on the Internet.

Please change the default to not 'call home'.  When starting Cura for the first time, a pop-up notify the users about these features, which is good.  Without it, I would not have discovered this right away.  Perhaps this pop-up can be changed to allow the settings to be changed using the pop-up, to make sure users have a quick and conscious choice to set or disable this before it send anything to Ultimaker and anyone else listening on the Internet.

ChrisTerBeke, 2017-10-10:
The popup is there to inform the user that anomalous statistics are collected. We do this to make Cura better for everyone. If you really don't want your data to be collected, disabling it once in the preferences window is a small step. This is a widely used method of handling privacy settings.

I'm not really sure I understand your comment about 'exposing private information on the Internet', as all traffic is encrypted over HTTPS.

nallath, 2017-10-10:
We could improve the message a bit; adding an action button so that the preference screen opens up would already improve this quite a bit.

petterreinholdtsen, 2017-10-10:
It would help a lot, as I as a new user was quite unsure how to find all 'call home' settings.

nallath, 2017-10-10:
But yeah, disable all of these by default isn't something i think is a good idea. The Update checker should always be on by default. The slicing data is completely anonymised; we rely on this data to get better profiles. You could argue if that should be on by default, but in that case, that discussion has already been held, so i doubt that will change.

petterreinholdtsen, 2017-10-10:
It make me sad to hear you insist of enabling these 'call home' features by default.  I guess others packaging Cura will reach a different conclusion regarding the default value for the 'call home' features, depending on their view on "privacy by default". :)

Anyway, by 'exposing private information on the Internet', I mean that calling home will create a TCP connection that exposes when Cura is used and IP address involved to everyone listening on the net, with NSA, GCHQ and FRA being the most well known examples.  Thanks to Snowden, it is confirmed that the former is able to break HTTPS connections, for exmaple by hijacking the TCP connection and reducing the level of encryption used, or by exploiting weaknesses in the HTTPS implementations in use.  If they can do it, so can others.  It will of course also make sure the unencrypted information is available for Ultimaker, including IP address, which is enough to do a rough location of the user and map the usage to other activity on the net.  The "anonymous" part is enforced on your end, and there is no way for a user to verify if you throw away the IP address or not, nor if it is combined with other sources collecting user information on the web like Google Analytics, or more relevant for ultimaker.com, myfonts.net and cloudfront.net. :)

closed 2017-10-10 by petterreinholdtsen

nallath, 2017-10-10:
True; But do note that Ultimaker is based in the Netherlands (and the server Cura connects to is as well). We have pretty strict laws regarding this. Companies can get fines of up to 4% of their revenue if they do not handle this kind of data correctly. 

No amount of encryption is truly ever safe, but care should be taken to look at your security compared to how dangerous it truly is the data falls into the wrong hands. I do feel strongly about privacy of our Users (as, in my experience, most developers do), but insane strong defence doesn't make a whole lot of sense if the data isn't that sensitive (This is for instance not health information. All it sends is an IP, user agent and some settings).  

Note that you also expose yourself by downloading Cura in the first place, as we (eg:ultimaker) must collect your IP in the first place in order to serve the download. 

When trying to defend against the online capacities of an entire nation, this is already a lost battle. No amount of security will protect you against billion dollar budgets.

Ghostkeeper, 2017-10-19:
With the GDPR we're going to have to change this to an opt-in, next May: http://www.eugdpr.org/

Ghostkeeper, 2017-10-19:
> No amount of security will protect you against billion dollar budgets.

Not sending the information would.

nallath, 2017-10-19:
No, as cura itself could still be targeted by a man in the middle attack when downloading.

The GDPR article 26 states;

> the principles of data protection should not apply to anonymous information, namely information which does not relate to an identified or identifiable natural person or to personal data rendered anonymous in such a manner that the data subject is not or no longer identifiable. This Regulation does not therefore concern the processing of such anonymous information, including for statistical or research purposes

As we've went to pretty great lengths to ensure that the slice data can not be tracked to a person in any way, nothing needs to change.

fieldOfView, 2017-10-19:
> As we've went to pretty great lengths to ensure that the slice data can not be tracked to a person in any way, nothing needs to change.

No matter how you slice it, Ultimaker still gets the IP addresses of the Cura instances. That does not mean that Ultimaker does anything with those IP addresses, but be have no way of knowing for sure that it doesn't, do we?

nallath, 2017-10-19:
This is true. But thats why the GDPR is now there. As far as I understood also appoints an agency that's checking for this. Note that most of the things in the GDPR were already in the Dutch Law.