Ultimaker/Cura #2577: prints 1st layer only 1/2 then full on 2nd layer

gummybearoz, 2017-10-10:
prints 1st layer only 1/2 then full on 2nd layer
[image]
[image]

ChrisTerBeke, 2017-10-10:
Seems like the bottom surface of your model is not entirely flat (or angled).

gummybearoz, 2017-10-10:
iv seen this comment on many other posts 
is this your go to cut past 
im not this silly 
iv done that test and cut a clean face then lowered it to 0-zero so this dis prooves the its not flat argument

DDDirk, 2017-10-10:
In Cura, did you do 'lay flat' (click the rotate-icon on the left, then the lay flat-icon)?
If that solves it, it may get printed at a (slight) angle, so it might be better to correct it in the model (make the surface completely horizontal).

ChrisTerBeke, 2017-10-10:
If you post the model here I can double check for you, but it's not the first time I've seen an issue with screenshots like these....

gummybearoz, 2017-10-10:
https://www.tinkercad.com/things/8BnQ6lkCTKD-9lay-flat-

ChrisTerBeke, 2017-10-10:
You have a link that doesn't force me to create an AutoDesk account?

gummybearoz, 2017-10-10:
In Cura, did you do 'lay flat' (click the rotate-icon on the left, then the lay flat-icon)?If that solves it, it may get printed at a (slight) angle, so it might be better to correct it in the model (make the surface completely horizontal).
--

yes this worked but iv never had to lay flat before in last year of printing with cura 
thank you for the tip ill make it part of my must do before saving.
as said above iv cut a new face then lowered to zero so my model is flat.

gummybearoz, 2017-10-10:
how to upload .stl didn't let me on making this post 
in zip?

fieldOfView, 2017-10-10:
I have checked the STL from TinkerCAD. Sorry, the block with the cylinder, triangle and square subtracted from it does _not_ have a flat bottom. The bottom two vertices on the front are .405 units (mm) higher than the other vertices.

gummybearoz, 2017-10-10:
[image]
wow im sorry for this mistake you make sence now its 1 am hear and uninstalled cura twice sorry for my rant 
apology from down under 
sorry

closed 2017-10-10 by gummybearoz