Ultimaker/Cura #2578: Cura - end of my rope

sbPhys, 2017-10-10:
I am writing this to beg for help. I want to be able to edit the Cura source code to add a few things and change the look. I really don't want to change anything that's there, just add some things to it. To that end, I have been trying all day to install Cura on Windows through github. It's driving me insane. 

I could list out all the things I've tried, but I feel that would be a bit of a waste of everybody's time. Instead I'm going to just ask for help, plain and simple.

Can anybody give me some step-by-step instructions on how to install everything I need to be able to run and edit the python code for Cura on Windows 10?

fieldOfView, 2017-10-10:
What do you want to change, _exactly_? A lot (though not all) of files are editable in the binary release. All the files in resources/themes (to change the looks), and in plugins (allmost everything is a plugin in Cura).

sbPhys, 2017-10-10:
Apart from the looks, which will need to be changed, the main thing I need to do is add a button which will a) open a browser to a specific webpage and b) upload the current model to that webpage through its api.

fieldOfView, 2017-10-10:
You can do both without running from source

sbPhys, 2017-10-10:
How? Where should I start?

sbPhys, 2017-10-10:
Ok, let me back up a bit. First question, when you say 'binary release' are you referring to the version I can install from https://ultimaker.com/en/products/cura-software?

sbPhys, 2017-10-10:
I'm sorry if I'm coming across like an idiot here. I like coding, but I'm still a novice in many areas.

fieldOfView, 2017-10-10:
Install the software. See what files are installed. Edit them. 
Themes are in resources\themes
For the button, you would likely write a plugin. Plugins are in the plugins folder.
See how current plugins do their thing. Adapt. Profit.

ChrisTerBeke, 2017-10-10:
We have some plugin examples available on the Ultimaker GitHub page as well: https://github.com/Ultimaker?utf8=✓&q=plugin&type=&language=

closed 2017-11-14 by ChrisTerBeke