Ultimaker/Cura #2579: Cura 3.0 Profiles

Thisismydigitalself, 2017-10-10:
Trying out Cura 3.0.
1. Exported my favorite profile from 2.7
2. Cleared all settings of  3.0
3. Imported 2.7 profile to 3.0.

Expected result: to see only my setting from 2.7. this is clearly not the case.
Either i get many more settings showing up which I don't want to see, or i get a clear list without any settings.

How do i import my 2.7 profiles with only my checked settings?

ChrisTerBeke, 2017-10-10:
What do you mean with cleared all settings of 3.0? Removing them from the UI or actually removing the user data folder?

Thisismydigitalself, 2017-10-11:
Removing from UI. 
uncheck all settings in 3.0 and importing profile from 2.7 i was expecting to see my exposed settings only. I am back to 2.7 cause i don't feel like going through all the settings and  unchecking what i wish not to see.

I might be doing something wrong here. Do I?

diegopradogesto, 2017-10-11:
When you export the profile, the visible settings are not stored. If you want to keep your visible settings, what you can do is save your configuration as a cura project file. This will store you visible settings in 2.7. Then you can import the project and both the visible settings and your profile will be there.

Let us know if this is what you want.

Thisismydigitalself, 2017-10-11:
Just tried that. does not work.
I am able to import the project but the visible settings are not the same.

diegopradogesto, 2017-10-11:
Could you add some screenshots to see what happen?

Thisismydigitalself, 2017-10-11:
I will but before I do you can check this yourself and see that when you import 2.7 curaproject to 3.0 you will get different settings visible.

diegopradogesto, 2017-10-12:
Oh, yeah, this is a known bug, we are working on it (CURA-4344). It needs to be tested and will be release probably in the next update.

Ghostkeeper, 2017-10-20:
That bug was that it shows that "0" settings are visible, in the dialogue when loading a project, not that it actually makes all settings invisible right? Seeing as how this also happened to Thisismydigitalself when loading single profiles, it seems to be a different bug.

We also know that _new_ settings are loaded in a default visibility state when you upgrade. As in, if you enabled all settings on 2.7 and then upgrade to 3.0, the settings for skin expansion will not be made visible because they didn't exist yet in 2.7. This is expected behaviour though.

diegopradogesto, 2017-10-20:
Oh, yes, this bug that I commented is related with 3.0 project files, not 2.7 ones. This is already fixed for next release. However, in some 2.7 project files, there is no information about the visible settings, and maybe that is the case.
It would be helpful if you can send the project file so we can take a look at it.

Thisismydigitalself, 2017-11-23:
Will 3.1 which is due any day now, fix this issue?

ChrisTerBeke, 2017-11-23:
It was indeed fixed and is in the 3.1 branch. The beta for that is due end of today.

nallath, 2017-11-27:
3.1 beta is out; Could you please re-test it?

Thisismydigitalself, 2017-11-27:
I am a bit lost when it comes to visible settings.
I am playing with 3.1. i was under the impression that my visible settings will be kept as part of my project or even with an empty 3MF file. it seems to not be the case. i was hoping that when i save a project with predefined settings and which are visible and which are not - this will be saved for me.

when i go to settings, deselect  all settings, there is no way back and i need to go through all settings and redefine all over again.

am i doing something wrong?

Ghostkeeper, 2017-12-11:
No, that should work. I'm trying right now and it seems to work. My steps:

1. Start from default visibility settings (actually I worked with a completely clean installation).
2. Make Infill Pattern visible.
3. Save project.
4. Close Cura. Remove preferences.
5. Open Cura. Add the same printer again. (Confirm that Infill Pattern is invisible.)
6. Load project.

At the end of this, Infill Pattern is visible again.

Thisismydigitalself, 2018-02-26:
This is an old case. since 3.2.x i started again to save/open projects.  Cura kept copying files from previous versions causing very bad results. what fixed most of my issues was completely deleting 3.2 folder from AppData and when Cura created new folder during next run things started working as they should. there is still this issue where Cura Project will not include global settings which are a must - see here: https://github.com/Ultimaker/Cura/issues/3201#issuecomment-364137893

ianpaschal, 2018-02-26:
@Thisismydigitalself Does this mean we can close this?

Thanks for helping us keep up to date.

Thisismydigitalself, 2018-02-26:
I would have said yes but it seems Case #3201 was closed as well which means it looks as though my main reported issue will not be resolved. please see what @fieldOfView said about my issue: https://github.com/Ultimaker/Cura/issues/3201#issuecomment-364137893

> Even though I hardly use projects myself, so this is not much of an issue for me personally, I agree with @Thisismydigitalself that the current behavior is not expected with the term "project". The data is there, and loads as expected for a split second, only for the parts to fall apart. I can imagine that would be frustrating.

ianpaschal, 2018-02-27:
As you are still not able to import your 2.7 profile?

Thisismydigitalself, 2018-02-27:
Nope. 
The bottom line of all these long project's reported issues is that saving and opening projects does not work as it should.  Please do not close case https://github.com/Ultimaker/Cura/issues/3201#issuecomment-364137893 cause it is the only good thread that towards the end summaries what's wrong with it.

thx.

ianpaschal, 2018-02-27:
Fair enough. I think @Ghostkeeper's reason for closing it is that it doesn't have a specific cause or fix and is therefore difficult to track as an issue. I will re-label it as a discussion and re-open it, although we may eventually close it again in favor of specific issue-n-fix tickets as they come up and are dealt with.

closed 2018-02-27 by ianpaschal