Ultimaker/Cura #2580: Unable to export 2 files from one session.  Requires shut down, reload, then save

Olive109, 2017-10-10:
When a drawing is loaded, sliced, and exported to an external drive, CURA allows changes to the slicing but will not allow exporting of the new sliced part to the external drive.  The only fix i've found is to close the program, open it, start over.  VERY inefficient.  

Is there another fix?

fieldOfView, 2017-10-10:
What version of Cura do you use?

Olive109, 2017-10-10:
2.6.2

fieldOfView, 2017-10-10:
Try Cura 2,7, or the Cura 3.0 beta that was released today.

Olive109, 2017-10-10:
Ok – trying to download 7 now

Olive109, 2017-10-10:
2.7 worked – I can save a second time to an external disc.

By the way, the update lists several great changes, but I’m not seeing them in my list of options when setting up a ‘custom’ print.  Things like Relative Z seam, Top surface skin, first layer line width, etc.  where do I go to activate or deactivate these functions?

Eric

nallath, 2017-10-11:
Check the settings visibility. You can find it in preferences->settings

Olive109, 2017-10-12:
Found it. Thank you.

From phone

closed 2017-10-16 by nallath