Ultimaker/Cura #2581: Some settings are not being properly saved in Profiles

EliW, 2017-10-11:
So I've been having no end of issues lately where my printer will seem to be printing FINE.  Then suddenly it goes all wonky, starts having adhesion issues, gets stringy again, etc.

I FINALLY found the Culprit.  Cura isn't saving some of my settings into profiles when I tell it to.   That is to say, it saves SOME things, but not others.   So later when I have a need to stop Cura and later restart it, the settings are reverted.

Ones that I can tell:
- Retraction Distance
- Combing Mode
- Coasting Volume

Attaching screenshot showing my saved profile and live settings showing that retraction is different (3.5 vs 4) ... and one showing that combing mode is different:  (none v all)  
[image]
[image]

EliW, 2017-10-11:
Cura 2.7 on Mac

ChrisTerBeke, 2017-10-11:
Tried to reproduce this using 2.7 and 3.0 (from source), but I can't. Can you share a project file (file -> save project)?

EliW, 2017-10-11:
Interesting, so I went back to reproduce, and the bug isn't quite what I thought.   If I reload a profile after updating it, it does in fact have the proper settings.  But the preview of settings in the Preferences/Profiles is wrong, as seen above.

To reproduce, take an existing profile you have.   change a few things, then do 'Create new profile from existing settings', then change some more settings, and do 'update profile from existing settings'.   The settings shown in the preferences window will be incorrect.  (At least, that's what happened to me)

Will a project file still help you @ChrisTerBeke ?

ChrisTerBeke, 2017-10-12:
Project files are always great as they come closest to duplicating your setup.

Know that you've described the issue a bit differently I think it's something we knew of and fixed in 3.0 (which will be released on Tuesday!). I'll confirm if that's the case.

closed 2018-02-26 by ianpaschal