Ultimaker/Cura #2582: Combing mode speeds up print, instead of slowing it down

EliW, 2017-10-11:
Theoretically having Combing mode on, should slow down the print process as it needs to make inefficient moves to get the head to where it needs to be instead of just going the most direct route.

However on a test print I've been doing, I found that going from combing being turned off (43min) that the printing time dropped dramatically when combing was turned on (27min) ...

EliW, 2017-10-11:
Cura 2.7 on Mac

smartavionics, 2017-10-11:
Well, doesn't that depend on whether it does a retraction when using a direct move and how long the retract/prime take compared to moving the extra distance involved in combing?

EliW, 2017-10-11:
I mean, sure, I guess it would.  But I would assume that 'no combing' would mean: "Just print the fastest you can".   No sense in minimizing movement via not combing, if it ends up taking longer due to retraction/prime.  IMO.

Ghostkeeper, 2017-10-20:
I think it's logical that "No combing" only removes combing and then automatically the retraction has to kick in more often to still get a passable result.

The main use case of disabling combing, in my opinion, is to prevent scars and have it z-hop instead. It could also be slightly faster if you don't have a Bowden tube and slow travel speed...

ianpaschal, 2018-03-09:
Since this is a general question and hasn't been discussed in 6+ months I'm closing.

closed 2018-03-09 by ianpaschal