Ultimaker/Cura #2587: how to host robeo on local host

hadoop-raiabhishek, 2017-10-11:
i want a python GUI to work remotely through a web url. i.e. when they hit the url they get the python interface to work on.  I am not getting a server for python like for Rstdio there is a shiny server which host the rstdio. 

Please help.

ChrisTerBeke, 2017-10-11:
Hi! It seems your question is not really related to Cura (this repository), but as far as I know there is no way to expose a Python GUI via the web other than using screenshare-like technologies (like Google Hangouts). I'm not familiar with Rstio but I think it's a native application. Another solution could be exposing a web server via Python (there are many frameworks for that, for example Flask), and building the GUI in web technologies (HTML/CSS/JavaScript).

(p.s. closing this issue as it's not a Cura bug, but you can still reply to it)

closed 2017-10-11 by ChrisTerBeke

Ghostkeeper, 2017-10-20:
Qt 5.10 will have a feature to stream a Qt GUI to a web browser, which does exactly what you need for Cura. But 5.10 hasn't been released yet.