Ultimaker/Cura #2589: Custom Cura Platform

mypeopoly, 2017-10-11:
Hi Cura Team,

Is there a way to customize how the build platform looks in Cura?
[image]

The reason I want to change it is that I want to help the user to identify where the peel force is strongest on which part of the platform and how orientation works in SLA printing. This platform will be part of a customized version of Cura (https://github.com/peopoly3d/moai-cura)  
This is how I envision the platform will look like:
[image]

fieldOfView, 2017-10-11:
Since you are making your own version of Cura, you could change the grid shader to add the colors in your sketch. The "sideflaps" can be added as a platform mesh, of which there are many examples.

The shader for the grid can be found here: 
https://github.com/Ultimaker/Cura/blob/master/resources/shaders/grid.shader

mypeopoly, 2017-10-11:
Thank you for your help

So is this piece code is defining plate color:
u_plateColor = [1.0, 1.0, 1.0, 1.0]
u_gridColor0 = [0.96, 0.96, 0.96, 1.0]
u_gridColor1 = [0.8, 0.8, 0.8, 1.0]

how to shade making gradient colors like the above?
--
What would be the place to add Mesh?

nallath, 2017-10-11:
Or even easier; Add a build plate with a texture. That means you don't even have to change anything in the code. 
OBJ items can have textures. The UM2 build plate uses this already to display the name.

mypeopoly, 2017-10-13:
here it is:

https://peopoly.net/blogs/news/how-to-customize-cura-build-platform-for-moai

thanks guys

closed 2017-10-13 by mypeopoly