Ultimaker/Cura #2596: Upgrading to Qt 5.9.2: Cura doesn't start

JPFrancoia, 2017-10-12:
I'm on Archlinux. After upgrading to Qt 5.9.2, Cura doesn't start anymore.

```
UM.Resources.__initializeStoragePaths [342]: Initializing storage paths
UM.Resources.__initializeStoragePaths [353]: Config storage path is /home/djipey/.config/cura/master
UM.Resources.__initializeStoragePaths [361]: Data storage path is /home/djipey/.local/share/cura/master
UM.Resources.__initializeStoragePaths [373]: Cache storage path is /home/djipey/.cache/cura/master
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [103]: Trying OpenGL context 4.1...
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [113]: Yay, we got at least OpenGL 4.1 core: 4.5 Core profile
UM.Qt.QtApplication.__init__ [80]: Detected most suitable OpenGL context version: 4.1 Core profile
2017-10-12 15:21:34,178 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ConsoleLogger
2017-10-12 15:21:34,207 - INFO - CuraEngineBackend.CuraEngineBackend.__init__ [64]: Found CuraEngine at: /usr/bin/CuraEngine
2017-10-12 15:21:34,208 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin CuraEngineBackend
2017-10-12 15:21:34,218 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin LocalFileOutputDevice
2017-10-12 15:21:34,254 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin UpdateChecker
2017-10-12 15:21:34,260 - INFO - UpdateChecker.UpdateCheckerJob.run [44]: Checking for new version of cura
2017-10-12 15:21:34,265 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin STLWriter
2017-10-12 15:21:34,268 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin OBJWriter
2017-10-12 15:21:34,272 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin OBJReader
2017-10-12 15:21:34,276 - WARNING - STLReader.STLReader.<module> [25]: Could not find numpy-stl, falling back to slower code.
2017-10-12 15:21:34,279 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin STLReader
2017-10-12 15:21:34,289 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin RotateTool
2017-10-12 15:21:34,294 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin TranslateTool
2017-10-12 15:21:34,298 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SelectionTool
2017-10-12 15:21:34,303 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin MirrorTool
2017-10-12 15:21:34,310 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ScaleTool
2017-10-12 15:21:34,314 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin CameraTool
2017-10-12 15:21:34,316 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SimpleView
2017-10-12 15:21:34,316 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin ConsoleLogger was already loaded
2017-10-12 15:21:34,318 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin FileLogger
2017-10-12 15:21:34,328 - ERROR - UM.Logger.logException [76]: Exception: Import error loading module UM3NetworkPrinting
2017-10-12 15:21:34,328 - ERROR - UM.Logger.logException [80]: Traceback (most recent call last):
2017-10-12 15:21:34,328 - ERROR - UM.Logger.logException [80]:   File "/usr/lib/python3.6/site-packages/UM/PluginRegistry.py", line 464, in _findPlugin
2017-10-12 15:21:34,328 - ERROR - UM.Logger.logException [80]:     module = imp.load_module(plugin_id, file, path, desc)
2017-10-12 15:21:34,329 - ERROR - UM.Logger.logException [80]:   File "/usr/lib/python3.6/imp.py", line 244, in load_module
2017-10-12 15:21:34,329 - ERROR - UM.Logger.logException [80]:     return load_package(name, filename)
2017-10-12 15:21:34,329 - ERROR - UM.Logger.logException [80]:   File "/usr/lib/python3.6/imp.py", line 216, in load_package
2017-10-12 15:21:34,329 - ERROR - UM.Logger.logException [80]:     return _load(spec)
2017-10-12 15:21:34,329 - ERROR - UM.Logger.logException [80]:   File "<frozen importlib._bootstrap>", line 675, in _load
2017-10-12 15:21:34,329 - ERROR - UM.Logger.logException [80]:   File "<frozen importlib._bootstrap>", line 655, in _load_unlocked
2017-10-12 15:21:34,329 - ERROR - UM.Logger.logException [80]:   File "<frozen importlib._bootstrap_external>", line 678, in exec_module
2017-10-12 15:21:34,329 - ERROR - UM.Logger.logException [80]:   File "<frozen importlib._bootstrap>", line 205, in _call_with_frames_removed
2017-10-12 15:21:34,329 - ERROR - UM.Logger.logException [80]:   File "/usr/lib/cura/plugins/UM3NetworkPrinting/__init__.py", line 3, in <module>
2017-10-12 15:21:34,329 - ERROR - UM.Logger.logException [80]:     from . import NetworkPrinterOutputDevicePlugin
2017-10-12 15:21:34,330 - ERROR - UM.Logger.logException [80]:   File "/usr/lib/cura/plugins/UM3NetworkPrinting/NetworkPrinterOutputDevicePlugin.py", line 7, in <module>
2017-10-12 15:21:34,330 - ERROR - UM.Logger.logException [80]:     from zeroconf import Zeroconf, ServiceBrowser, ServiceStateChange, ServiceInfo  # type: ignore
2017-10-12 15:21:34,330 - ERROR - UM.Logger.logException [80]: ModuleNotFoundError: No module named 'zeroconf'
2017-10-12 15:21:34,330 - WARNING - UM.PluginRegistry._populateMetaData [381]: Could not find plugin UM3NetworkPrinting
2017-10-12 15:21:34,332 - ERROR - UM.Logger.logException [76]: Exception: Import error loading module UM3NetworkPrinting
2017-10-12 15:21:34,332 - ERROR - UM.Logger.logException [80]: Traceback (most recent call last):
2017-10-12 15:21:34,332 - ERROR - UM.Logger.logException [80]:   File "/usr/lib/python3.6/site-packages/UM/PluginRegistry.py", line 464, in _findPlugin
2017-10-12 15:21:34,332 - ERROR - UM.Logger.logException [80]:     module = imp.load_module(plugin_id, file, path, desc)
2017-10-12 15:21:34,332 - ERROR - UM.Logger.logException [80]:   File "/usr/lib/python3.6/imp.py", line 244, in load_module
2017-10-12 15:21:34,332 - ERROR - UM.Logger.logException [80]:     return load_package(name, filename)
2017-10-12 15:21:34,332 - ERROR - UM.Logger.logException [80]:   File "/usr/lib/python3.6/imp.py", line 216, in load_package
2017-10-12 15:21:34,332 - ERROR - UM.Logger.logException [80]:     return _load(spec)
2017-10-12 15:21:34,332 - ERROR - UM.Logger.logException [80]:   File "<frozen importlib._bootstrap>", line 675, in _load
2017-10-12 15:21:34,332 - ERROR - UM.Logger.logException [80]:   File "<frozen importlib._bootstrap>", line 655, in _load_unlocked
2017-10-12 15:21:34,333 - ERROR - UM.Logger.logException [80]:   File "<frozen importlib._bootstrap_external>", line 678, in exec_module
2017-10-12 15:21:34,333 - ERROR - UM.Logger.logException [80]:   File "<frozen importlib._bootstrap>", line 205, in _call_with_frames_removed
2017-10-12 15:21:34,333 - ERROR - UM.Logger.logException [80]:   File "/usr/lib/cura/plugins/UM3NetworkPrinting/__init__.py", line 3, in <module>
2017-10-12 15:21:34,333 - ERROR - UM.Logger.logException [80]:     from . import NetworkPrinterOutputDevicePlugin
2017-10-12 15:21:34,333 - ERROR - UM.Logger.logException [80]:   File "/usr/lib/cura/plugins/UM3NetworkPrinting/NetworkPrinterOutputDevicePlugin.py", line 7, in <module>
2017-10-12 15:21:34,333 - ERROR - UM.Logger.logException [80]:     from zeroconf import Zeroconf, ServiceBrowser, ServiceStateChange, ServiceInfo  # type: ignore
2017-10-12 15:21:34,333 - ERROR - UM.Logger.logException [80]: ModuleNotFoundError: No module named 'zeroconf'
2017-10-12 15:21:34,341 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin UltimakerMachineActions
2017-10-12 15:21:34,345 - WARNING - 3MFWriter.<module> [9]: Could not import ThreeMFWriter; libSavitar may be missing
2017-10-12 15:21:34,350 - WARNING - 3MFWriter.<module> [9]: Could not import ThreeMFWriter; libSavitar may be missing
2017-10-12 15:21:34,350 - ERROR - UM.PluginRegistry.loadPlugin [220]: Plugin 3MFWriter did not return any objects to register
2017-10-12 15:21:34,387 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin USBPrinting
[... 91 lines cut ...]
[2017-10-11 14:29] [ALPM] upgraded python-pyqt5 (5.9-1 -> 5.9-2)
[2017-10-11 14:30] [ALPM] upgraded python2-pyqt5 (5.9-1 -> 5.9-2)
[2017-10-11 14:30] [ALPM] upgraded qt5-xmlpatterns (5.9.1-2 -> 5.9.2-1)
[2017-10-11 14:30] [ALPM] upgraded qt5-declarative (5.9.1-5 -> 5.9.2-1)
[2017-10-11 14:30] [ALPM] upgraded qt5-location (5.9.1-2 -> 5.9.2-1)
[2017-10-11 14:30] [ALPM] upgraded qt5-multimedia (5.9.1-2 -> 5.9.2-1)
[2017-10-11 14:30] [ALPM] upgraded qt5-quickcontrols (5.9.1-2 -> 5.9.2-1)
[2017-10-11 14:30] [ALPM] upgraded qt5-script (5.9.1-2 -> 5.9.2-1)
[2017-10-11 14:30] [ALPM] upgraded qt5-sensors (5.9.1-2 -> 5.9.2-1)
[2017-10-11 14:30] [ALPM] upgraded qt5-speech (5.9.1-2 -> 5.9.2-1)
[... cut ...]
I like Cura but every week, an upgrade will break it. I wouldn't mind if I was using it for hobby stuff, but I use it for work almost everyday. I'd like something more stable.

Of course, each time I find a workaround for a bug, I loose all my profiles/machines/settings. Even if I backup ```.local/share/cura``` and ```.config/cura``` and restore these folders, I never managed to restore my settings.

For now I'm going to use the AppImage. It's not ideal because I can't use my custom postProcessing scripts.

Any idea/advice to improve my situation is very welcome.

nallath, 2017-10-12:
I don't see how upgrading would influence your local files. Backing up those files (and restoring them) should work. I use that myself to set Cura to certain states (I am on fedora, but that shouldn't be an issue).

As for the custom scripts; The plugin now only checks in own location for scripts. If anyone has a good suggestion where else it should check, it's not that hard to expand on it.

Stunkymonkey, 2017-10-12:
i can confirm this issue
there are more people having the same problem: https://bugs.archlinux.org/task/55963?project=5&cat%5B0%5D=33&string=cura

JPFrancoia, 2017-10-12:
@nallath When I encounter a bug in Cura, I back up these files, then eras them, uninstall and reinstall cura. Just to make sure that my config files are not the problem.

But each time I try to restore, it fails, and I have to recreate machines/plugins/settings.

Is there any other file/directory I should back up ? How do you back up ? Do you have a script ?

LipuFei, 2017-10-13:
Looks like you need the `zeroconf` package. And this plugin of yours `/usr/lib/cura/plugins/SliceInfoPlugin` needs to be updated: we now require a `plugin.json` for each plugin.

sjoerdtimmer, 2017-10-13:
I can confirm the issue of the missing main window when upgrading to Qt 5.9.2. 
Other qt applications seem unaffected so on a hunch I'd say this is a cura/uranium issue.

sjoerdtimmer, 2017-10-13:
Here is a "clean" print of the console (after removing all user configs temporarily), showing that the issue is not related to the missing zeroconf or bad plugin:
```
 UM.Resources.__initializeStoragePaths [342]: Initializing storage paths
UM.Resources.__initializeStoragePaths [353]: Config storage path is /home/themba/.config/cura/master
UM.Resources.__initializeStoragePaths [361]: Data storage path is /home/themba/.local/share/cura/master
UM.Resources.__initializeStoragePaths [373]: Cache storage path is /home/themba/.cache/cura/master
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [103]: Trying OpenGL context 4.1...
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [113]: Yay, we got at least OpenGL 4.1 core: 4.5 Core profile
UM.Qt.QtApplication.__init__ [80]: Detected most suitable OpenGL context version: 4.1 Core profile
2017-10-12 19:25:47,818 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ConsoleLogger
2017-10-12 19:25:47,841 - INFO - CuraEngineBackend.CuraEngineBackend.__init__ [64]: Found CuraEngine at: /usr/bin/CuraEngine
2017-10-12 19:25:47,841 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin CuraEngineBackend
2017-10-12 19:25:47,855 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin STLReader
2017-10-12 19:25:47,859 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin OBJWriter
2017-10-12 19:25:47,862 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin OBJReader
2017-10-12 19:25:47,866 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin STLWriter
2017-10-12 19:25:47,875 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin RotateTool
2017-10-12 19:25:47,881 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin TranslateTool
2017-10-12 19:25:47,886 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin CameraTool
2017-10-12 19:25:47,896 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ScaleTool
2017-10-12 19:25:47,901 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin MirrorTool
2017-10-12 19:25:47,905 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SelectionTool
2017-10-12 19:25:47,908 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin FileLogger
2017-10-12 19:25:47,912 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SimpleView
2017-10-12 19:25:47,912 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin ConsoleLogger was already loaded
2017-10-12 19:25:47,919 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin LocalFileOutputDevice
2017-10-12 19:25:47,958 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin UpdateChecker
2017-10-12 19:25:47,967 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SolidView
2017-10-12 19:25:47,969 - INFO - UpdateChecker.UpdateCheckerJob.run [44]: Checking for new version of cura
2017-10-12 19:25:47,977 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin VersionUpgrade22to24
2017-10-12 19:25:47,984 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin VersionUpgrade21to22
2017-10-12 19:25:47,988 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin VersionUpgrade25to26
2017-10-12 19:25:47,992 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin VersionUpgrade26to27
2017-10-12 19:25:47,992 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin CuraEngineBackend was already loaded
2017-10-12 19:25:47,998 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin MachineSettingsAction
2017-10-12 19:25:48,048 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin UM3NetworkPrinting
2017-10-12 19:25:48,053 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin XRayView
2017-10-12 19:25:48,066 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin X3DReader
2017-10-12 19:25:48,070 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin GCodeProfileReader
2017-10-12 19:25:48,075 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ImageReader
2017-10-12 19:25:48,079 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin CuraProfileWriter
2017-10-12 19:25:48,079 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin STLReader was already loaded
2017-10-12 19:25:48,079 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin OBJWriter was already loaded
2017-10-12 19:25:48,080 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin OBJReader was already loaded
2017-10-12 19:25:48,080 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin STLWriter was already loaded
2017-10-12 19:25:48,088 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin XmlMaterialProfile
2017-10-12 19:25:48,092 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin CuraProfileReader
2017-10-12 19:25:48,104 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin LayerView
2017-10-12 19:25:48,104 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin RotateTool was already loaded
2017-10-12 19:25:48,104 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin TranslateTool was already loaded
2017-10-12 19:25:48,104 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin CameraTool was already loaded
2017-10-12 19:25:48,104 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin ScaleTool was already loaded
2017-10-12 19:25:48,105 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin MirrorTool was already loaded
2017-10-12 19:25:48,105 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin SelectionTool was already loaded
[... cut ...]
2017-10-12 19:25:58,715 - WARNING - UM3NetworkPrinting.DiscoverUM3Action._createAdditionalComponentsView [148]: Could not create ui components for UM3.
2017-10-12 19:25:58,725 - DEBUG - UM.Mesh.MeshData.calculateNormalsFromIndexedVertices [448]: Calculating normals took 0.0021505355834960938 seconds
2017-10-12 19:25:58,738 - DEBUG - UM.Mesh.MeshData.approximateConvexHull [392]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.003901243209838867 seconds. 55 input vertices. 55 output vertices.
2017-10-12 19:26:08,174 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
```

tyr431, 2017-10-13:
Thank god, there are more people experiencing this.
Can confirm, it does not crash, but does not open the main window after the loading screen either.

AiyionPrime, 2017-10-13:
Thought about zeroconf at first, but this stays after I added the module.
Does not run the main window.

```
UM.Resources.__initializeStoragePaths [342]: Initializing storage paths
UM.Resources.__initializeStoragePaths [353]: Config storage path is /home/aiyion/.config/cura/master
UM.Resources.__initializeStoragePaths [361]: Data storage path is /home/aiyion/.local/share/cura/master
UM.Resources.__initializeStoragePaths [373]: Cache storage path is /home/aiyion/.cache/cura/master
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [103]: Trying OpenGL context 4.1...
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [113]: Yay, we got at least OpenGL 4.1 core: 4.5 Core profile
UM.Qt.QtApplication.__init__ [80]: Detected most suitable OpenGL context version: 4.1 Core profile
2017-10-13 16:09:35,725 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ConsoleLogger
2017-10-13 16:09:35,743 - INFO - CuraEngineBackend.CuraEngineBackend.__init__ [64]: Found CuraEngine at: /usr/bin/CuraEngine
2017-10-13 16:09:35,744 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin CuraEngineBackend
2017-10-13 16:09:35,746 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SimpleView
2017-10-13 16:09:35,751 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin LocalFileOutputDevice
2017-10-13 16:09:35,753 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin FileLogger
2017-10-13 16:09:35,755 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin OBJReader
2017-10-13 16:09:35,758 - WARNING - STLReader.STLReader.<module> [25]: Could not find numpy-stl, falling back to slower code.
2017-10-13 16:09:35,760 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin STLReader
2017-10-13 16:09:35,762 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin OBJWriter
2017-10-13 16:09:35,765 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin STLWriter
2017-10-13 16:09:35,765 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin ConsoleLogger was already loaded
2017-10-13 16:09:35,771 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin RotateTool
2017-10-13 16:09:35,775 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin MirrorTool
2017-10-13 16:09:35,778 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SelectionTool
2017-10-13 16:09:35,784 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ScaleTool
2017-10-13 16:09:35,789 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin TranslateTool
2017-10-13 16:09:35,792 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin CameraTool
2017-10-13 16:09:35,832 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin UpdateChecker
2017-10-13 16:09:35,841 - INFO - UpdateChecker.UpdateCheckerJob.run [44]: Checking for new version of cura
2017-10-13 16:09:35,843 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin PerObjectSettingsTool
2017-10-13 16:09:35,848 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin PluginBrowser
2017-10-13 16:09:35,852 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin GCodeWriter
2017-10-13 16:09:35,855 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin XRayView
2017-10-13 16:09:35,860 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin RemovableDriveOutputDevice
2017-10-13 16:09:35,860 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin SimpleView was already loaded
2017-10-13 16:09:35,867 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin VersionUpgrade21to22
2017-10-13 16:09:35,871 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin VersionUpgrade25to26
2017-10-13 16:09:35,873 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin VersionUpgrade26to27
2017-10-13 16:09:35,877 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin VersionUpgrade22to24
2017-10-13 16:09:35,877 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin LocalFileOutputDevice was already loaded
2017-10-13 16:09:35,880 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ChangeLogPlugin
2017-10-13 16:09:35,880 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin FileLogger was already loaded
2017-10-13 16:09:35,880 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin OBJReader was already loaded
2017-10-13 16:09:35,881 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin STLReader was already loaded
2017-10-13 16:09:35,881 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin OBJWriter was already loaded
2017-10-13 16:09:35,881 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin STLWriter was already loaded
2017-10-13 16:09:35,885 - WARNING - 3MFReader.<module> [10]: Could not import ThreeMFReader; libSavitar may be missing
2017-10-13 16:09:35,899 - WARNING - 3MFReader.<module> [10]: Could not import ThreeMFReader; libSavitar may be missing
2017-10-13 16:09:35,899 - ERROR - UM.PluginRegistry.loadPlugin [220]: Plugin 3MFReader did not return any objects to register
2017-10-13 16:09:35,908 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin X3DReader
2017-10-13 16:09:35,914 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin XmlMaterialProfile
2017-10-13 16:09:35,914 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin ConsoleLogger was already loaded
2017-10-13 16:09:35,928 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin USBPrinting
2017-10-13 16:09:35,933 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin CuraProfileWriter
[... cut ...]
2017-10-13 16:09:46,603 - DEBUG - UM.Qt.Bindings.Theme.load [232]: Loaded theme /usr/share/cura/resources/themes/cura
2017-10-13 16:09:46,610 - DEBUG - UM3NetworkPrinting.DiscoverUM3Action._createAdditionalComponentsView [138]: Creating additional ui components for UM3.
QQmlComponent: Component is not ready
2017-10-13 16:09:46,611 - WARNING - UM3NetworkPrinting.DiscoverUM3Action._createAdditionalComponentsView [148]: Could not create ui components for UM3.
2017-10-13 16:09:56,708 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
```

sjoerdtimmer, 2017-10-13:
Yes, this affects all archlinux users who have recently updated. I hope someone from Ultimaker can fix this soon...

fieldOfView, 2017-10-13:
As far as I know, they already fixed it by switching to AppImage. According to the bugreport on bugs.archlinux.org that works. Any other Cura distribution is a community effort, so it is also a community effort to keep them running.

sjoerdtimmer, 2017-10-13:
I'd be more than happy to invest some time to keep this working. Do you have any suggestions where to start looking? Without an error or crash I'd be looking for a needle in a haystack... I'll start by search for changelogs in Qt.

JPFrancoia, 2017-10-13:
Also I didn't manage to use custom plugins with the AppImage. Any idea how to do it ?

JPFrancoia, 2017-10-16:
And where is that exactly ?

With AppImage 2.7, I tried to install my ```PostProcessingPlugin``` scripts in ```.local/share/cura/2.7/plugins```

But when I try to use them, it seems I'm using the default ```PostProcessingPlugin```

fieldOfView, 2017-10-16:
Ah, you mean postprocessing _scripts_, not Cura _plugins_. Yes, sorry, that's a bit of a problem, because they are only read from the application resources. Good point.

JPFrancoia, 2017-10-16:
Ah ! Yes the name is a bit ambiguous.

Would it require much work to make have a look at ```.local/share/cura/2.7/plugins``` ?

sjoerdtimmer, 2017-10-17:
Regarding the real issue: I verified against qt 5.10.0 and the issue persist. Sooner or later this is going to become an issue for a next appimage too... ;-)

@nallath, @fieldOfView, any suggestions where I might start looking? Have you seen such behavior (no crash, but no mainwindow) before or maybe a hunch about where things may go wrong?

fieldOfView, 2017-10-17:
> Have you seen such behavior (...) before or maybe a hunch

I have neither. My first try would be to see if a minimal pyqt application works.
Cura creates a window in QML here: https://github.com/Ultimaker/Cura/blob/master/resources/qml/Cura.qml
UM.MainWindow is a subclassed QQuickWindow: https://github.com/Ultimaker/Uranium/blob/master/UM/Qt/Bindings/MainWindow.py
I would also check if the MainWindow class gets instanced at all.

sjoerdtimmer, 2017-10-17:
@fieldOfView thanks. I already reverted all changes to qt5-quickcontrols made in 5.9.2 creating a package that is indentical to 5.9.1 (keeping the 5.9.2 version number to fool the version checks) and this results in the same issue. The problem is therefore likely not in quickcontrols. I'll do the same for other qt packages later.

sjoerdtimmer, 2017-10-17:
The MainWindow.py file is loaded but the constructor is indeed never called :s

Appesteijn, 2017-10-24:
When I upgraded to qt 5.9.2 on my MacOS (10.12.6) I experienced this also when running Cura from the commandline.

sjoerdtimmer, 2017-10-25:
@Appesteijn your phrasing suggests that there is another way to start cura that does work? Is that correct?

Appesteijn, 2017-10-25:
No unfortunately I haven't found a solution to get Cura running from the commandline with Qt 5.9.2

AiyionPrime, 2017-10-25:
@Appesteijn The way you explicitely stated, that it would not work, when starting it via commandline, suggests you are able to start it, if you don't do it via commandline, eg. a desktop shortcut or something;
Which is probably not the case, right?
So, what you really wanted to say is, you can't start Cura on MacOS anymore, after updating qt?

Appesteijn, 2017-10-25:
No I can run the download (and the daily builds) perfectly. Only when I run 'master' from source it doesn't work anymore.

sjoerdtimmer, 2017-10-25:
I guess @Appesteijn means that he can run the AppImages...

sjoerdtimmer, 2017-10-25:
I am quite stumped by qt-quicks decision to no longer launch the main window. The python class is run but never called. I have asked around in #qt-quick on freenode but havn't gotten an answer :s

hroncok, 2017-10-26:
Just to confirm. This is also a problem on Fedora 27 with Qt 5.9.2.

nmaggioni, 2017-10-27:
These are the packages that I needed to downgrade to in order to get Cura showing again:

```
pyqt5-common-5.9-1-x86_64.pkg.tar.xz
python2-pyqt5-5.9-1-x86_64.pkg.tar.xz
python-pyqt5-5.9-1-x86_64.pkg.tar.xz
qt5-3d-5.9.1-2-x86_64.pkg.tar.xz
qt5-base-5.9.1-2-x86_64.pkg.tar.xz
qt5-canvas3d-5.9.1-2-x86_64.pkg.tar.xz
qt5-connectivity-5.9.1-2-x86_64.pkg.tar.xz
qt5-declarative-5.9.1-5-x86_64.pkg.tar.xz
qt5-graphicaleffects-5.9.1-1-x86_64.pkg.tar.xz
qt5-imageformats-5.9.1-1-x86_64.pkg.tar.xz
qt5-location-5.9.1-2-x86_64.pkg.tar.xz
qt5-multimedia-5.9.1-2-x86_64.pkg.tar.xz
qt5-quickcontrols2-5.9.1-2-x86_64.pkg.tar.xz
qt5-quickcontrols-5.9.1-2-x86_64.pkg.tar.xz
qt5-script-5.9.1-2-x86_64.pkg.tar.xz
qt5-sensors-5.9.1-2-x86_64.pkg.tar.xz
qt5-serialport-5.9.1-2-x86_64.pkg.tar.xz
qt5-speech-5.9.1-2-x86_64.pkg.tar.xz
qt5-svg-5.9.1-2-x86_64.pkg.tar.xz
qt5-tools-5.9.1-3-x86_64.pkg.tar.xz
qt5-translations-5.9.1-1-any.pkg.tar.xz
qt5-wayland-5.9.1-2-x86_64.pkg.tar.xz
qt5-webchannel-5.9.1-2-x86_64.pkg.tar.xz
qt5-webengine-5.9.1-2-x86_64.pkg.tar.xz
qt5-webkit-5.212.0alpha2-4-x86_64.pkg.tar.xz
qt5-websockets-5.9.1-2-x86_64.pkg.tar.xz
qt5-x11extras-5.9.1-2-x86_64.pkg.tar.xz
qt5-xmlpatterns-5.9.1-2-x86_64.pkg.tar.xz
```

hroncok, 2017-10-31:
BTW When reverting a particular commit from Qt, the problem is gone.

Identified by Rex Dieter:

https://src.fedoraproject.org/rpms/qt5-qtdeclarative/c/5c5ecc404d8c670c7b1813bf221ac48d03a197c4?branch=master

d235j, 2017-10-31:
I can confirm that reverting the commit that @hroncok identified makes things work.

hroncok, 2017-11-02:
This is relevant for 2.7.0 as well as 3.0.3. (I'm not or Archlinux, but Fedora, however I thought it might help you.)

protist, 2017-11-02:
@JPFrancoia I just tried Cura 3.0.3-1… I get the splash screen, it disappears, then Cura crashes
```
2017-11-02 21:33:02,875 - CRITICAL - cura.CrashHandler.show [39]: An uncaught exception has occurred!
2017-11-02 21:33:02,876 - CRITICAL - cura.CrashHandler.show [42]: Traceback (most recent call last):
2017-11-02 21:33:02,876 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/bin/cura", line 82, in <module>
2017-11-02 21:33:02,876 - CRITICAL - cura.CrashHandler.show [42]:     app.run()
2017-11-02 21:33:02,876 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/lib/python3.6/site-packages/cura/CuraApplication.py", line 640, in run
2017-11-02 21:33:02,876 - CRITICAL - cura.CrashHandler.show [42]:     self._volume = BuildVolume.BuildVolume(root)
2017-11-02 21:33:02,876 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/lib/python3.6/site-packages/cura/BuildVolume.py", line 102, in __init__
2017-11-02 21:33:02,877 - CRITICAL - cura.CrashHandler.show [42]:     " with printed models."), title = catalog.i18nc("@info:title","Build Volume"))
2017-11-02 21:33:02,877 - CRITICAL - cura.CrashHandler.show [42]: TypeError: __init__() got an unexpected keyword argument 'title'

jonnor, 2017-11-02:
@protist did you also update uranium package? I have seen that issue when using Cura git with older uranium.
However this is slightly on the side of this bug report, recommend filing it separately if issue persists

ChrisTerBeke, 2017-11-02:
unexpected keyword argument 'title' is definitely a discrepancy between Cura and Uranium (we added a new property to the message class to accept a message title).

protist, 2017-11-02:
Ah apologies. It looks like Arch updated Cura but not Uranium. (Related: why is [the latest version](https://github.com/Ultimaker/Uranium/releases) of Uranium in such small type?)

However, when I manually update uranium to 3.0.3, cura now launches perfectly!

ChrisTerBeke, 2017-11-02:
It's small because a git tag was created but no GitHub release (which is sort of a wrapper around git tags if you wanna add release notes and binaries).

jelly, 2017-11-02:
Sorry for the partial update of uranium/curaengine. Arch has also revert the same commit as Fedora did to make cura launch again. Hopefully Qt fixes this in their new release.

protist, 2017-11-02:
@ChrisTerBeke Thanks for the info. It's quite confusing, especially as the *previous* version says "Latest release"!

@jelly No worries. It's all good now, and thank you for the quick response.

Ghostkeeper, 2017-11-08:
Looks like a bugfix might be coming up in Qt 5.9.3: https://bugreports.qt.io/browse/QTBUG-64017

d235j, 2017-11-08:
Yeah — there seems to be a proposed patch set at https://codereview.qt-project.org/#/c/209990/ but I have not had time to test it.

JPFrancoia, 2017-11-12:
Arch linux people, is it safe to upgrade Qt and related packages, or not yet ? From what I understand the problem comes from QtDeclarative and there is a new revision:

5.9.2-1 -> 5.9.2-2

jelly, 2017-11-13:
@JPFrancoia cura starts fine with that version.

JPFrancoia, 2017-11-26:
Ok, cura works fine now. The Qt fix in the repo fixed it. Also updated to cura 3, and all works fine. Closing issue.

closed 2017-11-26 by JPFrancoia