Ultimaker/Cura #2598: Crah

MarcelPalmen, 2017-10-12:
When I start the software it crashes with these massages:
Version: JGcreat-2.5.0
Platform: Windows-8.1-6.3.9600
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\lib\site-packages\cx_Freeze\initscripts\__startup__.py", line 12, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\lib\site-packages\cx_Freeze\initscripts\Console.py", line 24, in <module>
  File "./inst/bin/cura_app.py", line 72, in <module>
  File "D:/Cura_2.5/cura-build-2.5.0/build/inst/lib/python3.5/site-packages/UM\Application.py", line 316, in getInstance
  File "D:/Cura_2.5/cura-build-2.5.0/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 196, in __init__
  File "D:/Cura_2.5/cura-build-2.5.0/build/inst/lib/python3.5/site-packages/UM\Qt\QtApplication.py", line 110, in __init__
  File "D:/Cura_2.5/cura-build-2.5.0/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 478, in _loadPlugins
  File "D:/Cura_2.5/cura-build-2.5.0/build/inst/lib/python3.5/site-packages/UM\PluginRegistry.py", line 224, in loadPlugins
  File "D:/Cura_2.5/cura-build-2.5.0/build/inst/lib/python3.5/site-packages/UM\PluginRegistry.py", line 244, in getMetaData
  File "D:/Cura_2.5/cura-build-2.5.0/build/inst/lib/python3.5/site-packages/UM\PluginRegistry.py", line 359, in _populateMetaData
UM.PluginError.InvalidMetaDataError: Invalid metadata for plugin BarbarianPlugin

-------------------------
Thread 0x00001004 (most recent call first):
  File "C:\Program Files\JGcreat 2.5.0\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 862 in run
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 882 in _bootstrap

Thread 0x000017dc (most recent call first):
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 297 in wait
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\lib\site-packages\zeroconf.py", line 1569 in wait
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\lib\site-packages\zeroconf.py", line 1153 in run
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 882 in _bootstrap

Thread 0x0000230c (most recent call first):
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 297 in wait
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\lib\site-packages\zeroconf.py", line 1569 in wait
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\lib\site-packages\zeroconf.py", line 1017 in run
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 882 in _bootstrap

Thread 0x00001ec8 (most recent call first):
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\lib\site-packages\zeroconf.py", line 946 in run
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 882 in _bootstrap

Thread 0x00001e58 (most recent call first):
  File "C:\Program Files\JGcreat 2.5.0\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 58 in _updateThread
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 862 in run
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 882 in _bootstrap

Thread 0x0000025c (most recent call first):
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 293 in wait
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 424 in acquire
  File "D:/Cura_2.5/cura-build-2.5.0/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/Cura_2.5/cura-build-2.5.0/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 882 in _bootstrap

Thread 0x00001630 (most recent call first):
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 293 in wait
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 424 in acquire
  File "D:/Cura_2.5/cura-build-2.5.0/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/Cura_2.5/cura-build-2.5.0/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 882 in _bootstrap

Current thread 0x00001548 (most recent call first):
  File "D:/Cura_2.5/cura-build-2.5.0/build/inst/lib/python3.5/site-packages/cura\CrashHandler.py", line 100 in show
  File "./inst/bin/cura_app.py", line 45 in exceptHook

nallath, 2017-10-16:
I don't know what version of Cura you are using; "JGcreat-2.5.0" is not a version that i know of.

Other than that; It's using the wrong version of the barbarian plugin, which is not backwards compatible. Remove the plguin and try again

trennal, 2017-10-17:
Fatal exception at startup  Also had the issue with 2.7 after Windows 10 updates to Windows 10 Pro V 1790, Build 16299.19.
Crash Report
Version: 3.0.1-BETA
Platform: Windows-10-10.0.16299
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\Qt\Bindings\MainWindow.py", line 120, in mousePressEvent
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\Qt\QtMouseDevice.py", line 24, in handleEvent
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 212, in emit
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 315, in __performEmit
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\Controller.py", line 257, in event
  File "C:\Program Files\Cura 3.0\plugins\Tools\SelectionTool\SelectionTool.py", line 66, in event
    self._pixelSelection(event)
  File "C:\Program Files\Cura 3.0\plugins\Tools\SelectionTool\SelectionTool.py", line 104, in _pixelSelection
    item_id = self._selection_pass.getIdAtPosition(event.x, event.y)
AttributeError: 'NoneType' object has no attribute 'getIdAtPosition'

-------------------------
Thread 0x00003264 (most recent call first):
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 156 in _storeStderrToLogThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x0000059c (most recent call first):
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 148 in _storeOutputToLogThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002604 (most recent call first):
  File "C:\Program Files\Cura 3.0\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000034f0 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002460 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000025e8 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002b04 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap
[... 79 lines cut ...]
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1017 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00003a9c (most recent call first):
  File "zeroconf.py", line 946 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Current thread 0x0000371c (most recent call first):
  File "X:\3.0\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 100 in show
  File "X:\3.0\build\inst\bin\cura_app.py", line 45 in exceptHook
  File "X:\3.0\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 686 in run
  File "X:\3.0\build\inst\bin\cura_app.py", line 82 in <module>
  File "Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

nallath, 2017-10-17:
Could you also provide the entire log file (upload it to paste bin)? It seems something went wrong before the crash.

trennal, 2017-10-17:
I will provide the log this evening. I don’t think it’s a beta issue because I was getting the same response from the earlier version.

I’m pretty sure it has to do with a Windows upgrade, since that’s the only changes that have occurred. I’m hoping to resolve this so I csn get back to printing.

Thanks!
Lee

trennal, 2017-10-17:
Where can I find the entire log file?

trennal, 2017-10-18:
Logs attached.

nallath, 2017-10-18:
Replying with attachments over email doesn't work i'm afraid.

trennal, 2017-10-18:
Here is the stderr.log:
2017-10-17 17:30:25,490 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ConsoleLogger
2017-10-17 17:30:25,505 - INFO - CuraEngineBackend.CuraEngineBackend.__init__ [64]: Found CuraEngine at: C:\Program Files\Cura 3.0\CuraEngine.exe
2017-10-17 17:30:25,505 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CuraEngineBackend
2017-10-17 17:30:25,521 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CuraProfileReader
2017-10-17 17:30:25,537 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CuraProfileWriter
2017-10-17 17:30:25,537 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OBJReader
2017-10-17 17:30:25,557 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OBJWriter
2017-10-17 17:30:25,572 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin STLReader
2017-10-17 17:30:25,588 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin STLWriter
2017-10-17 17:30:25,588 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin FileLogger
2017-10-17 17:30:25,651 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin FirmwareUpdateChecker
2017-10-17 17:30:25,671 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeProfileReader
2017-10-17 17:30:25,671 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeReader
2017-10-17 17:30:25,686 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeWriter
2017-10-17 17:30:25,702 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ImageReader
2017-10-17 17:30:25,717 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LayerView
2017-10-17 17:30:25,733 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LegacyProfileReader
2017-10-17 17:30:25,749 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LocalFileOutputDevice
2017-10-17 17:30:25,764 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin MachineSettingsAction
2017-10-17 17:30:25,800 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OctoPrintPlugin
2017-10-17 17:30:25,816 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PerObjectSettingsTool
2017-10-17 17:30:25,831 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PluginBrowser
2017-10-17 17:30:25,847 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PostProcessingPlugin
2017-10-17 17:30:25,863 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin RemovableDriveOutputDevice
2017-10-17 17:30:25,881 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SliceInfoPlugin
2017-10-17 17:30:25,886 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SolidView
2017-10-17 17:30:25,901 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CameraTool
2017-10-17 17:30:25,917 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin MirrorTool
2017-10-17 17:30:25,932 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin RotateTool
2017-10-17 17:30:25,948 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ScaleTool
2017-10-17 17:30:25,964 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SelectionTool
2017-10-17 17:30:25,979 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin TranslateTool
2017-10-17 17:30:25,999 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UltimakerMachineActions
2017-10-17 17:30:26,031 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UM3NetworkPrinting
2017-10-17 17:30:26,046 - INFO - UpdateChecker.UpdateCheckerJob.run [44]: Checking for new version of cura
2017-10-17 17:30:26,046 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UpdateChecker
2017-10-17 17:30:26,078 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin USBPrinting
2017-10-17 17:30:26,093 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade21to22
2017-10-17 17:30:26,113 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade22to24
2017-10-17 17:30:26,129 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade25to26
2017-10-17 17:30:26,144 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade26to27
2017-10-17 17:30:26,160 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade27to30
2017-10-17 17:30:26,191 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SimpleView
2017-10-17 17:30:26,207 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin X3DReader
2017-10-17 17:30:26,227 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin XmlMaterialProfile
2017-10-17 17:30:26,241 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin XRayView
2017-10-17 17:30:26,241 - INFO - UM.Qt.QtApplication.__init__ [114]: Command line arguments: {'file': [], 'external-backend': False, 'single_instance': False, 'qmljsdebugger': None, 'disable-textures': False}
2017-10-17 17:30:26,257 - INFO - UM.VersionUpgradeManager.upgrade [112]: Looking for old configuration files to upgrade.
2017-10-17 17:30:26,288 - INFO - UpdateChecker.UpdateCheckerJob.run [78]: Found a new version of the software. Spawning message
2017-10-17 17:30:26,304 - DEBUG - UM.Backend.Backend._logSocketState [174]: Socket state changed to Listening
2017-10-17 17:30:26,320 - INFO - UM.Backend.Backend.startEngine [75]: Started engine process: C:\Program Files\Cura 3.0\CuraEngine.exe
[... cut ...]
2017-10-17 17:31:23,852 - CRITICAL - cura.CrashHandler.show [42]:     self._pixelSelection(event)
2017-10-17 17:31:23,868 - CRITICAL - cura.CrashHandler.show [42]:   File "C:\Program Files\Cura 3.0\plugins\Tools\SelectionTool\SelectionTool.py", line 104, in _pixelSelection
2017-10-17 17:31:23,868 - CRITICAL - cura.CrashHandler.show [42]:     item_id = self._selection_pass.getIdAtPosition(event.x, event.y)
2017-10-17 17:31:23,868 - CRITICAL - cura.CrashHandler.show [42]: AttributeError: 'NoneType' object has no attribute 'getIdAtPosition'

trennal, 2017-10-20:
Jaime,

I just downloaded the latest 3.0.3 version and it open to a white blank screen.  The only changes was the update from Windows.  When will you have an update that accommodates the new Windows 10 updates? Cura is very important to my ability to produce 3D prints.

Thanks,

stderr.log:
2017-10-20 16:56:02,058 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ConsoleLogger
2017-10-20 16:56:02,090 - INFO - CuraEngineBackend.CuraEngineBackend.__init__ [64]: Found CuraEngine at: C:\Program Files\Cura 3.0\CuraEngine.exe
2017-10-20 16:56:02,106 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CuraEngineBackend
2017-10-20 16:56:02,140 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CuraProfileReader
2017-10-20 16:56:02,175 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CuraProfileWriter
2017-10-20 16:56:02,206 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OBJReader
2017-10-20 16:56:02,223 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OBJWriter
2017-10-20 16:56:02,257 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin STLReader
2017-10-20 16:56:02,298 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin STLWriter
2017-10-20 16:56:02,323 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin FileLogger
2017-10-20 16:56:02,419 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin FirmwareUpdateChecker
2017-10-20 16:56:02,440 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeProfileReader
2017-10-20 16:56:02,474 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeReader
2017-10-20 16:56:02,507 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeWriter
2017-10-20 16:56:02,540 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ImageReader
2017-10-20 16:56:02,590 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LayerView
2017-10-20 16:56:02,623 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LegacyProfileReader
2017-10-20 16:56:02,657 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LocalFileOutputDevice
2017-10-20 16:56:02,690 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin MachineSettingsAction
2017-10-20 16:56:02,790 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OctoPrintPlugin
2017-10-20 16:56:02,845 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PerObjectSettingsTool
2017-10-20 16:56:02,873 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PluginBrowser
2017-10-20 16:56:02,907 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PostProcessingPlugin
2017-10-20 16:56:02,957 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin RemovableDriveOutputDevice
2017-10-20 16:56:03,006 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SliceInfoPlugin
2017-10-20 16:56:03,040 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SolidView
2017-10-20 16:56:03,073 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CameraTool
2017-10-20 16:56:03,118 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin MirrorTool
2017-10-20 16:56:03,156 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin RotateTool
2017-10-20 16:56:03,207 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ScaleTool
2017-10-20 16:56:03,240 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SelectionTool
2017-10-20 16:56:03,273 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin TranslateTool
2017-10-20 16:56:03,340 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UltimakerMachineActions
2017-10-20 16:56:03,439 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UM3NetworkPrinting
2017-10-20 16:56:03,457 - INFO - UpdateChecker.UpdateCheckerJob.run [44]: Checking for new version of cura
2017-10-20 16:56:03,473 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UpdateChecker
2017-10-20 16:56:03,573 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin USBPrinting
2017-10-20 16:56:03,640 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade21to22
2017-10-20 16:56:03,673 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade22to24
2017-10-20 16:56:03,707 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade25to26
2017-10-20 16:56:03,740 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade26to27
2017-10-20 16:56:03,783 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade27to30
2017-10-20 16:56:03,807 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SimpleView
2017-10-20 16:56:03,857 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin X3DReader
2017-10-20 16:56:03,907 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin XmlMaterialProfile
2017-10-20 16:56:03,940 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin XRayView
2017-10-20 16:56:03,957 - INFO - UM.Qt.QtApplication.__init__ [114]: Command line arguments: {'disable-textures': False, 'external-backend': False, 'qmljsdebugger': None, 'file': [], 'single_instance': False}
2017-10-20 16:56:03,992 - INFO - UM.VersionUpgradeManager.upgrade [112]: Looking for old configuration files to upgrade.
2017-10-20 16:56:04,157 - DEBUG - UM.Backend.Backend._logSocketState [174]: Socket state changed to Listening
2017-10-20 16:56:04,174 - INFO - UM.Backend.Backend.startEngine [75]: Started engine process: C:\Program Files\Cura 3.0\CuraEngine.exe
[... cut ...]
2017-10-17 17:31:23,852 - CRITICAL - cura.CrashHandler.show [42]:     self._pixelSelection(event)
2017-10-17 17:31:23,868 - CRITICAL - cura.CrashHandler.show [42]:   File "C:\Program Files\Cura 3.0\plugins\Tools\SelectionTool\SelectionTool.py", line 104, in _pixelSelection
2017-10-17 17:31:23,868 - CRITICAL - cura.CrashHandler.show [42]:     item_id = self._selection_pass.getIdAtPosition(event.x, event.y)
2017-10-17 17:31:23,868 - CRITICAL - cura.CrashHandler.show [42]: AttributeError: 'NoneType' object has no attribute 'getIdAtPosition'

Ghostkeeper, 2017-10-22:
I'm not seeing the error you're getting, but let's try to analyse where it came from:

    def _pixelSelection(self, event):
        # Find a node id by looking at a pixel value at the requested location
        item_id = self._selection_pass.getIdAtPosition(event.x, event.y)

So that means that `self._selection_pass` must be `None`. The only place where `_pixelSelection` is called is here:

    if self._selection_pass is None:
        self._selection_pass = self._renderer.getRenderPass("selection")

    self.checkModifierKeys(event)
    if event.type == MouseEvent.MousePressEvent and MouseEvent.LeftButton in event.buttons and self._controller.getToolsEnabled():
        # Perform a selection operation
        if self._selection_mode == self.PixelSelectionMode:
            self._pixelSelection(event)

Note the check at the start there to make sure that `self._selection_pass` is no longer `None`. It can now only be `None` if `self._renderer.getRenderPass("selection")` also returned None.

    def getRenderPass(self, name):
        for render_pass in self._render_passes:
            if render_pass.getName() == name:
                return render_pass

        return None

So that means that no render pass with the name "selection" was ever added. Where should it get added? Right here: https://github.com/Ultimaker/Uranium/blob/15cccf300f0de5405ac72a0344cf9aef9ce6ad73/UM/Qt/QtRenderer.py#L180

    def _initialize(self):
        supports_vao = OpenGLContext.supportsVertexArrayObjects()  # fill the OpenGLContext.properties
        Logger.log("d", "Support for Vertex Array Objects: %s", supports_vao)

        OpenGL.setInstance(OpenGL())
        self._gl = OpenGL.getInstance().getBindingsObject()

        self._default_material = OpenGL.getInstance().createShaderProgram(Resources.getPath(Resources.Shaders, "default.shader"))

        self._render_passes.add(DefaultPass(self._viewport_width, self._viewport_height))
        self._render_passes.add(SelectionPass(self._viewport_width, self._viewport_height))

The `DefaultPass` and `CompositePass` are also added there, so if you got a white screen that made sense. None of the three got added apparently. Why not? This _initialize function gets called upon the first call to `beginRendering`, which is called at the `beforeRendering` signal emit in `QQuickWindow`.

I can think of two reasons that might cause this to break specifically in your updated version of Windows 10:
* The `beforeRendering` signal is no longer being emitted by the window.
* There is an error somewhere along the way which gets hidden because it's in Qt (which is outside of Python's normal exception environment) and that thread gets shut down. Perhaps it's the `OpenGL()` intialization.

That's about as far as my knowledge reaches... I can't give a solution to this.

trennal, 2017-10-22:
Thanks for the thorough explanation. I will reach out to the video display adapter to see if they have an update that may address this issue.

trennal, 2017-10-23:
Driver update did the job. Thanks.

closed 2017-10-23 by ChrisTerBeke

MarcelPalmen, 2018-01-31:
Hello Jaime,

Sorry for my late reply.

How do I remov the Barbarian plugin?

Best regards,

Marcel Palmen

From: Jaime van Kessel 
Sent: Monday, October 16, 2017 9:26 AM
To: Ultimaker/Cura 
Cc: MarcelPalmen ; Author 
Subject: Re: [Ultimaker/Cura] Crash at start (#2598)

I don't know what version of Cura you are using; "JGcreat-2.5.0" is not a version that i know of.

Other than that; It's using the wrong version of the barbarian plugin, which is not backwards compatible. Remove the plguin and try again

—

Version: JGcreat-2.5.0

Platform: Windows-8.1-6.3.9600

Qt: 5.8.0

PyQt: 5.8.2

Exception:

Traceback (most recent call last):

File "<frozen importlib._bootstrap>", line 969, in _find_and_load

File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked

File "<frozen importlib._bootstrap>", line 664, in _load_unlocked

File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible

File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\lib\site-packages\cx_Freeze\initscripts\__startup__.py", line 12, in <module>

File "<frozen importlib._bootstrap>", line 969, in _find_and_load

File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked

File "<frozen importlib._bootstrap>", line 664, in _load_unlocked

File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible

File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\lib\site-packages\cx_Freeze\initscripts\Console.py", line 24, in <module>

File "./inst/bin/cura_app.py", line 72, in <module>

File "D:/Cura_2.5/cura-build-2.5.0/build/inst/lib/python3.5/site-packages/UM\Application.py", line 316, in getInstance

File "D:/Cura_2.5/cura-build-2.5.0/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 196, in __init__

File "D:/Cura_2.5/cura-build-2.5.0/build/inst/lib/python3.5/site-packages/UM\Qt\QtApplication.py", line 110, in __init__

File "D:/Cura_2.5/cura-build-2.5.0/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 478, in _loadPlugins

File "D:/Cura_2.5/cura-build-2.5.0/build/inst/lib/python3.5/site-packages/UM\PluginRegistry.py", line 224, in loadPlugins

File "D:/Cura_2.5/cura-build-2.5.0/build/inst/lib/python3.5/site-packages/UM\PluginRegistry.py", line 244, in getMetaData

File "D:/Cura_2.5/cura-build-2.5.0/build/inst/lib/python3.5/site-packages/UM\PluginRegistry.py", line 359, in _populateMetaData

[... 64 lines cut ...]
File "D:/Cura_2.5/cura-build-2.5.0/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run

File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 914 in _bootstrap_inner

File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 882 in _bootstrap

Thread 0x00002018 (most recent call first):

File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 293 in wait

File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 424 in acquire

File "D:/Cura_2.5/cura-build-2.5.0/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob

File "D:/Cura_2.5/cura-build-2.5.0/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run

File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 914 in _bootstrap_inner

File "C:\Users\Administrator\AppData\Local\Programs\Python\Python35\Lib\threading.py", line 882 in _bootstrap

Current thread 0x000021e4 (most recent call first):

File "D:/Cura_2.5/cura-build-2.5.0/build/inst/lib/python3.5/site-packages/cura\CrashHandler.py", line 100 in show

File "./inst/bin/cura_app.py", line 45 in exceptHook

LipuFei, 2018-02-01:
Hi @MarcelPalmen , you can find your plugins here: `C:\Users\<your-username>\AppData\Roaming\cura\<version-you-are-running>\plugins\<plugin-name>`. Remove the Barbarian plugin's folder there.