Ultimaker/Cura #2599: Unable to correctly scale 10x10cm x3d files generated by Terrafab

petterreinholdtsen, 2017-10-13:
When trying to load a 3D model of a landscape generated by http://terrafab.bengler.no/ into Cura 2.5.0, the gui show 'Loading file.x3d' for a while until it seem to be happy with the loading.  But the loaded object is way too big for the print bed, and the message "Nothing to slice because none of the models fit the build volume.  Please scale or rotate models to fit" is reported.  The size is reported to be 100352.0 x 100352.0 x 15360.0 mm.  It should be 0.1% of this.

fieldOfView, 2017-10-13:
In the preferences, on the "General" pane, there's an option to "Scale large models". It is turned off by default (by popular ~request~ demand), but you can turn it on.

petterreinholdtsen, 2017-10-13:
[Aldo Hoeben]
> In the preferences, on the "General" pane, there's an option to "Scale
> large models". It is turned off by default (by popular ~request~
> demand), but you can turn it on.

OK.  But the model in question _is_ 10x10 cm when imported into for
example meshlab (or send off to printing at Shapeways), so scaling large
models should not be needed.

-- 
Happy hacking
Petter Reinholdtsen

Ghostkeeper, 2017-10-22:
I took the example landscape from that site and looked at the file. Here's a snippet of the first few coordinates:

    <Coordinate point="-50 1.4733481009938205 -50 -49.69040247678019 1.3722817958899622 ...

Scrolling through them these coordinates seem to range from -50 to 50. So that means that it's 100mm wide or 10cm. I think our X3D loader is messing up the scale a bit here.

Ghostkeeper, 2018-02-05:
Our project manager removed this from our planning because it's older than 12 weeks.

petterreinholdtsen, 2018-02-05:
I do not understand what age has to do with the scaling issue.  Does this mean you no longer plan to spend time on fixing the incorrect scaling of X3D models?

Ghostkeeper, 2018-02-21:
Yeah. We may come back to it in the future, but we don't have it in our planning any more so it's uncertain when if ever.

The idea with the age is that if something was not reported more often than once in those 12 weeks then it would probably not be important enough.

closed 2023-06-13 by github-actions[bot]