Ultimaker/Cura #2600: Profile only has the option for fine. It was ok yesterday??

Jez73, 2017-10-13:


fieldOfView, 2017-10-13:
This needs lots more information. What version of Cura, what printer, what OS (yes, that can help troubleshooting), for starters.

Jez73, 2017-10-13:
Ok so i am running the latest version of cura 2.7.0... I have an ultimaker 3 and windows 8.
It was working ok yesterday with all the profile options, fast,normal,fine etc.
I have tried reinstalling but still only the option to print fine.

nallath, 2017-10-16:
Check the second extruder. It probably has a not supported profile.

ChrisTerBeke, 2017-10-18:
Like @nallath says, you only see profiles that are available with the material combination in both extruders. Make sure both extruders are fitted with a material that supports the wanted quality profile (this is done to ensure good printing results).

closed 2017-10-31 by ChrisTerBeke