Ultimaker/Cura #2602: Cura 3 Open/save file issue

gerardozq, 2017-10-13:
Hi I am facing this problem with all cura 2 and cura 3 software, can anyone enlighten me whats wrong and how to fix it?

Version: 3.0.1-BETA
Platform: Windows-10-10.0.15063
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "__startup__.py", line 12, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "Console.py", line 24, in <module>
  File "X:\3.0\build\inst\bin\cura_app.py", line 81, in <module>
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\Application.py", line 321, in getInstance
  File "X:\3.0\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 206, in __init__
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 112, in __init__
  File "X:\3.0\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 519, in _loadPlugins
RuntimeError: Could not load the backend plugin!

-------------------------
Thread 0x000027dc (most recent call first):
  File "C:\Program Files\Cura 3.0\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002ae8 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000009dc (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000000dc (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00003630 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00003104 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1153 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001fcc (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1017 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002e08 (most recent call first):
  File "zeroconf.py", line 946 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002180 (most recent call first):
  File "C:\Program Files\Cura 3.0\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 58 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x0000399c (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1153 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000024ec (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1017 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002bdc (most recent call first):
  File "zeroconf.py", line 946 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000003dc (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1017 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x0000263c (most recent call first):
  File "zeroconf.py", line 946 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Current thread 0x00002824 (most recent call first):
  File "X:\3.0\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 100 in show
  File "X:\3.0\build\inst\bin\cura_app.py", line 45 in exceptHook

nallath, 2017-10-16:
Probably a virus scanner that is preventing the backend to be started. Could try running it without your virus scanner / firewall?

ianpaschal, 2018-03-09:
Never heard back from OP so closing.

closed 2018-03-09 by ianpaschal