Ultimaker/Cura #2604: small print failure

fgvdla, 2017-10-14:
When I print a U profile with vertical legs of a different size, Cura prints on the inner side of the long leg a small edge at the position of the end of the small leg. The cad design is oke, Netfabb cannot find any issue.

nallath, 2017-10-16:
Could you provide some pictures of what you mean?

Ghostkeeper, 2017-10-22:
I think what happens is that the smaller leg is now stopped so the nozzle doesn't need to retract any more and the print speed is way faster. This causes a change in flow at that height and also a change in temperature (since there's less time to cool off).

You could try tuning it away by setting the minimum layer time higher, but you'll always have a little border there I'm afraid.

closed 2018-03-02 by ChrisTerBeke