Ultimaker/Cura #2607: PVA Default Standby Temperature

lgreenlee, 2017-10-14:
The PVA default standby temperature of 100 C results in substantial ooze when using the PVA material and will cause print failures. The default temperature should be just below the usage temperature. I've had a good experience with 185 C and I am continuing to test higher values.

The root cause of this issue is as follows: The Ultimaker 3 uses the Extruder 1 print time to heat up Extruder 2. The PVA material undergoes thermal expansion during this time (even with retraction after layer change enabled). While Extruder 2 is heating up Extruder 1 is still printing and the ooze worm that develops is large and inevitably gets dragged around the print and ends up somewhere it should not be. If the worm  happens to make it then it is placed at the start of Extruder 2's next run, which can be the tower or the print and ultimately ends up making a destroying the supports.

Perhaps other options include disabling the early heating?

Ghostkeeper, 2017-10-22:
We could disable the pre-heat, but this would roughly double the print time. Also, in my experience before pre-heating was implemented, heating after the nozzle switch just causes the material to ooze anyway while it's heating.

I haven't seen that much ooze that the nozzle lifting is insufficient to make the blob end up on the print while the previous extruder is still working. It happens after the extruder switch though. The prime tower doubles as a wipe tower for this purpose (it has a small wiping routine in there somewhere). Some material gets lost, but that's why the prime tower is called a prime tower: It makes sure that the material is properly flowing before starting with your pretty walls.

If the prime tower is disabled, then the blob and consequent underextrusion ends up in the actual print. That's bad, but you saved time by not printing the prime tower. It's a quality vs. time balance here.

If you find a better stand-by temperature, we'll ping @tomheijmans to let him try it out if he pleases.

lgreenlee, 2017-10-22:
Thanks, I have been using 165-185C as the standby temp and things are OK with this. I tried 195C but some cooking occurred.

I have been using the prime tower too, I've had issues with the mixed material towers, they have all failed with PLA and PVA. A part that I'm printing now is 100mm tall and I have not had any towers survive this height. I submitted a separate issue for having separate prime towers though.

Ghostkeeper, 2017-10-31:
Which PVA brand was this?

Ghostkeeper, 2017-10-31:
We've also seen the problem with mixed PVA/PLA prime towers. If I recall, it's the main reason why we disabled the prime tower for that combination.

lgreenlee, 2017-10-31:
Hi Thanks,

I am using the Ultimaker PVA.

The towers are disabled by default, the issue is that the flow of the PVA
is not well managed and makes a mess of models with large vertical
supports. I have mitigated the heat up worm by using the 165C standby
temperature but there continues to be excessive flow after the layer
completes in the default profile. The tower helps grab this when switching
materials.

I also noticed that the 230C initial temperature is above Ultimkers listed
boiling point of 228C. I believe that the default settings are too hot for
the PVA material.

Thank you,
Logan

Ghostkeeper, 2017-11-13:
The initial layer temperature we list for PVA is 220, not 230.

@tomheijmans tested the higher stand-by temperature and didn't get good results:

[image]

The material degraded pretty quickly in the nozzle.

lgreenlee, 2017-11-13:
Thanks, Sorry the initial layer temperature is 225 not 230. Still very
close to the boiling point. I have noticed that especially during the blob
making that the PVA appears to become carbonized and sticks to the model.
This causes extruded PVA to stick to the melted PVA. I have started
applying high temperature grease to the nozzle to help mitigate the
adhesion, it helps for a while but ultimately is not a solution.

Could you send me this model? I'd like to try myself.

[image: image.png]

Ghostkeeper, 2017-11-20:
@tomheijmans Do you still have that model?

tomheijmans, 2017-11-20:
@Ghostkeeper, don't know how I got involved in this thread. I'm not familiar with Cura. Think you mentioned the wrong person.

Ghostkeeper, 2017-11-27:
Ah sorry Tom. I mistook you for a colleague, I think.

closed 2018-03-02 by ChrisTerBeke