Ultimaker/Cura #2608: Improved prime tower behavior

lgreenlee, 2017-10-14:
This request is based on my experience using PVA and the Ultimaker 3.

The combined prime tower approach works well for single or like filaments. When printing with PVA the tower can sometimes have a hard time forming if the PVA and the other material do not bond well.

Having individual prime tower configurations could help bonding by only having like material within a single structure. This will be a big help when printing with PVA.

Ghostkeeper, 2017-10-22:
So you suggest creating a separate prime tower for each extruder.

What would happen if you print, say, a [traffic cone](https://www.thingiverse.com/thing:21773) that doesn't need the other extruder for a few layers? Will it have to switch extruders every layer to just print the second prime tower? Or will it go down to the old height of the prime tower? Both have significant disadvantages.

lgreenlee, 2017-10-22:
This is true, there may be cases where the support layer is not used.  If the option is selected, then a tower layer is printed regardless of need. I'd offer the separate towers as an option.

In my case, the tower is getting printed and is eventually getting destroyed due to the PVA making a mess. This is causing the support to be dragged into the model as well as the PLA. I would take the extra time for the layer changes over having a product that I need to "clean up" or that has defects. I can share pictures if you like.

maukcc, 2017-10-23:
> I would take the extra time for the layer changes over having a product that I need to "clean up" or that has defects

I will have to agree with that.(switch extruders every layer)

In the end, the user should decide: do I want to print 1 or multiple towers

ChrisTerBeke, 2017-12-21:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/1936 and https://github.com/Ultimaker/CuraEngine/issues/588 (the latter of which might have been using the same diamond extruder).

lgreenlee, 2017-12-21:
> Ultimaker/CuraEngine#588

Hello, this is not a duplicate of that case. My request is for an Ultimaker
3 Extended, a product you actually produce. I can understand not want to
support esoteric 3rd party hardware, but the reality is that Ulitmaker has
sold the community a bill of good that doesn't exist with PVA extrusion. It
simply does not produce prints that are a reasonable quality over the
entire dimension of the build volume. Part of the reason for this is
switching between PVA and the main material makes a mess and can ruin the
entire print, especially as the prints get taller. Having separate build
towers would prevent this from happening.

ChrisTerBeke, 2017-12-21:
Fair enough, although both issues require the same solution.

lgreenlee, 2017-12-21:
Thank you, Chris

yyh1002, 2018-01-29:
May I suggest an idea which may make the prime tower more reliable?
Adding solid wall or column printed purely in PLA to the prime tower. 
According to my experience of slicing soluble support models, primary extruder is used for every layer in most cases. Therefore adding PLA to every layer of prime tower would neither add unnecessary nozzle switching nor extend print time very much. There can be columns or wall around the tower printed in PLA to keep the tower rigid and help the inconsistent bond between PLA and PVA. 
Hope the following image shows the idea clearly. 
[image]
It would also be good to have dedicated brim setting for prime tower.

Ghostkeeper, 2018-01-29:
Cura has no notion of which extruder would be desirable to use for filling the prime tower though. You'd have to have a selector drop-down for that. Or try to choose *the other one than the support extruder* automatically, but that would further eliminate feasibility of using more than 2 extruders in the future.

yyh1002, 2018-01-29:
The column doesn't stop 3 extruders or wiping nozzle. It's just adding a consistent structural element that has proper layer adhesion and layer height. The strength of the tower won't be affected by bonding of two material and accuracy of z offset between nozzles. Any of the extruders can be used to produce the column, and it will always be strong because it's adhering with itself. It's only that it would be more expensive to have PVA in the column head.

kevinjones4642, 2018-02-26:
Not sure if this is the right place to ask, but usually when I use a prime tower it is hollow, but in my last print it was filled. I don't think I changed any settings that would have caused this. Does anyone know why?

Ghostkeeper, 2018-03-16:
Maybe you've set the "Prime Tower Minimum Volume" setting way too high?

kevinjones4642, 2018-03-21:
Here's some follow up information. I'm now using Cura 3.2.1, and I have found some very peculiar behavior with both the prime tower and the use of support material. They both seem to be linked to the layer height. I sliced a part using 0.15 layer height, and Cura skips support material under most of the part, and builds a prime tower with walls that are about 2.3mm thick. I re-slice the same part with 0.10 layer height, and Cura now provides support everywhere that it is needed, but the prime tower is now completely filled. The only parameter I changed was layer height. The part is being printed with TPU and the support is PLA. I'm not too concerned about the filled prime tower (only about 1cm cubed of material), but the lack of support material for the 0.15 slice height is completely trashing the part.

kevinjones4642, 2018-03-21:
Maybe I just figured out the prime tower issue. It seems that the fill of the prime tower is directly related to the layer height. I'm guessing that it is trying to purge a given volume of material, so the thinner the layer, the thicker the wall needs to be. Does this sound about right?

I would still really like to figure out the support material issue though. As the part wont print right without support, but print time with 0.1 layer height is nearly 8 hours - it's a small part.

Ghostkeeper, 2018-03-21:
Yes, the prime tower changes in thickness when you change the layer height because it's specified by the "Prime Tower Purge Volume" setting. It puts a specific amount of material volume on the prime tower each layer. So if you have thinner layers this volume gets squashed over a larger area.

The logic behind this is that the amount of volume oozed out of the nozzle in stand-by stance does not depend on layer height. But you can also specify a specific prime tower thickness if you want.

Ghostkeeper, 2019-06-17:
This feature request has been implemented in Cura 3.5 onwards. It's taken a bit of a different form, but the behaviour is the same: One prime tower is printed inside another, causing one of the extruders to always print which makes the prime tower strong regardless of how well the two materials adhere to each other.

closed 2019-06-17 by Ghostkeeper