Ultimaker/Cura #2609: Crash: Exiting Editing Machine Settings

azide0x37, 2017-10-15:
Occurs regardless if any settings are changed, or if auto slicing is enabled or disabled. 

Machine/printer object structure changed between 2.7 and 3? 

Version: 3.0.1-BETA
Platform: Darwin-17.0.0-x86_64-i386-64bit
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "/Applications/Cura.app/Contents/MacOS/plugins/plugins/MachineSettingsAction/MachineSettingsAction.py", line 82, in onFinishAction
    if self._backend.determineAutoSlicing():
AttributeError: 'NoneType' object has no attribute 'determineAutoSlicing'

-------------------------
Thread 0x0000700009fd8000 (most recent call first):
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 156 in _storeStderrToLogThread
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700009ad5000 (most recent call first):
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 148 in _storeOutputToLogThread
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007000095d2000 (most recent call first):
  File "/Applications/Cura.app/Contents/MacOS/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007000090cf000 (most recent call first):
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/site-packages/zeroconf.py", line 1153 in run
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700008bcc000 (most recent call first):
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/site-packages/zeroconf.py", line 1017 in run
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007000086c9000 (most recent call first):
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007000081c6000 (most recent call first):
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/site-packages/zeroconf.py", line 1017 in run
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700007cc3000 (most recent call first):
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007000077c0000 (most recent call first):
  File "/Applications/Cura.app/Contents/MacOS/plugins/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 58 in _updateThread
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 882 in _bootstrap
[... 78 lines cut ...]

Thread 0x000070000409f000 (most recent call first):
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Current thread 0x00007fffb00ba340 (most recent call first):
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 100 in show
  File "/Users/ultimaker/build/3.0/build/inst/bin/cura_app.py", line 45 in exceptHook
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 686 in run
  File "/Users/ultimaker/build/3.0/build/inst/bin/cura_app.py", line 82 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

azide0x37, 2017-10-15:
Race condition in `onFinishAction()`?

`     if self._backend:  # This sometimes triggers before backend is loaded.`

fieldOfView, 2017-10-15:
Yes, that would fix it. I'll make a PR.

PR #2610 (2017-10-16): Fix crash when exiting Machine Settings
This PR fixes #2609
commit ad1c0046 (2017-10-16): Fix crash when exiting Machine Settings

fixes #2609
closed 2017-10-16 by alekseisasin