Ultimaker/Cura #2611: bfb flavor not working

Joshuawilkin, 2017-10-15:
i would love to use cure on my bfb 3d touch but the gcode files do not work (the extruder does not switch on) can you address this problem please, i am not the only person that would like this to be fixed as shown an many forums.

nallath, 2017-10-16:
We can't (and won't) put time into debugging third party machines. We don't own any of those and our priority is in getting Cura to work for our own machines. If someone else decides to fix this, pull requests are of course always welcome.

Ghostkeeper, 2017-10-22:
A lot of those problems are probably in CuraEngine somewhere. It has some exceptions riddled throughout GcodeExport specifically for the BFB g-code flavour.

Perhaps you know what could be changed to "switch on" that extruder? What changes would you need to be able to print? As Nallath said, we can't test anything because we don't have any BFB printer. And what's more, we're not allowed to spend time on it during working hours. Only in our free time. But if we know what changes must be made to the output, perhaps a pull request can be added to fix it up.

closed 2018-03-02 by ChrisTerBeke