Ultimaker/Cura #2615: Cura 3 Beta: Raft does not go to correct size when printing with supports

DubSidious, 2017-10-17:
Just thought i'd let you know that im trying to print with a raft on the beta of cura 3, and the raft does not seem to go to the correct size, meaning part of the first layer would print in mid air.
[image]

Picture attached to show what i mean. Not moaning in anyway, i know its a beta :) Just thought i'd let you guys know about the issue

Thanks for the software, and keep up the good work. Love using cura, have done since trying it on my first day of printing.

Ghostkeeper, 2017-10-27:
Usually when this happens it turns out that the bottom of the model was not entirely flat at the bottom, or not laid flat properly.

DubSidious, 2017-10-27:
Seems the problem went away with the release of the latest version.
And can assure you the bottoms of those objects were perfectly flat. Have printed this item with those stl's many times before.

closed 2017-10-31 by ChrisTerBeke