Ultimaker/Cura #2616: Bed re leveling constantly only on x and y axis

zerithcj, 2017-10-17:
Printer is randomly homing the X and Y axis during a print and this only happens which cura 2.x and 3 but does not happen with 15.04.6. I have tried looking at start and end gcode but they both are pretty much the same. I even did a re install but the problem still persists and can not figure out why this is happening whatsoever

nallath, 2017-10-18:
What machine are you using?

zerithcj, 2017-10-18:
@nallath i am currently using maker select 2.1

Ghostkeeper, 2017-10-29:
Maker Select 2.1 is not one of the pre-configured printers. So it's hard to say what's happening. Perhaps a fragment of g-code that contains the homing commands could help, or else exporting a project file (by loading up a model which reproduces the problem, then clicking File -> Save Project).

I predict that it's one of these two causes:
* The Maker Select 2.1 has weird firmware that interprets one of our commands as a command to home, such as the acceleration command or something.
* Your configuration is such that it happens to move to position 0,0. I've seen this happen quite a lot with dual-extrusion prints because the default nozzle switch location is 0,0.

zerithcj, 2017-10-29:
The issue happens with every model I try to print however the gcode looks very very similar (if not the same) to 15.04.

Cura 3.0.3 Start GCode
```
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G1 Z15.0 F9000 ;move the platform down 15mm
G92 E0 ;zero the extruded length
G1 F200 E3 ;extrude 3mm of feed stock
G92 E0 ;zero the extruded length again
G1 F9000
;Put printing message on LCD screen
M117 Printing...
```
and 15.04.6
```
G21        ;metric values
G90        ;absolute positioning
M82        ;set extruder to absolute mode
M107       ;start with the fan off
G28 X0 Y0  ;move X/Y to min endstops
G28 Z0     ;move Z to min endstops
G1 Z15.0 F{travel_speed} ;move the platform down 15mm
G92 E0                  ;zero the extruded length
G1 F200 E3              ;extrude 3mm of feed stock
G92 E0                  ;zero the extruded length again
G1 F{travel_speed}
;Put printing message on LCD screen
M117 Printing...
```

Ghostkeeper, 2017-11-06:
I wouldn't think it's the starting g-code since you indicated that it happened halfway through the print.

zerithcj, 2017-11-06:
[](https://youtu.be/KTY4yaRUbo0)
Heres a video of what it is doing

ChrisTerBeke, 2017-11-06:
Are you printing over USB or SD card? I've seen this happening to many printers when using a USB connecting, regardless of firmware or USB host software. In most of those cases the issue was half or incorrectly executed G-code lines (if the G-code sent to the printer is not validated and checked by the firmware using N numbering commands). Try a few different printing methods and host softwares and see if you can find a pattern.

zerithcj, 2017-11-12:
I am printing over USB and it also seems that saving the model into gcode and using repetier host seems to also do good

TreyDevDenver, 2018-11-26:
I had to setup Cura (3.6.0) from scratch to work with my Maker Select v2.1 today and encountered this issue. The problem is that the start/stop gcode included in the profile for the Wanhao Duplicator i3 (same machine as the MS) has a reference to `travel_speed` that does not get replaced (and warns: `CuraEngineBackend.StartSliceJob.get_value [72]: Unable to replace 'travel_speed' placeholder in start/end g-code`). 

Fix is to replace the deprecated `travel_speed` in the start and stop gcode for this machine with `speed_travel`. Looks like there's an open PR for this: #4717.

GregValiant, 2024-11-13:
This appears to be a non-UM Cura issue.  Can we close this?

closed 2024-11-14 by cooljas-was-taken